espressif/esp-idf #13782: ESP32 BLE+light-sleep issue with power supply

AhmedSalem636, 2024-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello ESP community,

I have a problem that I would like to discuss here hoping someone could give me some insights.

**Project overview:**
I am working on a battery-powered device that uses ESP32 for BLE connection. Since the device is battery powered, the power saving is an essential here. In order to save power, I was advised to follow the esp-idf power-save example that runs BLE with automatic light-sleep using external crystal. I have done that which does in fact save power as expected. 

**Issue:**
However, after running the system for some time, the ESP halts. After some investigations using oscilloscope, I found out that the problem was associated with power supply.
The power supply I am using is a switching mode buck converter that's capable of delivering up to 1.5Ampare. 
So, I have run a few tests with automatic light-sleep enabled and disabled and here are the results:
1- disabled: the system runs pretty well, and the power supply can run the ESP with no issue " it doesn't even heat up a bit"
2- enabled:  when the light-sleep is enabled, I see the issue comes back. So, after looking at the 3v3 line with oscilloscope, I saw that using the automatic light-sleep with the idf power management component, the ESP keeps switching very fast between active and sleeping. 
So, I concluded that the power supply can deliver the required power, but it is just unable to cope with the speed at which the ESP is switching on&off.

ESP itself wakes up once very 30 seconds then goes back to sleep as it is idle most of the time. I believe that this switching frequency has something to do with the external crystal and the Bluetooth module itself.

So, I am wondering if there is a way to control or slow down this switching frequency a little bit so that the power supply has more time to respond to the Bluetooth actions. 
" one solution is that I used large capacitors, it worked but sometimes it fails, and I feel like this is not the ideal solution. 

**NOTE:** 90% of the buck converters that has a quiescent current under 400uA has a slow response time + I am limited to the local and available components in my area.

xiongweichao, 2024-05-13:
Hi @AhmedSalem636 , As you said, Bluetooth uses automatic light sleep and constantly switches between active and sleeping. Because Bluetooth will exit sleep before each packet is sent and received. If you want to reduce the frequency of switching, I suggest you increase the connection interval.

AhmedSalem636, 2024-05-14:
Hi Xiongweichao,

Thanks for your quick reply. So let's say I have a connection interval of 1 second and assuming the ESP itself is sleeping for 40 seconds until the next task to be active. Does that mean that during these 40 seconds, the ESP will wake-up once every second?

Also will it help if I change the supervision timeout or the connection latency?
Thanks in advance,Salem

xiongweichao, 2024-05-15:
Hi @AhmedSalem636 ,

> Does that mean that during these 40 seconds, the ESP will wake-up once every second?

Yeah!

> Also will it help if I change the supervision timeout or the connection latency?

Only affected by the connection interval.

Thanks

closed 2024-06-02 by AhmedSalem636