espressif/esp-idf #13783: [v5.1.4][S3] Toggling VCC during boot BRICKS THE DEVICE!

chipweinberger, 2024-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-s3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom pcb

### Power Supply used.

USB

### What is the expected behavior?

Device can survive power loss without bricking itself

### What is the actual behavior?

the bootloader becomes corrupted, (invalid header: 0xce5209fc)

### Steps to reproduce.

1. I unplug and replug the device quickly many times
2. eventually the device does not boot anymore (invalid header: 0xce5209fc)

Note: I have a custom bootloader that waits 1 second before booting into app (not sure if necessary for repro)

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048839
0x40048839: uart_tx_one_char_uart in ROM

invalid header: 0xce5209fc
invalid header: 0xce5209fc
invalid header: 0xce5209fc
invalid header: 0xce5209fc
invalid header: 0xce5209fc
invalid header: 0xce5209fc
```

chipweinberger, 2024-05-11:
This is a super serious issue! 

I can't ship a product that bricks itself!

chipweinberger, 2024-05-11:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/15284386/sdkconfig.txt)

chipweinberger, 2024-05-11:
[image]

I have a simple power on reset circuit.

chipweinberger, 2024-05-11:
note: I can still recover the device using GPIO 0 to put it into DOWNLOAD mode.

chipweinberger, 2024-05-11:
@igrr , appreciate your thoughts here.

chipweinberger, 2024-05-11:
full schematic. But I don't think my pcb is the problem. just for reference.

[image]

chipweinberger, 2024-05-11:
I want to be clear, `invalid header: 0xce5209fc` is triggered solely by unplugging the device from power at some unfortunate time, and then replugging it in. I have not re-flashed the device.

I use host-generated flash encryption keys, so I can normally re-flash the encrypted device, but when the device "bricks", I can not re-flash it anymore. `invalid header: 0xce5209fc` is permanent.

Its as if the flash encryption keys on the device are corrupted.

chipweinberger, 2024-05-12:
So I've been trying to reproduce this for the past 45 minutes.

**Original repro scenario (~10 attempts):**
* **pcb**: custom pcb
* **power supply:** AC usb-c charger
* **power cable:** usb-c to usb-b cable
* **uart**:  unconnected
* **enclosure**: custom 3d printed enclosure

**did not repro (~150 attempts)**
* **pcb**: dev-kit-c
* **power supply:** computer usb port
* **power cable:** usb-c to micro usb
* **uart**: connected as typical
* **enclosure**: none

**did not repro (~150 attempts)**
* **pcb**: custom pcb
* **power supply:** computer usb port
* **power cable:** connected via modified dev-kit-c
* **uart**: connected via modified dev-kit-c
* **enclosure**: none

**This leaves a couple suspects:**
1. maybe the AC usb-c charger is bad?
2. maybe the 3d printed enclosure is causing a problem?

## 3d printed enclosure

This may actually be related to the problem.

* my device was not always turning on because the enclosure would sometimes accidentally hold down a button (tied to GPIO0)

## GPIO0

My current theory is that some combination of toggling both VCC & GPIO 0 is causing the problem.

chipweinberger, 2024-05-12:
@mahavirj

chipweinberger, 2024-05-12:
**Wow no way!!!!**

**I wanted to check if somehow flash encryption got disabled, so I flashed a plaintext binary, AND IT BOOTED AS NORMAL!!**

I know this sounds incredible, but I am 100% sure flash encryption *was enabled* on this device.

Somehow, flash encryption was disabled, *after* being enabled in production mode.

My device has a command to print the fuses, and these are all the fuses that are currently set on it:

```
	"set":	[{
			"label":	"WR_DIS",
			"bits":	32,
			"offset":	"blk0:0",
			"value":	"0x0"
		}, {
			"label":	"WR_DIS_KEY_PURPOSE_1",
			"bits":	1,
			"offset":	"blk0:9",
			"value":	"0x1"
		}, {
			"label":	"WR_DIS_BLOCK_KEY1",
			"bits":	1,
			"offset":	"blk0:24",
			"value":	"0x1"
		}, {
			"label":	"WR_DIS_BLOCK_KEY2",
			"bits":	1,
			"offset":	"blk0:25",
			"value":	"0x1"
		}, {
			"label":	"RD_DIS",
			"bits":	7,
			"offset":	"blk0:32",
			"value":	"0x2"
		}, {
			"label":	"RD_DIS_BLOCK_KEY1",
			"bits":	1,
			"offset":	"blk0:33",
			"value":	"0x1"
		}, {
			"label":	"DIS_DOWNLOAD_DCACHE",
			"bits":	1,
			"offset":	"blk0:43",
			"value":	"0x1"
		}, {
			"label":	"DIS_PAD_JTAG",
			"bits":	1,
			"offset":	"blk0:51",
			"value":	"0x1"
		}, {
			"label":	"DIS_DOWNLOAD_MANUAL_ENCRYPT",
			"bits":	1,
			"offset":	"blk0:52",
			"value":	"0x1"
		}, {
			"label":	"VDD_SPI_XPD",
			"bits":	1,
			"offset":	"blk0:68",
			"value":	"0x1"
		}, {
			"label":	"VDD_SPI_TIEH",
			"bits":	1,
			"offset":	"blk0:69",
			"value":	"0x1"
		}, {
			"label":	"VDD_SPI_FORCE",
			"bits":	1,
			"offset":	"blk0:70",
			"value":	"0x1"
		}, {
			"label":	"KEY_PURPOSE_1",
			"bits":	4,
			"offset":	"blk0:92",
			"value":	"0x2"
		}, {
			"label":	"KEY_PURPOSE_2",
			"bits":	4,
			"offset":	"blk0:96",
			"value":	"0x3"
		}, {
			"label":	"DIS_USB_JTAG",
			"bits":	1,
			"offset":	"blk0:118",
			"value":	"0x1"
		}, {
			"label":	"DIS_DIRECT_BOOT",
			"bits":	1,
			"offset":	"blk0:129",
			"value":	"0x1"
		}, {
			"label":	"ENABLE_SECURITY_DOWNLOAD",
			"bits":	1,
			"offset":	"blk0:133",
			"value":	"0x1"
		}, {
			"label":	"MAC",
			"bits":	48,
			"offset":	"blk1:40",
			"value":	"0x68b6b3472ae0"
		}, {
			"label":	"WAFER_VERSION_MINOR_LO",
			"bits":	3,
			"offset":	"blk1:114",
			"value":	"0x1"
		}, {
			"label":	"BLK_VERSION_MINOR",
			"bits":	3,
			"offset":	"blk1:120",
			"value":	"0x2"
		}, {
			"label":	"K_RTC_LDO",
			"bits":	7,
			"offset":	"blk1:141",
			"value":	"0x53"
		}, {
			"label":	"K_DIG_LDO",
			"bits":	7,
			"offset":	"blk1:148",
			"value":	"0x4a"
		}, {
			"label":	"V_RTC_DBIAS20",
			"bits":	8,
			"offset":	"blk1:155",
			"value":	"0x99"
		}, {
			"label":	"V_DIG_DBIAS20",
			"bits":	8,
			"offset":	"blk1:163",
			"value":	"0x8e"
		}, {
			"label":	"DIG_DBIAS_HVT",
			"bits":	5,
			"offset":	"blk1:171",
			"value":	"0x1a"
		}, {
			"label":	"ADC2_CAL_VOL_ATTEN3",
			"bits":	6,
			"offset":	"blk1:186",
			"value":	"0x23"
		}, {
			"label":	"OPTIONAL_UNIQUE_ID",
			"bits":	128,
			"offset":	"blk2:0",
			"value":	"0xa3aba04cdf8693c296ed73c5426f423e"
		}, {
			"label":	"BLK_VERSION_MAJOR",
			"bits":	2,
			"offset":	"blk2:128",
			"value":	"0x1"
		}, {
			"label":	"TEMP_CALIB",
			"bits":	9,
			"offset":	"blk2:132",
			"value":	"0x171"
		}, {
			"label":	"OCODE",
			"bits":	8,
			"offset":	"blk2:141",
			"value":	"0x52"
		}, {
			"label":	"ADC1_INIT_CODE_ATTEN0",
			"bits":	8,
			"offset":	"blk2:149",
			"value":	"0x74"
		}, {
			"label":	"ADC1_INIT_CODE_ATTEN1",
			"bits":	6,
			"offset":	"blk2:157",
			"value":	"0x2b"
		}, {
			"label":	"ADC1_INIT_CODE_ATTEN2",
			"bits":	6,
			"offset":	"blk2:163",
			"value":	"0x1e"
		}, {
			"label":	"ADC1_INIT_CODE_ATTEN3",
			"bits":	6,
			"offset":	"blk2:169",
			"value":	"0x22"
		}, {
			"label":	"ADC2_INIT_CODE_ATTEN0",
			"bits":	8,
			"offset":	"blk2:175",
			"value":	"0x90"
		}, {
			"label":	"ADC2_INIT_CODE_ATTEN1",
			"bits":	6,
			"offset":	"blk2:183",
			"value":	"0x2d"
		}, {
			"label":	"ADC2_INIT_CODE_ATTEN2",
			"bits":	6,
			"offset":	"blk2:189",
			"value":	"0x12"
		}, {
			"label":	"ADC2_INIT_CODE_ATTEN3",
			"bits":	6,
			"offset":	"blk2:195",
			"value":	"0x20"
		}, {
			"label":	"ADC1_CAL_VOL_ATTEN0",
			"bits":	8,
			"offset":	"blk2:201",
			"value":	"0x80"
		}, {
			"label":	"ADC1_CAL_VOL_ATTEN1",
			"bits":	8,
			"offset":	"blk2:209",
			"value":	"0x7f"
		}, {
			"label":	"ADC1_CAL_VOL_ATTEN2",
			"bits":	8,
			"offset":	"blk2:217",
			"value":	"0x73"
		}, {
			"label":	"ADC1_CAL_VOL_ATTEN3",
			"bits":	8,
			"offset":	"blk2:225",
			"value":	"0x8d"
		}, {
			"label":	"ADC2_CAL_VOL_ATTEN0",
			"bits":	8,
			"offset":	"blk2:233",
			"value":	"0x78"
		}, {
			"label":	"ADC2_CAL_VOL_ATTEN1",
			"bits":	7,
			"offset":	"blk2:241",
			"value":	"0x3c"
		}, {
			"label":	"ADC2_CAL_VOL_ATTEN2",
			"bits":	7,
			"offset":	"blk2:248",
			"value":	"0x41"
		}, {
			"label":	"KEY2",
			"bits":	256,
			"offset":	"blk6:0",
			"value":	"0x45e59"
		}],

```

chipweinberger, 2024-05-12:
on a different _proper functioning device_ I can see `SPI_BOOT_CRYPT_CNT` is set as expected.

```
		}, {
			"label":	"SPI_BOOT_CRYPT_CNT",
			"bits":	3,
			"offset":	"blk0:82",
			"value":	"0x1"
		}, {
```

All these fuses are burned at the same time in my production script

```
# burn fuses
    burnXtsAesKey1(st)
    burnXtsAesKey2(st)
    burnSpiBootCryptCount(st)
    burnDisableDirectBoot(st)
    burnDisableDownloadManualEncrypt(st)
    burnDisableDownloadDCache(st)
    burnDisableDownloadICache(st)
    burnHardDisableJtag(st)
    burnDisableUsbJtag(st)
    burnEnableSecureDownloadMode(st)
```

so really, on the affected device, all evidence points to `SPI_BOOT_CRYPT_CNT` was 0x1, and after the incident happened now it is 0x0. And this caused the device to fail to boot.

chipweinberger, 2024-05-12:
also note, my devices does not use `CONFIG_SECURE_FLASH_ENC_ENABLED`. Instead I burn the fuses myself manually in my production script.

mahavirj, 2024-05-13:
@chipweinberger 

Few questions:

1. For the device where you suspect that `SPI_BOOT_CRYPT_CNT` is reset to 0, are you able to again write it to enable the flash enc?
2. Are you able to recreate this issue reliably on multiple devices?

chipweinberger, 2024-05-13:
> For the device where you suspect that SPI_BOOT_CRYPT_CNT is reset to 0, are you able to again write it to enable the flash enc?

I was waiting for your recommendation before I tried it.

I have not yet reproduced it, but i have not tried the exact repro scenario yet. I only have a few devices and was not sure if I would permanently break them. 

I will try the exact repro scenario again now that I think i can reverse the damage. will try soon

do you have any guesses what is happening?

mahavirj, 2024-05-13:
> it appears SPI_BOOT_CRYPT_CNT somehow got reset back to 0, causing flash encryption to become disabled

So the sequence goes something like this:

1. First the device was running the encrypted artifacts
2. Power cycle device multiple times
3. Device started running into boot failure 
4. Flash plaintext artifacts and the device could boot successfully

Is this correct understanding? CC @KonstantinKondrashov 

> do you have any guesses what is happening?

No. The issue looks entirely different and not something that we heard/observed previously. 

(So far) Most of the failures in the security features workflow can be attributed to errors in the efuse configuration:

- Device being abruptly power cycled whilst it was enabling the flash encryption workflow
- Programming errors in the efuse write operation. We had added some protection measures for this in the software and they are part of all latest branches. ([Reference](https://www.espressif.com/sites/default/files/advisory_downloads/AR2022-006%20Optimization%20of%20eFuse%20Programming.pdf))

chipweinberger, 2024-05-13:
> Is this correct understanding?

yes exactly. Power cycle / gpio 0 toggling.

chipweinberger, 2024-05-14:
> For the device where you suspect that SPI_BOOT_CRYPT_CNT is reset to 0, are you able to again write it to enable the flash enc?

I have results.

1. I had my affected device in SPI_BOOT_CRYPT_CNT 0, as I described
2. because Secure Download Mode is enabled, I wrote simple firmware to set SPI_BOOT_CRYPT_CNT 1
3. I flashed the firmware
4. I booted

These are the logs I got when I booted. 

```
--- esp-idf-monitor 1.4.0 on /dev/cu.usbmodem1101 460800 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x400490e9
0x400490e9: usb_uart_device_rx_one_char in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3c58,len:0x37bc
load:0x403c9700,len:0x4
load:0x403c9704,len:0x1c94
load:0x403cc700,len:0x4550
entry 0x403c9a8c
I (32) boot: ESP-IDF v5.1.1-1884-g04ab2fe17e 2nd stage bootloader
I (32) boot: compile time May 13 2024 18:14:28
I (32) boot: Multicore bootloader
I (33) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (35) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (37) boot: Enabling RNG early entropy source...
I (39) boot: Partition Table:
I (40) boot: ## Label            Usage          Type ST Offset   Length
I (41) boot:  0 jOtaData         OTA data         01 00 00010000 00002000
I (43) boot:  1 jCoredump        Unknown data     01 03 00012000 00010000
I (45) boot:  2 nvsKeys          NVS keys         01 04 00022000 00001000
I (47) boot:  3 jPhyInit         RF data          01 01 00023000 00001000
I (49) boot:  4 efuse_em         efuse            01 05 00024000 00001000
I (51) boot:  5 jOtaAesKey       Unknown data     01 06 00025000 00001000
I (53) boot:  6 jFabInfo         Unknown data     01 06 00026000 00013000
I (54) boot:  7 jStSafe          Unknown data     01 06 00039000 00004000
I (56) boot:  8 jJamcorderUuid   Unknown data     01 06 0003d000 00001000
I (58) boot:  9 nvs              WiFi data        01 02 0003e000 00033000
I (60) boot: 10 jSafeBoot        Unknown data     01 06 00071000 00006000
I (62) boot: 11 jDevCmds         Unknown data     01 06 00077000 00004000
I (64) boot: 12 jLifeStats       Unknown data     01 06 0007b000 00042000
I (66) boot: 13 jImmortal        Unknown data     01 06 000bd000 00008000
I (67) boot: 14 jJfuArgs         Unknown data     01 06 000c5000 00008000
I (69) boot: 15 jJfwmArgs        Unknown data     01 06 000cd000 00008000
I (71) boot: 16 jExtra1          Unknown data     01 06 000d5000 00004000
I (73) boot: 17 jExtra2          Unknown data     01 06 000d9000 00004000
I (75) boot: 18 jExtra3          Unknown data     01 06 000dd000 00002000
I (77) boot: 19 jExtra4          Unknown data     01 06 000df000 00001000
I (79) boot: 20 jJfuApp          OTA app          00 11 000e0000 00220000
I (81) boot: 21 jJfwmApp         OTA app          00 10 00300000 00500000
I (82) boot: End of partition table
I (84) boot: No factory image, trying OTA 0
I (85) jboot: bootloader_utility_get_selected_boot_partition: 0
I (86) jboot: Clearing ota_data Partition...
I (88) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (220) boot_comm:  0 jOtaData         data  00010000 00002000 [yes]
I (220) boot_comm:  1 jCoredump        data  00012000 00010000 [no]
I (221) boot_comm:  2 nvsKeys          data  00022000 00001000 [no]
I (222) boot_comm:  3 jPhyInit         data  00023000 00001000 [no]
I (224) boot_comm:  4 efuse_em         data  00024000 00001000 [no]
I (226) boot_comm:  5 jOtaAesKey       data  00025000 00001000 [no]
[... 47 lines cut ...]
I (2368) efuse: BURN BLOCK0
E (2370) efuse: BURN BLOCK0 - ERROR (written bits != read bits)
W (2371) efuse: BLOCK0: next retry to fix an error [2/3]...
I (2372) efuse: BURN BLOCK0
I (2375) efuse: BURN BLOCK0 - OK (all write block bits are set)
I (2375) efuse: Batch mode. Prepared fields are committed
E (2376) esp_image: image at 0x0 has invalid magic byte (nothing flashed here?)
W (2378) flash_encrypt: no valid bootloader was found
E (2380) boot: Encryption flash contents failed (261)
E (2381) boot: OTA app partition slot 0 is not bootable
E (2383) esp_image: image at 0xe0000 has invalid magic byte (nothing flashed here?)
E (2385) boot: OTA app partition slot 1 is not bootable
E (2386) boot: No bootable app partitions in the partition table
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cde6d
invalid header: 0xc9f45dd9
invalid header: 0xc9f45dd9
invalid header: 0xc9f45dd9
invalid header: 0xc9f45dd9
invalid header: 0xc9f45dd9
invalid header: 0xc9f45dd9

```

chipweinberger, 2024-05-14:
did more stuff..

**step 6**: flash encrypted firmware
**step 7**: reboot. works as normal in encrypted mode.

So in the end, I was able to recover the device. 

However, this is of course very unacceptable. A consumer of my product will not be able to recover.

mahavirj, 2024-05-16:
@chipweinberger 

> Key Purposes: (4, 2, 3, 0, 0, 0, 12)

This suggests that device has following 3 key purposes:

- `XTS_AES_128_KEY`
- `XTS_AES_256_KEY_1`
- `XTS_AES_256_KEY_2`

In this case the key used will still be 512 bit and derived from key purpose 2/3. Please refer to section 23.4.2 in the S3 TRM for more details. 

Q: you wrote this new key purpose block (4) intentionally? Which key you are using to encrypt the artifacts here?

> I (2365) efuse: BURN BLOCK0
E (2366) efuse: BURN BLOCK0 - ERROR (written bits != read bits)
W (2367) efuse: BLOCK0: next retry to fix an error [1/3]...

Are these errors common for your hardware design? Or they are occurring only on this specific device?

> I (49) boot:  4 efuse_em         efuse            01 05 00024000 00001000

Any specific purpose for this partition? This is used for the virtual efuse scenario but not something that is recommended outside of IDF test environment.

chipweinberger, 2024-05-16:
> Q: you wrote this new key purpose block (4) intentionally?

No. I have no idea why it is being written. 

> Are these errors common for your hardware design?

I've never seen them before, AFAIK.

> Any specific purpose for this partition? 

I used it during development, and never got rid of it. I think it is okay to keep?

chipweinberger, 2024-06-07:
I have a video showing how to reproduce strange behavior with GPIO 0.

* Note: the USB-A port is plugged into my mac using a "illegal" USB-A to USB-A cable for USB-Serial-JTAG

https://github.com/espressif/esp-idf/assets/1863934/473971eb-8bdd-4bf6-a92a-66401db40fbe

@igrr @mahavirj 

note: My schematic is above.

chipweinberger, 2024-06-27:
@mahavirj 

hoping you guys have time to investigate it further. Bricking devices is very severe.

mahavirj, 2024-06-28:
@chipweinberger 

IMO, getting the hardware design reviewed first would be helpful here. Would it be possible for you to contact our support email (sales at espressif) and send across the design files?

chipweinberger, 2024-06-28:
pcb is here: https://github.com/espressif/esp-idf/issues/13783#issuecomment-2106045922