espressif/esp-idf #13784: hid_host_example.c does not collect data from generic devices (ie. wired PS4 controller)

BotSpace, 2024-05-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My goal is to have my ESP32-S2 act as a USB host for my wired PS4 controller. I'm using windows 10, VS code, and ESP-IDF v5.2.1. I'm using UART for flashing and monitoring, leaving the USB peripheral free to act as USB Host. I have created a project that is an unmodified copy of [hid_host_example.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c).

After flashing and monitoring, I plug in a wired mouse to the USB peripheral, shake it a bit, and disconnect it:
```
Waiting for the device to reconnect..
I (128226) example: HID Device, protocol 'KEYBOARD' CONNECTED
I (128226) example: HID Device, protocol 'MOUSE' CONNECTED
I (128226) example: HID Device, protocol 'NONE' CONNECTED

Mouse
X: -00006       Y: 000322       | | |
Generic
E (240326) USBH: Device 1 gone
I (240326) example: HID Device, protocol 'KEYBOARD' DISCONNECTED
I (240326) example: HID Device, protocol 'MOUSE' DISCONNECTED
I (240336) example: HID Device, protocol 'NONE' DISCONNECTED
```
The mouse connection works in that the mouse state is printed to the terminal in real time as expected. There seems to be some erroneous info messages but otherwise it works.

Then I connect the PS4 controller and do my hardest to provide input.
```
I (385816) example: HID Device, protocol 'NONE' CONNECTED
E (398406) USBH: Device 1 gone
I (398406) example: HID Device, protocol 'NONE' DISCONNECTED
E (401006) HUB: Root port reset failed
```
The S2 sees the HID device but despite my efforts, no data is read to the terminal. When I unplug the PS4 controller, that is reported.

The PS4 controller is known good, when I plug it into my PC it shows up in device manager and I can see inputs using the windows game controller utility.

I have spent the better part of the last 2 days reading the [Device Class Definition for Human Interface Devices (HID)](https://www.usb.org/sites/default/files/hid1_11.pdf) but am completely stumped. Any advice is appriciated!

peter-marcisovsky, 2024-05-13:
Hi @BotSpace 
Could you please provide us a USB descriptor of the PS4 controller you are using?
Thank you.

BotSpace, 2024-05-13:
My controller is a PS4 SADES PS-403 (C200?). I don't have too much experience with HID but this is what I found by connecting the controller to my laptop and sniffing that connection with Wireshark. I based my results from [this information](https://eleccelerator.com/wiki/index.php?title=DualShock_4). The results are mostly similar, except for the bcdHID 1.17 field and both bEndpointAddresses 

Device Descriptor
```
0x12,        // bLength
0x01,        // bDescriptorType (Device)
0x00, 0x02,  // bcdUSB 2.00
0x00,        // bDeviceClass (Use class information in the Interface Descriptors)
0x00,        // bDeviceSubClass 
0x00,        // bDeviceProtocol 
0x40,        // bMaxPacketSize0 64
0x4C, 0x05,  // idVendor 0x054C
0xC4, 0x05,  // idProduct 0x05C4
0x00, 0x01,  // bcdDevice 1.00
0x01,        // iManufacturer (String Index)
0x02,        // iProduct (String Index)
0x00,        // iSerialNumber (String Index)
0x01,        // bNumConfigurations 1
```

Configuration Descriptor
```
0x09,        // bLength
0x02,        // bDescriptorType (Configuration)
0x29, 0x00,  // wTotalLength 41
0x01,        // bNumInterfaces 1
0x01,        // bConfigurationValue
0x00,        // iConfiguration (String Index)
0xC0,        // bmAttributes Self Powered
0xFA,        // bMaxPower 500mA

0x09,        // bLength
0x04,        // bDescriptorType (Interface)
0x00,        // bInterfaceNumber 0
0x00,        // bAlternateSetting
0x02,        // bNumEndpoints 2
0x03,        // bInterfaceClass
0x00,        // bInterfaceSubClass
0x00,        // bInterfaceProtocol
0x00,        // iInterface (String Index)

0x09,        // bLength
0x21,        // bDescriptorType (HID)
0x00, 0x01,  // bcdHID 1.17
0x00,        // bCountryCode
0x01,        // bNumDescriptors
0x22,        // bDescriptorType[0] (HID)
0xD3, 0x01,  // wDescriptorLength[0] 467

0x07,        // bLength
0x05,        // bDescriptorType (Endpoint)
0x83,        // bEndpointAddress (IN/D2H)
0x03,        // bmAttributes (Interrupt)
0x40, 0x00,  // wMaxPacketSize 64
0x05,        // bInterval 5 (unit depends on device speed)

0x07,        // bLength
0x05,        // bDescriptorType (Endpoint)
0x02,        // bEndpointAddress (OUT/H2D)
0x03,        // bmAttributes (Interrupt)
0x40, 0x00,  // wMaxPacketSize 64
0x05,        // bInterval 5 (unit depends on device speed)
```

Marcel-1, 2024-06-07:
This is exactly what I am trying todo as well. Have you made any progress on this that you would share?

BotSpace, 2024-06-08:
Honestly I have been pretty frustrated trying to hook a wired gamepad up to an ESP32-S2. Espressif advertises the USB-OTG Host capabilities of the S2 which has been out for 5 years, but if you are trying to host for any device that is not a keyboard or mouse you seem pretty screwed to find a working example.

I gave up on ESP-IDF and turned to Arduino, where there you are still hard pressed to find working examples. Part of the problem is that I feel like there are four or five different forks of TinyUSB, each with their own quirks and spotty examples. I tried one or two but had little success with those either.

This is compounded by the fact that I believe tinyUSB is still broken on the esp32-S3 in some ways.

I have had my only success with [this project by badjeff](https://github.com/badjeff/ESP32-USB-Host-HID-Bridge) (we love badjeff) which he says is based off of esp-idf, but it what way I do not understand. There is very little in common between his code and the modern esp-idf example. And even then Jeff's work only works with some of my PS controllers, but not all of them. I have tried to add more with little success.

I suppose I shouldn't be too quick to hold it against espressif. As part of this project I have spent many hours trying to learn and understand USB and I have found it to be an order of magnitude more complex than the serial/i2s/spi/can protocols I am more familiar with. It seems like what makes controllers different is not just the format of the data they report, but even the sequence of messages needed to get them to spit out data.

I have more respect for modern operating systems that can handle so many gamepads from different manufactures.

So yea in the end I gave up on ESP-IDF and TinyUSB after toying with both for many hours, and am basing my MVP off Jeff's work.

Perhaps someone will come along and tell me about what I missed with ESP-IDF or TinyUSB. I would welcome that in that case, but for now I am somewhat miffed.

Marcel-1, 2024-06-09:
Thank you for the feedback and your experience. I'm currently going through the same exercise as you did to understand and modify ESP-IDF example. I have been experimenting with ChatGPT to help me write some code for this but so far haven't been successful. If I do get this to work I will help you know.

alkurop, 2024-06-25:
It may be possible to connect with bluetooth

BotSpace, 2024-06-25:
I'm sure it would be possible, but a wired connection is a design requirement for my use case. And even if it wasn't, I would still think Espressif should have good wired gamepad software support for their MCUs with HID host capability.

Marcel-1, 2024-06-25:
Hey @alkurop I was able to connect a DualShock 4 using Bluetooth with the Bluepad32 library but I assume Bluetooth is going to add some latency which is why I started to explore working on a wired connection. Bluepad32 is really good and easy to setup but I still prefer a wired connection.

SaintSampo, 2025-10-16:
Hi, thank you for looking into this. I do still have the project I need this for on the back burner (awhile ago I designed a PCB that turns a gamepad into an ELRS transmitter. I don't consider BLE reliable enough in the field to fly a drone).

I used [this library](https://github.com/esp32beans/ESP32_USB_Host_HID) to create a minimum viable product with a ps4 controller, it works and everything!

My original goal was to support all gamepad types. I have done more research since then and for Xbox controllers, I have little hope of finding time to write a DirectX driver. I have slightly more hope of adding Nintendo support, it at least uses HID. I appreciate Sony for keeping the PS controller protocols relatively simple.

For the espressif example, I will give you suggestions a try when I get a chance. Assuming that any device that is not a mouse or keyboard does not use the BOOT protocol, wouldn't it make sense for generic devices to be opened and started by default? I might look into putting in a PR for this.

TLDR, this is still relevant, thank you for the info, Ill look into this again soon.