espressif/esp-idf #13787: The freemodbus stack does not support bulk payload of 150 registers for write request with function code(16) 0x10

BestBot7, 2024-05-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a slave device which supports writing 150 registers at once which is around 300 bytes. I have framed a modbus request which is greater than 300 bytes including slave ID, function code, register start address, number of registers, number of bytes, data and CRC and sent it to the modbus slave device using standard modbus master template code. In standard documentations it is mentioned that the modbus can support writing multiple registers of 123 registers at once. Is there any way to that can support the bulk and huge payload of writing multiple registers?

alisitsyn, 2024-05-13:
The maximum buffer for the frame is 256 bytes = absolute maximum is 128 registers but the max number of holding registers to read/write is not 128. From the spec:

"The quantity of registers to be read, combined with all the other fields in the expected response, must not exceed the allowable length of Modbus messages: 256 bytes.". 
If we take into account the standard fields of Modbus the maximum number of register to read with FC=0x03 drops to [125 registers](https://github.com/espressif/esp-modbus/blob/master/freemodbus/modbus/functions/mbfuncholding_m.c#L56). 

The limitations is hard coded in the stack and for now it is not planned to extend the limit of registers to support non standard slaves.

alisitsyn, 2024-09-19:
This information will be described in the documentation on how to override the standard limitations in such slaves. This is pretty untypical case and because of this this will not be implemented in the stack for now to satisfy the Modbus standard. The issue will be closed for now. Feel free to reopen if you have comments.

closed 2024-09-19 by alisitsyn