espressif/esp-idf #13788: esp32s3, restarting cdc software serial doesn't work

greenaddress, 2024-05-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4 and 5.2 release branch

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3 tdisplay s3

### Power Supply used.

USB

### What is the expected behavior?

I expect to be able to call pseudo code that looks like
```
tinyusb_driver_install
tusb_cdc_acm_init

// use cdc serial

tusb_cdc_acm_deinit
tinyusb_driver_uninstall

// sleep a bit or do something else usb mass storage
tinyusb_driver_install
tusb_cdc_acm_init
```

and I would expect the cdc software serial working again but it doesn't show up with lsusb after a cdc and driver deinit and uninstall 

I get no errors, everything is ESP_OK but the device doesn't show up - an esp_restart is required for things to show up again.

using
  espressif/tinyusb: "^0.15.0~9"                                                 
  espressif/esp_tinyusb: "^1.4.4"   

and tried with esp-idf tags v5.1.3, v5.1.4 and 5.2 release branch

### What is the actual behavior?

I would expect to be able to cycle the usb software cdc on/off/on 

### Steps to reproduce.

Something like https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_serial_device could be modified to also deinit cdc device software and uninstall tinyusb

### Debug Logs.

```plain
logs of first init vs next init don't seem to differ.
```

### More Information.

basically same issue (1) as https://www.reddit.com/r/esp32/comments/1c50bn1/esp32s3_tinyusb_issue/ which to me also doesn't seem resolved (same as OP)

peter-marcisovsky, 2024-05-14:
Hi @greenaddress
Thanks for opening the issue, as you pointed out, I was already trying to solve it on Reddit. 
I will try to reproduce it and let you know.

greenaddress, 2024-05-20:
@peter-marcisovsky thanks for getting back

Any news on reproducing? would it be of help if we provided some minimal/hello world for both 5.2 and 5.1.4 to show the issue?

roma-jam, 2024-05-21:
Hi @greenaddress ,

we have all necessary information and we are able to reproduce the behavior, thanks a lot.

Also, to re-start tinyusb driver we need to shutdown it properly (this feature is not implemented in upstream as well: https://github.com/hathach/tinyusb/blob/a1abbd70cbc00878a34f0b7db24805fc78b6b8d8/src/tusb.h#L142)

Based on the requirements, this could be done in several different ways, so we will return back with some solution asap. 

Meanwhile, do you have specific requirements and do you plan to change the USB device class during the tinyusb driver uninstall-install routine?

greenaddress, 2024-05-22:
@roma-jam I plan to use usb storage as a host (msc) when not using cdc as device.

msc works but i can't bring back cdc (even if i never used msc)

greenaddress, 2024-05-31:
@roma-jam any news or suggestion/work around?

roma-jam, 2024-06-10:
Hi @greenaddress ,

Sorry for such a long delay with a reply. 

Currently, there is not so much to do, because to use tinyusb driver again (which means to restart the device but with another configuration) we need to: 
- Disable all enabled class drivers (if there is one)
- Disable USB interrupt
- Deinit underlying hardware layer and flush all internal variables in tinyusb driver to be able to start again.

So, when the according API are not available, the only one option is to fulfill a whole reset of the mcu unit. 

To make it available without the whole reset, changes should be done in: 
- esp_tinyusb component (https://github.com/espressif/esp-usb/tree/master/device/esp_tinyusb) ;
- tinyusb component (https://github.com/espressif/tinyusb);