espressif/esp-idf #13795: cannot open linker script file esp32.peripherals.ld: No such file or directory

summerfind, 2024-05-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all.
I installed vscode and extension: esp-idf/python; and run example blink, got error:
`[105/106] Generating binary image from built executable
esptool.py v4.7.0
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated C:/Users/Xinzhou/esp/blink/build/bootloader/bootloader.bin
[106/106] cmd.exe /C "cd /D C:\Users\Xinzhou\esp\blink\build\bootloader\esp-idf\esptool_py && c:\espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:/espressif/v5.2.1/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/Xinzhou/esp/blink/build/bootloader/bootloader.bin"
Bootloader binary size 0x6860 bytes. 0x7a0 bytes (7%) free.
[921/923] Linking CXX executable blink.elf
FAILED: blink.elf
cmd.exe /C "cd . && C:\espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=C:/Users/Xinzhou/esp/blink/build/blink.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\blink.elf.rsp -o blink.elf  && cd ."
C:/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot open linker script file esp32.peripherals.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.`

please let me know what info. need.
Thanks
Adam

dobairoland, 2024-05-15:
Can you check `C:\Users\Xinzhou\esp\blink\build\build.ninja` and find the line starting with `LINK_PATH`?

Mine is the following building with pure ESP-IDF:

```
LINK_PATH = -LC:/Users/Roland/esp/esp-idf/components/soc/esp32/ld   -LC:/Users/Roland/esp/esp-idf/components/esp_rom/esp32/ld   -LC:/Users/Roland/esp/esp-idf/examples/get-started/blink/build/esp-idf/esp_system/ld   -LC:/Users/Roland/esp/esp-idf/components/esp_phy/lib/esp32   -LC:/Users/Roland/esp/esp-idf/components/esp_wifi/lib/esp32
```

What is the content of yours?

I'm not sure where your error originates. It would help if you could try first the build from pure ESP-IDF terminal (ESP-IDF CMD) without Vscode.

summerfind, 2024-05-16:
Thank you [dobairoland](https://github.com/dobairoland).
I found `build.ninja` and the line: ` LINK_PATH = -LC:/Users/Xinzhou/esp/v5.2.1/esp-idf/components/esp_rom/esp32/ld   -LC:/Users/Xinzhou/esp/v5.2.1/esp-idf/components/soc/esp32/ld   -LC:/Users/Xinzhou/esp/v5.2.1/esp-idf/components/bootloader/subproject/main/ld/esp32
`

I did build blink in CMD, this time better than before which also failed.
this time not got failed, but:
`-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Xinzhou/esp/blink/build/bootloader
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\Xinzhou\esp\blink\build\log\idf_py_stderr_output_4336 and C:\Users\Xinzhou\esp\blink\build\log\idf_py_stdout_output_4336`

see if you need the full error message.

fhrbata, 2024-05-16:
Hello @summerfind ,
it would be helpful to know which subcommand failed and the reason for its failure. Could you please provide the logs or the exact error message of the failed command? Thank you very much.

summerfind, 2024-05-16:
> Hello @summerfind , it would be helpful to know which subcommand failed and the reason for its failure. Could you please provide the logs or the exact error message of the failed command? Thank you very much.

sure.
I rebuild it by: idf.py --no-ccache build actually got : `[899/901] Linking CXX executable hello_world.elfFAILED: hello_world.elf` again. 

`C:\Users\Xinzhou\esp\blink>idf.py --no-ccache build
Executing action: all (aliases: build)
Running cmake in directory C:\Users\Xinzhou\esp\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.1_py3.12_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 C:\Users\Xinzhou\esp\blink"...
-- git describe returned 'fatal: bad revision 'HEAD''
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] espressif/led_strip (2.5.3) from file:///C:/Espressif/registry
[2/2] idf (5.1.4)
-- Project sdkconfig file C:/Users/Xinzhou/esp/blink/sdkconfig
Compiler supported targets: xtensa-esp32-elf

-- App "blink" version: 1
-- Adding linker script C:/Users/Xinzhou/esp/blink/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.4/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__led_strip esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.1.4/components/app_trace C:/Espressif/frameworks/esp-idf-v5.1.4/components/app_update C:/Espressif/frameworks/esp-idf-v5.1.4/components/bootloader C:/Espressif/frameworks/esp-idf-v5.1.4/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt C:/Espressif/frameworks/esp-idf-v5.1.4/components/cmock C:/Espressif/frameworks/esp-idf-v5.1.4/components/console C:/Espressif/frameworks/esp-idf-v5.1.4/components/cxx C:/Espressif/frameworks/esp-idf-v5.1.4/components/driver C:/Espressif/frameworks/esp-idf-v5.1.4/components/efuse C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp-tls C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_adc C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_coex C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_common C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_eth C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_event C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_hid C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_http_client C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_http_server C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_https_server C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_lcd C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_local_ctrl C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_mm C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_netif C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_netif_stack C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_partition C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_phy C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_pm C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_psram C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_ringbuf C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_system C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_timer C:/Espressif/frameworks/esp-idf-v5.1.4/components/esp_wifi C:/Espressif/frameworks/esp-idf-v5.1.4/components/espcoredump C:/Users/Xinzhou/esp/blink/managed_components/espressif__led_strip C:/Espressif/frameworks/esp-idf-v5.1.4/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.1.4/components/fatfs C:/Espressif/frameworks/esp-idf-v5.1.4/components/freertos C:/Espressif/frameworks/esp-idf-v5.1.4/components/hal C:/Es [... cut ...]
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.peripherals.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\Xinzhou\esp\blink\build\log\idf_py_stderr_output_4964 and C:\Users\Xinzhou\esp\blink\build\log\idf_py_stdout_output_4964`

fhrbata, 2024-05-16:
Hello @summerfind,
there was a very similar issue reported here https://github.com/espressif/esp-idf/issues/13112 and it seems that running install and export script from powershell fixed the problem. Please see https://github.com/espressif/esp-idf/issues/13112#issuecomment-1929169554. Maybe this can help in your case too(cc @jakub-kocka ), but the root cause of this problem is still not clear to me.  Perhaps there is something wrong with your installation and following 
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html might help.
Thank you

summerfind, 2024-05-17:
Thank you [fhrbata](https://github.com/fhrbata)
#13112  solved the headache.

fhrbata, 2024-05-17:
Hello @summerfind ,

`v4.2` is EOL and not supported. You can see the list of supported versions and targets here https://github.com/espressif/esp-idf?tab=readme-ov-file#esp-idf-release-and-soc-compatibility. I would suggest to use the windows installer https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html#esp-idf-tools-installer to install version of  esp-idf environment you require/need. HTH and thank you

summerfind, 2024-05-17:
> is EOL and not supported
Thanks.
I know that, question is:

1. I used Windows installer,  installed all Python/Vscode/Espressif and then got build error of 'cc1 xxxx' when 'idf.py build' from Vscode terminal; I used 'idf.py --no-ccache build ' eliminated 'cc1 error', but got 'Failed: hello_world.elf';

2. ref. to [Get Started] : https://docs.espressif.com/projects/esp-idf/zh_CN/release-v4.2/esp32/get-started/index.html 
I used (I would say it is wrong use?): 'git clone -b release/v4.2 --recursive https://github.com/espressif/esp-idf.git'
in my '~/esp', got lot of thing downloaded and installed, and I can 'idf.py build' from my CMD and succeed and flash to ESP32;

3. when I try to use ESP32S3, got Error: Invalid value for '{esp32|esp32s2}': 'esp32c3' is not one of 'esp32', 'esp32s2'. I tried to update by: 'git clone -b release/v4.4 --recursive https://github.com/espressif/esp-idf.git' or even v5.2 all failed by strange errors, don't know why?

fhrbata, 2024-05-18:
Hello @summerfind,

I'm glad you figured it out! I found this page https://idf.espressif.com/ that I hadn't seen before. It offers a nice starting point based on the chip you plan to use and directs you to the relevant documentation and installation steps. Maybe it can be handy.

I rarely use Windows myself, so I'm not familiar with all the potential issues. However, generally, fetching or checking out the required branch, followed by `git submodule update --init --recursive --checkout` and running the `install` and `export` scripts works for me in Windows `cmd.exe`.

Anyway, it's great to hear that it's working for you now!

Thank you

closed 2024-05-23 by adokitkat