espressif/esp-idf #13796: 芯片掉固件

WonderfulMoment, 2024-05-16:
你好，目前这边遇到个问题，设备烧写程序之后，设备发到现场前测试OK的
设备发到客户现场之后发现起不来了，把esp32的芯片拿回来分析发现log打印如下：
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

重新烧写程序又是正常的

这个现象不是所有芯片都出现，目前有5芯片出现这个情况

请问一下，我们的esp32-s3在什么情况下会丢固件

zztiger123, 2024-05-17:
您好，

请问您目前烧录的固件大概是怎样的应用？里面会有 flash 读写操作吗？方便读取这五片芯片和正常的芯片的固件相关信息给我们比对吗？指令如下：
esptool.py read_flash xxxxx   // 读取芯片固件，xxx 需要按您实际的 flash 大小填写 offset  
esptool.py flash_id                  // 读取 flash id
esptool.py read_flash_status --bytes 3    // 读取 flash 状态

pancankun123, 2024-05-17:
你好，我这边也遇到了一样的问题，丢固件的芯片的信息如下：
C:\Users\143206\Desktop\ESP TOOL\20240506>esptool.exe read_flash_status --bytes 3
esptool.py v4.6.2
Found 2 serial ports
Serial port COM56
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Traceback (most recent call last):
  File "esptool.py", line 37, in <module>
  File "esptool\__init__.py", line 1073, in _main
  File "esptool\__init__.py", line 868, in main
  File "esptool\cmds.py", line 1171, in read_flash_status
  File "esptool\loader.py", line 1359, in read_status
  File "esptool\loader.py", line 1296, in run_spiflash_command
  File "esptool\loader.py", line 712, in read_reg
  File "esptool\loader.py", line 419, in command
esptool.util.UnsupportedCommandError: This command (0xa) is not supported in Secure Download Mode
[19548] Failed to execute script 'esptool' due to unhandled exception!

C:\Users\143206\Desktop\ESP TOOL\20240506>esptool flash_id
esptool.py v4.6.2
Found 2 serial ports
Serial port COM56
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Traceback (most recent call last):
  File "esptool.py", line 37, in <module>
  File "esptool\__init__.py", line 1073, in _main
  File "esptool\__init__.py", line 868, in main
  File "esptool\cmds.py", line 1088, in flash_id
  File "esptool\loader.py", line 891, in flash_id
  File "esptool\loader.py", line 1296, in run_spiflash_command
  File "esptool\loader.py", line 712, in read_reg
  File "esptool\loader.py", line 419, in command
esptool.util.UnsupportedCommandError: This command (0xa) is not supported in Secure Download Mode
[12152] Failed to execute script 'esptool' due to unhandled exception!

C:\Users\143206\Desktop\ESP TOOL\20240506>esptool.exe read_flash
usage: esptool read_flash [-h] [--spi-connection SPI_CONNECTION] [--no-progress] address size filename
esptool read_flash: error: the following arguments are required: address, size, filename

正常芯片的信息如下：
C:\Users\143206\Desktop\ESP TOOL\20240506>esptool.exe read_flash
usage: esptool read_flash [-h] [--spi-connection SPI_CONNECTION] [--no-progress] address size filename
esptool read_flash: error: the following arguments are required: address, size, filename

C:\Users\143206\Desktop\ESP TOOL\20240506>esptool flash_id
esptool.py v4.6.2
Found 2 serial ports
Serial port COM3
Connecting......................................
COM3 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port COM22
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Traceback (most recent call last):
  File "esptool.py", line 37, in <module>
  File "esptool\__init__.py", line 1073, in _main
  File "esptool\__init__.py", line 868, in main
  File "esptool\cmds.py", line 1088, in flash_id
  File "esptool\loader.py", line 891, in flash_id
  File "esptool\loader.py", line 1296, in run_spiflash_command
  File "esptool\loader.py", line 712, in read_reg
  File "esptool\loader.py", line 419, in command
esptool.util.UnsupportedCommandError: This command (0xa) is not supported in Secure Download Mode
[23116] Failed to execute script 'esptool' due to unhandled exception!

C:\Users\143206\Desktop\ESP TOOL\20240506>esptool.exe read_flash_status --bytes 3
esptool.py v4.6.2
Found 2 serial ports
Serial port COM3
Connecting......................................
COM3 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port COM22
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Traceback (most recent call last):
  File "esptool.py", line 37, in <module>
  File "esptool\__init__.py", line 1073, in _main
  File "esptool\__init__.py", line 868, in main
  File "esptool\cmds.py", line 1171, in read_flash_status
  File "esptool\loader.py", line 1359, in read_status
  File "esptool\loader.py", line 1296, in run_spiflash_command
  File "esptool\loader.py", line 712, in read_reg
  File "esptool\loader.py", line 419, in command
esptool.util.UnsupportedCommandError: This command (0xa) is not supported in Secure Download Mode
[25172] Failed to execute script 'esptool' due to unhandled exception!

我这里烧录的固件都是烧录密钥后烧断保险丝进行加密的，但有一部分芯片出现了丢固件的问题

pancankun123, 2024-05-17:
另外，我这里还有一部分芯片出现了无法烧断保险丝的情况，烧录情况如下：
C:\Users\143206\Desktop\ESP TOOL\20240506>espefuse.exe --port COM59 --chip esp32s3 --do-not-confirm burn_bit BLOCK0 4
espefuse.py v4.6.2
Connecting...

=== Run "burn_bit" command ===
bit_number:   [191]........................................................[0]
BLOCK0    : 0x000000000000000000000000000000000000000000000010
BLOCK0          (                ) [0 ] regs_to_write: 00000010 00000000 00000000 00000000 00000000 00000000

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[00] BLOCK0               is not empty
        (written ): 0x000000000000010200400000041cd1f500180c0100800116
        (to write): 0x000000000000000000000000000000000000000000000010
        All wr_data bits are set in the written block, continue with EMPTY_BLOCK.
Nothing to burn, see messages above.
这种问题应该如何解决

pancankun123, 2024-05-17:
另外，我这里还有一部分芯片出现了无法烧断保险丝的情况，烧录情况如下：
[image]

这种问题应该如何解决

WonderfulMoment, 2024-05-17:
> 您好，
> 
> 请问您目前烧录的固件大概是怎样的应用？里面会有 flash 读写操作吗？方便读取这五片芯片和正常的芯片的固件相关信息给我们比对吗？指令如下： esptool.py read_flash xxxxx // 读取芯片固件，xxx 需要按您实际的 flash 大小填写 offset esptool.py flash_id // 读取 flash id esptool.py read_flash_status --bytes 3 // 读取 flash 状态

谢谢回复，esp32主要是作为网卡使用的，就是通用的wifi功能+一个外接4G模组，没有特殊的flash读写操作，

1. flash id和 flash status这两条能正常读取信息，read_flash命令出错（正常芯片和问题芯片都是同样的错误，尝试不同的波特率和读取长度降低，错误是一样的），输出内容如下：

esptool.py -p /dev/ttyACM0 -b 230400 read_flash 0 0x20000 flash_contents.bin

esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 48:27:e2:2d:7f:f0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.

A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0x7dc bytes

2. 读取的flash id  和 flash status 的内容如下：

1）, 其中一个问题芯片的flash id和flash status的内容
$ esptool.py -p /dev/ttyACM0 flash_id

esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 80:65:99:d7:81:b8
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4017
Detected flash size: 8MB
Flash type set in eFuse: quad (4 data lines)
Hard resetting via RTS pin...

$ esptool.py -p /dev/ttyACM0 read_flash_status --bytes 3

esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 80:65:99:d7:81:b8
Uploading stub...
Running stub...
Stub running...
Status value: 0x200200
Hard resetting via RTS pin...

2）, 正常使用的卡读取的flash id和flash status 如下：
 $ esptool.py -p /dev/ttyACM0 flash_id

esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 48:27:e2:2d:7f:f0
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4017
Detected flash size: 8MB
Flash type set in eFuse: quad (4 data lines)
Hard resetting via RTS pin...

$ esptool.py -p /dev/ttyACM0 read_flash_status --bytes 3

esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 48:27:e2:2d:7f:f0
Uploading stub...
Running stub...
Stub running...
Status value: 0x200000
Hard resetting via RTS pin...

zztiger123, 2024-05-17:
> 谢谢回复，esp32主要是作为网卡使用的，就是通用的wifi功能+一个外接4G模组，没有特殊的flash读写操作，
>
> 1. flash id和 flash status这两条能正常读取信息，read_flash命令出错（正常芯片和问题芯片都是同样的错误，尝试不同的波特率和读取长度降低，错误是一样的），输出内容如下：
>
> esptool.py -p /dev/ttyACM0 -b 230400 [...]

在 read flash 指令里加上 --no-stub 试试。比如 esptool.py --no-stub read_flash 0 0x100000 flash_dump.bin

然后一直是这五个 S3 每过一段时间，固件就会稳定消失吗？有没有一定的规律？

WonderfulMoment, 2024-05-17:
> 在 read flash 指令里加上 --no-stub 试试。比如 esptool.py --no-stub read_flash 0 0x100000 flash_dump.bin
>
> 然后一直是这五个 S3 每过一段时间，固件就会稳定消失吗？有没有一定的规律？

没有专门正对这几个卡做跟踪测试，其中有一个卡重新烧录程序是能正常运行的，其他几个卡没进行恢复，为了保留环境给到你们分析，家里测试的样片没见到丢程序的

生产的情况是，第一批做的小批量验证，上线的200块，运行了两个月没看到有异常的；
前几天重新生产了一千台，部分上电之后就发现有这5块程序掉了，还有大部分没有上电，所以还不知道会不会还有别的掉程序的

附件是读到的flash的数据，一个是有问题的flash内容一个是家里用的没有问题的读出来的内容，应为github不支持bin文件，所以改后缀为txt了，请查阅[
[flash_contents.txt](https://github.com/espressif/esp-idf/files/15346753/flash_contents.txt)
[问题芯片-flash_contents.txt](https://github.com/espressif/esp-idf/files/15346760/-flash_contents.txt)
](url)

pancankun123, 2024-05-17:
> > 另外，我这里还有一部分芯片出现了无法烧断保险丝的情况，烧录情况如下：![企业微信截图_17159164268256](https://private-user-images.githubusercontent.com/151031536/331449890-2766ec1c-ce1a-4c33-858c-b1d0b38955d2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU5Mjk4NzAsIm5iZiI6MTcxNTkyOTU3MCwicGF0aCI6Ii8xNTEwMzE1MzYvMzMxNDQ5ODkwLTI3NjZlYzFjLWNlMWEtNGMzMy04NThjLWIxZDBiMzg5NTVkMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxN1QwNzA2MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYTMzZmVjNmMxOGU5YWZkYmIzZDY4ODRjYjdiN2I5MjFmNWM3MzRjZTZmM2FjNGI0N2Y2OWUyMWVmYzdkYjFjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9._y34hp1f2PudPDQQmNT4836Hmj3lzz9kZaLSiQ_GqQc)
> > 这种问题应该如何解决
> 
> 这个不算是个问题，从 log 可以看到你要写入的 efuse 位已经都是 1 了，所以默认不做操作。

pancankun123, 2024-05-18:
> 这个不算是个问题，从 log 可以看到你要写入的 efuse 位已经都是 1 了，所以默认不做操作。
但是即使efuse 位已经都是 1了，还是无法正确烧录文件，另外，我这边烧录固件的时候，已经在烧录指令里面加 指令里加上 --no-stub和--force，一部分固机烧录重新上电后还是出现了丢固件的问题

pancankun123, 2024-05-18:
但是即使efuse 位已经都是 1了，还是无法正确烧录文件，另外，我这边烧录固件的时候，已经在烧录指令里面加 指令里加上 --no-stub和--force，一部分固机烧录重新上电后还是出现了丢固件的问题

> 这个不算是个问题，从 log 可以看到你要写入的 efuse 位已经都是 1 了，所以默认不做操作。

但是即使efuse 位已经都是 1了，还是无法正确烧录文件，另外，我这边烧录固件的时候，已经在烧录指令里面加 指令里加上 --no-stub和--force，一部分固机烧录重新上电后还是出现了丢固件的问题

zztiger123, 2024-05-20:
> 没有专门正对这几个卡做跟踪测试，其中有一个卡重新烧录程序是能正常运行的，其他几个卡没进行恢复，为了保留环境给到你们分析，家里测试的样片没见到丢程序的
>
> 生产的情况是，第一批做的小批量验证，上线的200块，运行了两个月没看到有异常的； 前几天重新生产了一千台，部分上电之后就发现有这5块程序掉了，还有大部分没有上电，所以还不知道会不会还有别的掉程序的 [...]

分析了下您提供的 flash 文件，确实异常的 S3 flash 里面为空了。

您是用的什么方式烧录的固件？如果是 flash download tool，在以下 log 文件夹里可以查询烧录了哪些 mac 地址对应的芯片，可以先看看这些芯片是否真的有成功烧录过固件：
[image]

pancankun123, 2024-05-20:
> espefuse.exe summary

一般情况下，我的设备是需要烧录密钥和熔断保险丝的，espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_key BLOCK_KEY0 CYH00124012202511.bin XTS_AES_128_KEY
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_efuse DIS_DOWNLOAD_MANUAL_ENCRYPT 1
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_efuse DIS_DOWNLOAD_DCACHE 1
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_efuse DIS_DOWNLOAD_ICACHE 1
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_efuse DIS_USB_JTAG 1
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_efuse DIS_DIRECT_BOOT 1
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_efuse HARD_DIS_JTAG 1
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_bit BLOCK0 1
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_bit BLOCK0 2
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_efuse SPI_BOOT_CRYPT_CNT 7
espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_bit BLOCK0 4
执行完以上步骤后，芯片是无法通过efuse summary 查询到里面信息了的，但现在 SPI_BOOT_CRYPT_CNT 的值不可以写入，芯片的信息也可以查询，那就是没有烧断保险丝加密成功，那我应该可以重复上面的步骤，继续烧断保险丝，但是一直显示SPI_BOOT_CRYPT_CNT 的值已经写入1了，不可更改，这个时候如果我强行烧录芯片的固件，烧录后芯片可以启动，但我需要读取到的信息是无法读取的。
另外，我这边也存在另一个用户的问题，我这边的问题和另一个用户的区别就是我这边需要先烧录密钥和保险丝，我这边的问题是烧录密钥和保险丝都是没有问题的通过 espefuse.exe summary 也无法查看芯片的信息，芯片已经加密成功，但是这个时候烧录加密后的固件，就会出现和另一个用户一样的问题。我这边的设备是最开始已经烧录成功了，也正确的读取到了信息，放置一段时间后出现了丢固件的问题，这是一部分的芯片，另一部分芯片是烧录完成后就出现丢固件问题

zztiger123, 2024-05-20:
> 一般情况下，我的设备是需要烧录密钥和熔断保险丝的，espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm burn_key BLOCK_KEY0 CYH00124012202511.bin XTS_AES_128_KEY espefuse.exe --port COM60 --chip esp32s3 --do-not-confirm [...]

这几个损坏的板子，此时下载固件和芯片本地的运行 log 分别是怎样的？

WonderfulMoment, 2024-05-20:
> 分析了下您提供的 flash 文件，确实异常的 S3 flash 里面为空了。
>
> 您是用的什么方式烧录的固件？如果是 flash download tool，在以下 log 文件夹里可以查询烧录了哪些 mac 地址对应的芯片，可以先看看这些芯片是否真的有成功烧录过固件： [...]

工厂那边说，他们电脑坏了，所以部分升级的日志没有找到，还有其它方式可以确认是否烧写过固件么？

zztiger123, 2024-05-20:
> 工厂那边说，他们电脑坏了，所以部分升级的日志没有找到，还有其它方式可以确认是否烧写过固件么？

目前应该没有别的方法了，因为 flash 里的记录是可以被全部擦除的。不过您重新烧录固件后恢复正常的那片 S3，目前是否继续复现问题？

WonderfulMoment, 2024-05-22:
> 目前应该没有别的方法了，因为 flash 里的记录是可以被全部擦除的。不过您重新烧录固件后恢复正常的那片 S3，目前是否继续复现问题？

你好，目前有一块料是客户现场上用了一段时间才出问题的，所以这块料肯定是正常烧录过的，log打印也是 invalid header: 0xffffffff，但是这颗料出问题之后，usb-jtag口也无法识别了，目前只能看到串口打印，但是用打印的串口进行flash内容读取没有成功，请问有没有办法从串口读取flash内容？

WonderfulMoment, 2024-05-23:
> 你好，目前有一块料是客户现场上用了一段时间才出问题的，所以这块料肯定是正常烧录过的，log打印也是 invalid header: 0xffffffff，但是这颗料出问题之后，usb-jtag口也无法识别了，目前只能看到串口打印，但是用打印的串口进行flash内容读取没有成功，请问有没有办法从串口读取flash内容？

=================================

项目上运行一段时间之后才出问题的这科料的状态和flash也读出来了，内容如下及附件，麻烦看一下是什么问题，谢谢
[flash_dump.txt](https://github.com/espressif/esp-idf/files/15410334/flash_dump.txt)
：
PS C:\Espressif\frameworks\esp-idf-v4.4.5> esptool.py -p COM24 flash_id
esptool.py v3.3.4-dev
Serial port COM24
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 80:65:99:d7:7e:e8
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4017
Detected flash size: 8MB
Hard resetting via RTS pin...
PS C:\Espressif\frameworks\esp-idf-v4.4.5>
PS C:\Espressif\frameworks\esp-idf-v4.4.5> esptool.py -p COM24  read_flash_status --bytes 3
esptool.py v3.3.4-dev
Serial port COM24
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 80:65:99:d7:7e:e8
Stub is already running. No upload is necessary.
Status value: 0x200200
Hard resetting via RTS pin...

zztiger123, 2024-05-23:
> =================================
>
> 项目上运行一段时间之后才出问题的这科料的状态和flash也读出来了，内容如下及附件，麻烦看一下是什么问题，谢谢 [flash_dump.txt](https://github.com/espressif/esp-idf/files/15410334/flash_dump.txt) ： PS [...]

Hi @WonderfulMoment  这样的话，您方便发送邮件到 sales@espressif.com 吗？记得在邮件里提到当前这个 github 链接，他们会进一步接手并分析您这个问题。

WonderfulMoment, 2024-05-23:
> > > > > > > > > > 您好，
> > > > > > > > > > 请问您目前烧录的固件大概是怎样的应用？里面会有 flash 读写操作吗？方便读取这五片芯片和正常的芯片的固件相关信息给我们比对吗？指令如下： esptool.py read_flash xxxxx // 读取芯片固件，xxx 需要按您实际的 flash 大小填写 offset esptool.py flash_id // 读取 flash id esptool.py read_flash_status --bytes 3 // 读取 flash 状态
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > 谢谢回复，esp32主要是作为网卡使用的，就是通用的wifi功能+一个外接4G模组，没有特殊的flash读写操作，
> > > > > > > > > 
> > > > > > > > > 1. flash id和 flash status这两条能正常读取信息，read_flash命令出错（正常芯片和问题芯片都是同样的错误，尝试不同的波特率和读取长度降低，错误是一样的），输出内容如下：
> > > > > > > > > 
> > > > > > > > > esptool.py -p /dev/ttyACM0 -b 230400 read_flash 0 0x20000 flash_contents.bin
> > > > > > > > > esptool.py v4.7.0 Serial port /dev/ttyACM0 Connecting... Detecting chip type... ESP32-S3 Chip is ESP32-S3 (QFN56) (revision v0.2) Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3) Crystal is 40MHz MAC: 48:27:e2:2d:7f:f0 Uploading stub... Running stub... Stub running... Changing baud rate to 230400 Changed.
> > > > > > > > > A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0x7dc bytes
> > > > > > > > > 
> > > > > > > > > 2. 读取的flash id  和 flash status 的内容如下：
> > > > > > > > > 
> > > > > > > > > 1）, 其中一个问题芯片的flash id和flash status的内容 $ esptool.py -p /dev/ttyACM0 flash_id
> > > > > > > > > esptool.py v4.7.0 Serial port /dev/ttyACM0 Connecting... Detecting chip type... ESP32-S3 Chip is ESP32-S3 (QFN56) (revision v0.2) Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3) Crystal is 40MHz MAC: 80:65:99:d7:81:b8 Uploading stub... Running stub... Stub running... Manufacturer: c8 Device: 4017 Detected flash size: 8MB Flash type set in eFuse: quad (4 data lines) Hard resetting via RTS pin...
> > > > > > > > > $ esptool.py -p /dev/ttyACM0 read_flash_status --bytes 3
> > > > > > > > > esptool.py v4.7.0 Serial port /dev/ttyACM0 Connecting... Detecting chip type... ESP32-S3 Chip is ESP32-S3 (QFN56) (revision v0.2) Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3) Crystal is 40MHz MAC: 80:65:99:d7:81:b8 Uploading stub... Running stub... Stub running... Status value: 0x200200 Hard resetting via RTS pin...
> > > > > > > > > 2）, 正常使用的卡读取的flash id和flash status 如下： $ esptool.py -p /dev/ttyACM0 flash_id
> > > > > > > > > esptool.py v4.7.0 Serial port /dev/ttyACM0 Connecting... Detecting chip type... ESP32-S3 Chip is ESP32-S3 (QFN56) (revision v0.2) Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3) Crystal is 40MHz MAC: 48:27:e2:2d:7f:f0 Uploading stub... Running stub... Stub running... Manufacturer: c8 Device: 4017 Detected flash size: 8MB Flash type set in eFuse: quad (4 data lines) Hard resetting via RTS pin...
> > > > > > > > > $ esptool.py -p /dev/ttyACM0 read_flash_status --bytes 3
> > > > > > > > > esptool.py v4.7.0 Serial port /dev/ttyACM0 Connecting... Detecting chip type... ESP32-S3 Chip is ESP32-S3 (QFN56) (revision v0.2) Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3) Crystal is 40MHz MAC: 48:27:e2:2d:7f:f0 Uploading stub... Running stub... Stub running... Status value: 0x200000 Hard resetting via RTS pin...
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 在 read flash 指令里加上 --no-stub 试试。比如 esptool.py --no-stub read_flash 0 0x100000 flash_dump.bin
> > > > > > > > 然后一直是这五个 S3 每过一段时间，固件就会稳定消失吗？有没有一定的规律？
> > > > > > > 
> > > > > > > 
> > > > > > > 没有专门正对这几个卡做跟踪测试，其中有一个卡重新烧录程序是能正常运行的，其他几个卡没进行恢复，为了保留环境给到你们分析，家里测试的样片没见到丢程序的
> > > > > > > 生产的情况是，第一批做的小批量验证，上线的200块，运行了两个月没看到有异常的； 前几天重新生产了一千台，部分上电之后就发现有这5块程序掉了，还有大部分没有上电，所以还不知道会不会还有别的掉程序的
> > > > > > > 附件是读到的flash的数据，一个是有问题的flash内容一个是家里用的没有问题的读出来的内容，应为github不支持bin文件，所以改后缀为txt了，请查阅[ [flash_contents.txt](https://github.com/espressif/esp-idf/files/15346753/flash_contents.txt) [问题芯片-flash_contents.txt](https://github.com/espressif/esp-idf/files/15346760/-flash_contents.txt) ](url)
> > > > > > 
> > > > > > 
> > > > > > 分析了下您提供的 flash 文件，确实异常的 S3 flash 里面为空了。
> > > > > > 您是用的什么方式烧录的固件？如果是 flash download tool，在以下 log 文件夹里可以查询烧录了哪些 mac 地址对应的芯片，可以先看看这些芯片是否真的有成功烧录过固件： ![image](https://private-user-images.githubusercontent.com/40046810/331931283-c815f8c0-8819-42c8-8649-d77ee1d2cdbe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxOTI3MjksIm5iZiI6MTcxNjE5MjQyOSwicGF0aCI6Ii80MDA0NjgxMC8zMzE5MzEyODMtYzgxNWY4YzAtODgxOS00MmM4LTg2NDktZDc3ZWUxZDJjZGJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIwVDA4MDcwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkMGQ5NGU4NzYzNGYxNDAwMWY1ZDU5MDIzM2VjNDUxMTg0ZTFkMjA0MGRjZGEyMzMzNzA1ZTYyMWY4YjJiZGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TrJnGFMDQAnmEj1bPeiJmcoqmsJBVZbiQOOAzRqEHSY)
> > > > > 
> > > > > 
> > > > > 工厂那边说，他们电脑坏了，所以部分升级的日志没有找到，还有其它方式可以确认是否烧写过固件么？
> > > > 
> > > > 
> > > > 目前应该没有别的方法了，因为 flash 里的记录是可以被全部擦除的。不过您重新烧录固件后恢复正常的那片 S3，目前是否继续复现问题？
> > > 
> > > 
> > > 你好，目前有一块料是客户现场上用了一段时间才出问题的，所以这块料肯定是正常烧录过的，log打印也是 invalid header: 0xffffffff，但是这颗料出问题之后，usb-jtag口也无法识别了，目前只能看到串口打印，但是用打印的串口进行flash内容读取没有成功，请问有没有办法从串口读取flash内容？
> > 
> > 
> > =================================
> > 项目上运行一段时间之后才出问题的这科料的状态和flash也读出来了，内容如下及附件，麻烦看一下是什么问题，谢谢 [flash_dump.txt](https://github.com/espressif/esp-idf/files/15410334/flash_dump.txt) ： PS C:\Espressif\frameworks\esp-idf-v4.4.5> esptool.py -p COM24 flash_id esptool.py v3.3.4-dev Serial port COM24 Connecting.... Detecting chip type... ESP32-S3 Chip is ESP32-S3 (revision v0.2) Features:  [... cut ...] _status --bytes 3 esptool.py v3.3.4-dev Serial port COM24 Connecting... Detecting chip type... ESP32-S3 Chip is ESP32-S3 (revision v0.2) Features: WiFi, BLE Crystal is 40MHz MAC: 80:65:99:d7:7e:e8 Stub is already running. No upload is necessary. Status value: 0x200200 Hard resetting via RTS pin...
> 
> Hi @WonderfulMoment 这样的话，您方便发送邮件到 [sales@espressif.com](mailto:sales@espressif.com) 吗？记得在邮件里提到当前这个 github 链接，他们会进一步接手并分析您这个问题。

之前有同步发邮件的，就是邮件反应比较慢，所以才到这上面来问，现在量上来了，担心线上大批量出现类似问题，这样售后就特别麻烦，所以比较着急。

zztiger123, 2024-05-23:
> 之前有同步发邮件的，就是邮件反应比较慢，所以才到这上面来问，现在量上来了，担心线上大批量出现类似问题，这样售后就特别麻烦，所以比较着急。

方便重新再发一封新的邮件并附上此 github issue 链接吗？在 github 上我们需要保护客户的信息，所以不方便问您的邮箱来在邮件系统里查询。您未来的新邮件我已经通知商务要内部提高优先级尽快处理。

WonderfulMoment, 2024-05-23:
> 方便重新再发一封新的邮件并附上此 github issue 链接吗？在 github 上我们需要保护客户的信息，所以不方便问您的邮箱来在邮件系统里查询。您未来的新邮件我已经通知商务要内部提高优先级尽快处理。

已经发了，麻烦查收一下，有什么需要配合的，及时回复邮件，谢谢；

closed 2024-09-24 by Alvin1Zhang

Gui-ye, 2024-12-07:
> 您好，
> 
> 请问您目前烧录的固件大概是怎样的应用？里面有 flash 读写操作吗？方便读取这五片芯片和正常的芯片的固件相关信息给我们比对吗？指令如下： esptool.py read_flash xxxxx // 读取芯片硬件，xxx 需要按照您实际的 flash 大小填写 offset esptool.py flash_id // 读取 flash id esptool.py read_flash_status --bytes 3 // 读取 flash 状态

你好，我现在也遇到了这个情况，同一批的4片ESP32C6模组（因为仅用于测试所以只购买了4片）均出现该情况
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xfESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002949e
0x4002949e: uart_hal_get_txfifo_count in ROM
附带您要求的硬件信息
[芯片信息.txt](https://github.com/user-attachments/files/18048736/default.txt)
这是执行esptool.py read_flash 0x000000 0x800000 esp32err.bin 后产生的结果，由于GITHUB不支持bin文件因此修改了文件的后缀名。
[esp32err.txt](https://github.com/user-attachments/files/18048740/esp32err.txt)
我确定这些模组都被刷入过固件并进行过测试，但放置几周后就出现了该情况，目前我对其中的2块重新刷入固件后可以正常使用并且经过大约1周也没有再出现掉固件的情况。

Gui-ye, 2024-12-07:
你好，我也遇到了这个问题，请问最后是否成功找到了原因？

zztiger123, 2024-12-09:
> 被刷入过固件并进行过测试，但

@Gui-ye 你好，经查看您的 [esp32err.txt](https://github.com/user-attachments/files/18048740/esp32err.txt) ，里面的所有内容都是 0xFF, 请问您的场景是否有主控连接 ESP，或者有其他可能在应用层读写 flash 的行为？

目前还没有类似的问题上报

Gui-ye, 2024-12-09:
> > 被刷入过固件并进行过测试，但
> 
> @Gui-ye 你好，经查看您的 [esp32err.txt](https://github.com/user-attachments/files/18048740/esp32err.txt) ，里面的所有内容都是 0xFF, 请问您的场景是否有主控连接 ESP，或者有其他可能在应用层读写 flash 的行为？
> 
> 目前还没有类似的问题上报

@zztiger123 您好，我的测试固件是连接WIFI后建立MQTT通信，然后发送扫描到的AP信息至MQTT频道。我不确定连接WIFI时进行的nvs_flash_init是否是您所说的读写flash行为，为了防止有您提到的读写flash行为被我忽略我将附上我的测试代码，测试代码中的账号密码已经全部弃用。其中simple_wifi_sta(.h).txt是simple_wifi_sta.h。
[main.txt](https://github.com/user-attachments/files/18056037/main.txt)
[simple_wifi_sta(.h).txt](https://github.com/user-attachments/files/18056038/simple_wifi_sta.h.txt)
[simple_wifi_sta.txt](https://github.com/user-attachments/files/18056039/simple_wifi_sta.txt)

Gui-ye, 2024-12-09:
> @Gui-ye您好。
> 
> 您的代码理论上不会出现闪存被占用的情况。
> 
> 目前您的 4 片 C6 仅烧录您发的​​代码对应的固件，一周后 4 片 C6 的 flash 都完全被困了吗？换成任何其他固件后就不会一直被困吗？
> 
> 如果情况如上，总的来说我们还是更怀疑在哪里进行了误操作破坏了整个闪存

@ zztiger123 您好，可能我的描述让您有一些误解。我的意思是我在一个多月以前向这些模组烧录了这些固件，而大约一周前我在向这些模组烧录新的固件时发现了固件丢失的情况，在一周前的这一次烧录后直到现在也没有再出现丢固件的情况。
在一个多月前的那一次烧录后直到一周前的那一次烧录期间我只是把它们放置在静电防护袋中没有进行任何操作。
在我第一次烧录固件前我曾使用加热台将该模组焊接至PCB板上，模组型号是ESP32-C6-MINI-1-H8，焊接峰值温度250度。在第一次烧录后我没有再进行焊接。是否有可能因焊接温度控制误差而造成这种情况？

Rainbow-Cai, 2024-12-09:
Hi @Gui-ye  请问你使用的 esp-idf SDK  是什么版本的呢？你能提供固件正常运行时的完整日志吗？

Gui-ye, 2024-12-09:
> 你好@Gui-ye 请问你使用的 esp-idf SDK 是什么版本的呢？你能提供固件正常运行时的完整日志吗？

@Rainbow-Cai您好，我使用的是esp-idf-v5.2.2，并通过ESP-IDF  v1.8.1连接到VSCode。以下是我刚刚为一块丢失固件的模组第一次烧录（上文提到的剩余2块丢失固件后还没有重新刷入固件的模组之一）并正常运行时“监控设备”窗口传回的全部内容。
[log.txt](https://github.com/user-attachments/files/18060043/log.txt)

Rainbow-Cai, 2024-12-09:
@Gui-ye  可以对正常运行的设备尝试挂机测试看看能不能复现呢？挂机测试时麻烦保存设备运行日志到本地。

Gui-ye, 2024-12-09:
> @Gui-ye 可以对正常运行的设备尝试挂机测试看看能不能复现呢？挂机测试时麻烦保存设备运行日志到本地。

@Rainbow-Cai 您好，能不能麻烦您描述一下如何进行测试？我不是很清楚如何进行这项测试。

Rainbow-Cai, 2024-12-09:
@Gui-ye  设备下载固件后，通过 串口连接到电脑，在电脑端打开串口调试工具来捕捉设备运行日志，在串口调试工具页面将日志保存到本地，然后对设备进行完整的应用功能的测试，比如你之前提到过的测试环境和测试应用。主要是希望可以捕捉到你所说的现象的完整过程的日志。

Gui-ye, 2024-12-10:
> @Gui-ye 设备下载固件后，通过 串口连接到电脑，在电脑端打开串口调试工具来捕捉设备运行日志，在串口调试工具页面将日志保存到本地，然后对设备进行完整的应用功能的测试，比如你之前提到过的测试环境和测试应用。主要是希望可以捕捉到你所说的现象的完整过程的日志。

@Rainbow-Cai 您好，我无法复现该问题，因为在一周前的那次烧录后直到现在都没有再出现这个问题。即便我使用出现过这个问题的固件重新进行测试也没有再一次出现。这似乎是一个偶发的问题。

Rainbow-Cai, 2024-12-12:
@Gui-ye  如果无法进行复现或无法找到复现方式，则无法做进一步分析了。