espressif/esp-idf #13797: cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred

HDLA-BG, 2024-05-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32s3

### Power Supply used.

External 5V

### What is the expected behavior?

No errors when downloading file via Modem

### What is the actual behavior?

A lot of errors printed in the log coming from cdc_acm

### Steps to reproduce.

1. Connect to Internet via Modem using USB
2. Request OTA update
...

### Debug Logs.

```plain
E (156034) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156075) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156255) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156387) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156391) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156392) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156393) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156393) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156394) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156395) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156397) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156403) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156405) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156508) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156510) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156511) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
E (156511) cdc_acm: cdc_acm_host_data_tx_blocking(1183): Incorrect number of bytes transferred
```

### More Information.

We are using ESPModem driver and Modem SIMCOM A7672
The modem is Connected via USB.
The connection is stable until we request OTA update using https server.
After file transfer is started cfc_acm starts sending **"Incorrect number of bytes transferred"** and continue doing it during whole download process.

The problem doesn't appear every time but it is pretty often.

roma-jam, 2024-05-16:
Hi @HDLA-BG, 

Thanks for reporting the issue. 

Despite the printed error, the OTA firmware update finishes successfully? Or when errors appears, the the OTA firmware update process finishes with error as well?

HDLA-BG, 2024-05-16:
Hi @roma-jam 

OTA update finishes successfully.

roma-jam, 2024-05-21:
Hi @HDLA-BG ,

thanks for clarification, it is good to know that data is not corrupted during the transfer. 

Meanwhile, the error should not be there. 
Let me check some details and I will return back asap.

radiotommy, 2024-06-13:
@HDLA-BG 
I think we got  the same issue, I got it fixed and made a PR:
https://github.com/espressif/esp-usb/pull/40

HDLA-BG, 2024-06-13:
@radiotommy 

This is great, thanks, I hope it'll be merged soon.

closed 2024-06-17 by tore-espressif