espressif/esp-idf #13801: Strange websocket crashes at runtime

hitecSmartHome, 2024-05-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.6 2nd stage bootloader

### Espressif SoC revision.

ESP-32 Wrover-E revision v3.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-32 Wrover-E

### Power Supply used.

External 3.3V

### What is the expected behavior?

To not crash on ws send or ws connect or ws close

### What is the actual behavior?

It crashes randomly on ws send or ws connect or ws close

### Steps to reproduce.

Create a ws client, connect to an external server with ssl and let it run

### Debug Logs.

```plain
==================== THREAD 1 (TCB: 0x3ffcca70, name: 'loopTask') =====================
#0  0x40083170 in panic_abort (details=0x3ffcc5b0 "assert failed: 0x4008c7d6") at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_system\panic.c:408
#1  0x40089900 in esp_system_abort (details=0x3ffcc5b0 "assert failed: 0x4008c7d6") at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_system\esp_system.c:137
#2  0x4008e768 in __assert_func (file=<optimized out>, line=0, func=0x0, expr=0x0) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components
ewlib\assert.c:47
#3  0x4008c7d9 in tlsf_free (tlsf=0x3f803294, ptr=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\heap_tlsf.c:964
#4  0x4008d087 in multi_heap_free_impl (p=0x3f835bb4, heap=0x3f803280) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\multi_heap.c:212
#5  multi_heap_free_impl (heap=0x3f803280, p=0x3f835bb4) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\multi_heap.c:200
#6  0x40083efa in heap_caps_free (ptr=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\heap_caps.c:382
#7  0x4008e798 in free (ptr=0x3f835bb4) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components
ewlib\heap.c:39
#8  0x40264dcc in esp_tls_internal_event_tracker_destroy (h=0x3f835bb4) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp-tls\esp_tls_error_capture.c:46
#9  0x40264c6e in esp_tls_conn_destroy (tls=0x3f835bbc) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp-tls\esp_tls.c:111
#10 0x401f452a in base_close (t=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components cp_transport    ransport_ssl.c:270
#11 0x40275f2e in esp_transport_close (t=0x3ffe3288) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components     cp_transport    ransport.c:172
#12 0x401f4b94 in ws_close (t=0x3ffe3300) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components        cp_transport    ransport_ws.c:540
#13 0x40275f2e in esp_transport_close (t=0x3ffe3300) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components     cp_transport    ransport.c:172
#14 0x401bb9a6 in esp_websocket_client_abort_connection (client=0x3f8195c4) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c:161
#15 0x401bc4fe in esp_websocket_client_send_with_opcode (client=0x3f8195c4, opcode=<optimized out>, data=0x3f82f408 <error: Cannot access memory at address 0x3f82f408>, len=4892, timeout=500) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c:882
#16 0x401bc52c in esp_websocket_client_send_text (client=0x3f8195c4, data=0x3f82f408 <error: Cannot access memory at address 0x3f82f408>, len=4892, timeout=500) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c:829
#17 0x401b4041 in WebsocketClient::send (len=4892, data=0x3f82f408 <error: Cannot access memory at address 0x3f82f408>, this=0x3ffb350c <serverRouter>) at lib/WebsocketClient/WebsocketClient.cpp:38
[... 71 lines cut ...]
#7  0x4008e798 in free (ptr=0x3f835bb4) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components
ewlib\heap.c:39
#8  0x40264dcc in esp_tls_internal_event_tracker_destroy (h=0x3f835bb4) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp-tls\esp_tls_error_capture.c:46
#9  0x40264c6e in esp_tls_conn_destroy (tls=0x3f835bbc) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp-tls\esp_tls.c:111
#10 0x401f452a in base_close (t=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components cp_transport    ransport_ssl.c:270
#11 0x40275f2e in esp_transport_close (t=0x3ffe3288) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components     cp_transport    ransport.c:172
#12 0x401f4b94 in ws_close (t=0x3ffe3300) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components        cp_transport    ransport_ws.c:540
#13 0x40275f2e in esp_transport_close (t=0x3ffe3300) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components     cp_transport    ransport.c:172
#14 0x401bb9a6 in esp_websocket_client_abort_connection (client=0x3f8195c4) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c:161
#15 0x401bc4fe in esp_websocket_client_send_with_opcode (client=0x3f8195c4, opcode=<optimized out>, data=0x3f82f408 <error: Cannot access memory at address 0x3f82f408>, len=4892, timeout=500) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c:882
#16 0x401bc52c in esp_websocket_client_send_text (client=0x3f8195c4, data=0x3f82f408 <error: Cannot access memory at address 0x3f82f408>, len=4892, timeout=500) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c:829
#17 0x401b4041 in WebsocketClient::send (len=4892, data=0x3f82f408 <error: Cannot access memory at address 0x3f82f408>, this=0x3ffb350c <serverRouter>) at lib/WebsocketClient/WebsocketClient.cpp:38
#18 WebsocketClient::send (this=0x3ffb350c <serverRouter>, data=0x3f82f408 <error: Cannot access memory at address 0x3f82f408>, len=4892) at lib/WebsocketClient/WebsocketClient.cpp:36
#19 0x40142904 in HsHServerRouter::sendTrigger (this=0x3ffb350c <serverRouter>, triggerType=0x3f416e25 "sysInfo", message=...) at src/HsHServerRouter/Handlers.cpp:57
#20 0x4018cba2 in <lambda(int, int)>::operator() (__closure=0x3ffcc930, prev=<optimized out>, now=<optimized out>) at src/System/SystemInfo.cpp:11
#21 std::_Function_handler<void(int, int), Sys::sendSysInfo()::<lambda(int, int)> >::_M_invoke(const std::_Any_data &, int &&, int &&) (__functor=..., __args#0=<optimized out>, __args#1=<optimized out>) at c:\users\pc\.platformio\packages       oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
#22 0x40196ec0 in std::function<void (int, int)>::operator()(int, int) const (this=0x3ffcc930, __args#0=<optimized out>, __args#1=<optimized out>) at c:\users\pc\.platformio\packages  oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
#23 0x40196f88 in Time::emitMinChange (now=15, prev=14, this=<optimized out>) at src/Time/TimeEvents.cpp:45
#24 Time::emitMinChange (this=<optimized out>, prev=14, now=15) at src/Time/TimeEvents.cpp:39
#25 0x4019710a in Time::monitorMinChange (this=0x3ffb3d74 <hshTime>) at src/Time/TimeEvents.cpp:154
#26 0x40197133 in Time::handleEvents (this=0x3ffb3d74 <hshTime>) at src/Time/TimeEvents.cpp:205
#27 0x40195e79 in Time::loop (this=0x3ffb3d74 <hshTime>) at src/Time/Time.cpp:211
#28 0x401a0d1b in loop () at src/main.cpp:91
#29 0x401c41f4 in loopTask (pvParameters=<optimized out>) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

hitecSmartHome, 2024-05-17:
Here is the current decoded coreDump

```cpp
espcoredump.py v1.11.0
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffcca70, name: 'loopTask', GDB name: 'process 1073531504'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x401d6a14
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40083170          0x40083170 <panic_abort+24>
lbeg           0x4000c349          1073791817
lend           0x4000c36b          1073791851
lcount         0x0                 0
sar            0x10                16
ps             0x60723             395043
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x80089900          -2146920192
a1             0x3ffcc570          1073530224
a2             0x3ffcc5b0          1073530288
a3             0x3ffcc5c1          1073530305
a4             0x10                16
a5             0x0                 0
a6             0x0                 0
a7             0x1                 1
a8             0x0                 0
a9             0x1                 1
a10            0x3ffcc5c1          1073530305
a11            0x3ffcc5c1          1073530305
a12            0x3                 3
a13            0x3ffbf108          1073475848
a14            0x60f20             397088
a15            0x1                 1

==================== CURRENT THREAD STACK =====================
#0  0x40083170 in panic_abort (details=0x3ffcc5b0 "assert failed: 0x4008c7d6") at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_system\panic.c:408
#1  0x40089900 in esp_system_abort (details=0x3ffcc5b0 "assert failed: 0x4008c7d6") at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_system\esp_system.c:137
#2  0x4008e768 in __assert_func (file=<optimized out>, line=0, func=0x0, expr=0x0) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components
ewlib\assert.c:47
#3  0x4008c7d9 in tlsf_free (tlsf=0x3f803294, ptr=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\heap_tlsf.c:964
#4  0x4008d087 in multi_heap_free_impl (p=0x3f835bb4, heap=0x3f803280) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\multi_heap.c:212
#5  multi_heap_free_impl (heap=0x3f803280, p=0x3f835bb4) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\multi_heap.c:200
#6  0x40083efa in heap_caps_free (ptr=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\heap\heap_caps.c:382
[... 294 lines cut ...]
.coredump.tasks.data 0x3ffc46e0 0x1f0 RW
.coredump.tasks.data 0x3ffd1d00 0x164 RW
.coredump.tasks.data 0x3ffd4d70 0x1a0 RW
.coredump.tasks.data 0x3ffaf7cc 0x164 RW
.coredump.tasks.data 0x3ffaf620 0x1a0 RW
.coredump.tasks.data 0x3ffc0848 0x164 RW
.coredump.tasks.data 0x3ffc06d0 0x170 RW
.coredump.tasks.data 0x3ffbdee4 0x164 RW
.coredump.tasks.data 0x3ffbdd30 0x1a0 RW
.coredump.tasks.data 0x3ffc3778 0x164 RW
.coredump.tasks.data 0x3ffdfda0 0x1c0 RW
.coredump.tasks.data 0x3ffc61c0 0x164 RW
.coredump.tasks.data 0x3ffc5ff0 0x1c0 RW
.coredump.tasks.data 0x3ffd1a30 0x164 RW
.coredump.tasks.data 0x3ffd2860 0x1f0 RW
.coredump.tasks.data 0x3ffafa54 0x164 RW
.coredump.tasks.data 0x3ffba680 0x1a0 RW
.coredump.tasks.data 0x3ffda028 0x164 RW
.coredump.tasks.data 0x3ffd9d20 0x300 RW
.coredump.tasks.data 0x3ffd1b98 0x164 RW
.coredump.tasks.data 0x3ffd4550 0x1b0 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
```

hitecSmartHome, 2024-05-22:
I have tried to guard the send with a mutex

```cpp
boolean WebsocketClient::send(const char* data, size_t len) {
    if( !isConnected() || !data || !len ){ return false; }
    std::lock_guard<std::mutex> lock(sendMutex);
    if (esp_websocket_client_send_text(client, data, len, pdMS_TO_TICKS(500)) != -1) {
        return true;
    }
    return false;
}
```

Doesnt matter. The resulted crash is the same. It's inside esp_websocket_client_send_text

hitecSmartHome, 2024-05-22:
So it seems to me that it crashes when I reinit the websocket connection after a stop.

Sometimes my app calls this method

```cpp
boolean WebsocketClient::halt() {
    if( xSemaphoreTake( sendSemaphore, pdMS_TO_TICKS(1000) ) != pdTRUE ){ return false; }
    disconnect();
    stop();
    if( xSemaphoreGive( sendSemaphore )  != pdTRUE ){ return false; }
    return true;
}
```

which calls these

```cpp
void WebsocketClient::disconnect() {
    if (!isConnected()) {
        return;
    }
    esp_websocket_client_close(client, pdMS_TO_TICKS(1000));
}

void WebsocketClient::stop() {
    setAutoReconnect(false);
    esp_websocket_client_destroy(client);
}

void WebsocketClient::setAutoReconnect(boolean autoReconnect) {
    config.disable_auto_reconnect = !autoReconnect;
}
```

After it reinits with this

```cpp
void WebsocketClient::init() {
    client = esp_websocket_client_init(&config);
    esp_websocket_register_events(
        client,
        WEBSOCKET_EVENT_ANY,
        reinterpret_cast<esp_event_handler_t>(eventHandler),
        static_cast<void*>(this));
    esp_websocket_client_start(client);
}
```

And I've got a crash

```
0x400839bd: panic_abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_system\panic.c line 408
0x4008a131: esp_system_abort at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_system\esp_system.c line 137
0x4008f1a5: __assert_func at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\newlib\assert.c line 47
0x4008a725: xPortEnterCriticalTimeout at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_hw_support\include/soc/spinlock.h line 75
0x4008afc6: xQueueSemaphoreTake at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\freertos\port\xtensa\include/freertos/portmacro.h line 578
0x4027b0c1: xQueueTakeMutexRecursive at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\freertos\queue.c line 731
0x401bcf96: esp_websocket_client_send_with_opcode at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c line 854
0x401bd025: esp_websocket_client_send_text at C:\Users\Pc\.platformio\packages\framework-espidf@3.40406.240122\components\esp_websocket_client\esp_websocket_client.c line 829
0x401b43af: WebsocketClient::send(char const*, unsigned int) at lib/WebsocketClient/WebsocketClient.cpp line 41
0x401428b6: HsHServerRouter::reply(ArduinoJson::V700PB2::JsonDocument, bool) at src/HsHServerRouter/Handlers.cpp line 14
0x40140fbc: HsHServerRouter::dbDownload(ArduinoJson::V700PB2::JsonDocument, ArduinoJson::V700PB2::JsonDocument) at src/HsHServerRouter/DatabaseHandlers.cpp line 30
0x401411d7: HsHServerRouter::handleDBReq(std::__cxx11::basic_string  , std::allocator  >&, ArduinoJson::V700PB2::JsonDocument, ArduinoJson::V700PB2::JsonDocument) at src/HsHServerRouter/DatabaseHandlers.cpp line 7
0x4014589d: HsHServerRouter::handleRequest(ArduinoJson::V700PB2::JsonDocument, ArduinoJson::V700PB2::JsonDocument) at src/HsHServerRouter/HsHServerRouter.cpp line 123
0x40145cc6: HsHServerRouter::handleQueue() at src/HsHServerRouter/HsHServerRouter.cpp line 99
0x40145d14: HsHServerRouter::handleQueue() at src/HsHServerRouter/HsHServerRouter.cpp line 157
0x40145d3f: HsHServerRouter::loop() at src/HsHServerRouter/HsHServerRouter.cpp line 152
0x40145d5a: serverRouterTask(void*) at src/HsHServerRouter/HsHServerRouter.cpp line 9
```

closed 2024-07-02 by hitecSmartHome

andreichalapco, 2024-08-07:
Is there any resolution here? Did you find the fix?

hitecSmartHome, 2024-08-07:
Sorry. I have dropped idf websocket on 4.4.7 and made my own. It had a lot of issues.