espressif/esp-idf #13802: ESP32-S3 - IDF v5.4-dev-421 fails to build

AtkinEngineering, 2024-05-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-421-g4523f3d674

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Building the `hello_world` example should build correctly.

### What is the actual behavior?

Always produces the following error:
`error: unexpected output section continuous line "*(.debug)" at line 12052 in "Linker script and memory map" section in "/home/dev/esp/esp-idf/examples/get-started/hello_world/build/hello_world.map" map file`

### Steps to reproduce.

VSCode & ESP-IDF extension:
1) Configure ESP-IDF Extension, Express with ESP-IDF Version set to master (development branch)
2) Open /get-started/hello_world folder
3) Set target to ESP32s3 with ESP-PROG
4) Build

### Build or installation Logs.

```plain
*  Executing task: /home/dev/.espressif/python_env/idf5.4_py3.11_env/bin/python /home/dev/esp/master/esp-idf/tools/idf_size.py /home/dev/esp/master/esp-idf/examples/get-started/hello_world/build/hello_world.map 

error:  unexpected output section continuous line "*(.debug)" at line 12052 in "Linker script and memory map" section in 
"/home/dev/esp/master/esp-idf/examples/get-started/hello_world/build/hello_world.map" map file

 *  The terminal process "/usr/bin/bash '-c', '/home/dev/.espressif/python_env/idf5.4_py3.11_env/bin/python /home/dev/esp/master/esp-idf/tools/idf_size.py /home/dev/esp/master/esp-idf/examples/get-started/hello_world/build/hello_world.map'" terminated with exit code: 1.
```

### More Information.

I have tried this with my development code and the examples and both the same.

fhrbata, 2024-05-18:
Hello @AtkinEngineering,

thank you for reporting this issue and apologies for the inconvenience. The problem stems from recent changes in the linker scripts (https://github.com/espressif/esp-idf/commit/2b36636f6fb11e37671d384055db47ca2e0446ed) that were implemented to properly handle orphan sections.

While the build itself is fine, it appears that the `VSC` extension is automatically running `esp-idf-size` after the build. Unfortunately, it doesn't recognize empty output sections introduced by the aforementioned changes. We have already developed a fix for `esp-idf-size` and hope it will be merged soon.

In the meantime, you can use the legacy version of `esp-idf-size` as a workaround, because it doesn't have this problem. For context, `esp-idf-size` currently has two implementations: legacy and refactored, which can be dynamically switched. The refactored version has a more rigorous link map file parser, supports multiple output formats, and is easily extendable. It's by default enabled since `v5.3`.

To switch to the legacy version in `VSC`:

1. Go to `File` -> `Preferences` -> `Settings` and search for `idf.customExtraVars`.
2. Choose whether you want to change the `esp-idf-size` behavior globally by selecting the `User` tab, or just for a specific project by selecting the `Workspace` tab.
3. Click on `Add Item`.
4. Add the `ESP_IDF_SIZE_LEGACY` item(env. variable) with the value `1`.

Alternatively, you can directly edit the `.vscode/settings.json` file in your project/workspace. Here are my settings for reference.
```
{
    "C_Cpp.intelliSenseEngine": "default",
    "idf.adapterTargetName": "esp32s3",
    "idf.customExtraPaths": "/home/fhrbata/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/fhrbata/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:/home/fhrbata/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin:/home/fhrbata/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240305/riscv32-esp-elf/bin:/home/fhrbata/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/fhrbata/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:/home/fhrbata/.espressif/tools/esp-rom-elfs/20240305",
    "idf.customExtraVars": {
        "OPENOCD_SCRIPTS": "/home/fhrbata/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts",
        "ESP_ROM_ELF_DIR": "/home/fhrbata/.espressif/tools/esp-rom-elfs/20240305/",
        "ESP_IDF_SIZE_LEGACY": "1"
    },      
    "idf.espIdfPath": "/home/fhrbata/esp/master/esp-idf",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32.cfg"
    ],      
    "idf.port": "/dev/ttyUSB0",
    "idf.pythonBinPath": "/home/fhrbata/.espressif/python_env/idf5.4_py3.12_env/bin/python",
    "idf.toolsPath": "/home/fhrbata/.espressif",
    "idf.flashType": "UART"
}
```

Hopefully, this can temporarily help to address this problem.

Thank you

AtkinEngineering, 2024-05-18:
Hello @fhrbata,
This is fantastic, and makes sense.
Your suggestion works.
Thank you.

closed 2024-05-20 by espressif-bot