espressif/esp-idf #13803: Make standalone heap tracing more usefull and easy to read and analyse

masterxq, 2024-05-18:
### Is your feature request related to a problem?

tracing down memory leaks currently works like this:
```
heap_trace_start();
//Wait or try something
heap_trace_stop();
heap_trace_dump();
```

Now you will see a lot of allocated memory leaks that are false positive. And will be freed soon.

This can be avoided with a some small changes to heap trace api.

### Describe the solution you'd like.

As already mentioned many of the false positive will be freed soon and we don't want to see them.

This can be solved if it is possible only to stop tracing malloc and continue tracing free. Then it would be possible to wait longer or restart some services. So many of the false positive leaks will now be freed and new mallocs will not be collected. Now we have a more relevant output of the dump.

```
heap_trace_start();
//Wait or try something
heap_trace_stop_tracing_malloc(); //Only stop tracing malloc, continue tracing free.
//Wait longer for frees or test some actions that could trigger free
heap_trace_stop();
heap_trace_dump();
```
Now we should have much less false positives...

The standard workflow would still work, and it should be easily possible without breaking changes.

### Describe alternatives you've considered.

Even better would be if this workflow would also be possible, so we better know when or with which action stuff will be freed.
```
heap_trace_start();
//Wait or try something
heap_trace_stop_tracing_malloc(); //Only stop tracing malloc, continue tracing free.
//Wait longer or test some actions
while(trace_heap)
{
 if(button_pressed)
   heap_trace_dump(); //Don't know if it is currently possible to call dump before stop
}
heap_trace_stop();
```

### Additional context.

I think this would help a lot to make good bug reports about memory leaks without complicated jtag setup, what really can be hard to setup in products where the problem apears.

After all this could help to fix a lot of memory leaks, by helping users by helping.

Additionally, it would help the user to find leaks in user code. Without have tons of false positives from the framework that will be freed the next free seconds :)

Thanks for reading :)

SoucheSouche, 2024-05-20:
Hello @masterxq, thanks for submitting this feature request.

It is perfectly fine to use `heap_caps_dump` when the tracing is running.
Concerning the feature request, I will investigate it but it seems to make sense to be able to stop recording new allocations while continuing to record free calls.

I will keep you updated about the solution I will choose to implement.

chipweinberger, 2024-05-20:
i agree it makes sense

SoucheSouche, 2024-05-22:
Hi @masterxq,
I just merged the new feature locally. It will be available with next sync.

masterxq, 2024-05-22:
Very fast, very cool and very useful!

I will try to honor the work invested and give it a test by using the new feature as soon as possible to create a bug report that is as qualitative as possible.

Can it also be merged into 4.4.x?

Thank you!

SoucheSouche, 2024-05-23:
@masterxq,
The documentation  has been updated and will now list the new function next to `heap_trace_start` and `heap_trace_stop`
Since it is a new feature, it will not be backported.

masterxq, 2024-05-26:
Understand, thank you very much!

Best Regards!

closed 2024-05-26 by masterxq