espressif/esp-idf #13804: USB Remote Wakeup is ignored (Example project "tusb_hid")

kerryland, 2024-05-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1 and git master (commit 4523f2d67465373f0e732a3264273a8e84a1a6d1)

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3 WROOM FREENOVE, and Custom Board

### Power Supply used.

USB

### What is the expected behavior?

The example project "tusb_hid" (https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_hid), which illustrates the ability to wake a host device, should work.

### What is the actual behavior?

The application moves the mouse around, and simulates pressing the letter 'a', but the configuration descriptor `TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP` on [line 52](https://github.com/espressif/esp-idf/blob/4523f2d67465373f0e732a3264273a8e84a1a6d1/examples/peripherals/usb/device/tusb_hid/main/tusb_hid_example_main.c#L52) is ignored, i.e. Pressing the BOOT button, when the host computer is in sleep mode, does not wake the computer.

If you examine Device Manager you see that the ESP32 device does not support any power management or "wake' functionality:

[image]

### Steps to reproduce.

Using Visual Studio Code...
- Install esp-idf 5.2.1
- Create a project using example "tusb_hid"
- Change target to esp32s3 via built-in usb jtag
- Build the example: `idf.py all`
- Flash it
- Examine Device Manager, notice no "Power Management" functions
- Put computer to sleep
- Press BOOT on the ESP32
- Notice that the computer stays asleep

Build and flash logs attached. I tried 5.2.1 and git master (commit 4523f2d67465373f0e732a3264273a8e84a1a6d1)
[build-and-flash-5-4-dev.txt](https://github.com/espressif/esp-idf/files/15360757/build-and-flash-5-4-dev.txt)
[build-and-flash-5-2-1.txt](https://github.com/espressif/esp-idf/files/15360758/build-and-flash-5-2-1.txt)

[This forum post](https://forums.adafruit.com/viewtopic.php?t=193703) describes the same issue, with a proposed fix, but I couldn't find the code mentioned in the posting.

kerryland, 2024-05-18:
I have since noticed that if I press BOOT when the Windows machine is in SLEEP mode the computer's power button starts blinking, but does nothing more than that.

roma-jam, 2024-05-21:
Hi @kerryland, 

thanks a lot for such a detailed description!  

I will take a look and return back asap.