espressif/esp-idf #13808: Openthread border router can't use the Ethernet port

francisp2, 2024-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

esp32s3 chip revision: v0.2 /  esp32h2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP Thread Border Router/Zigbee Gateway Development Board Espressif Systems ESP32-H2 Series Thread/Zigbee/BLE

### Power Supply used.

USB

### What is the expected behavior?

I expected the Ethernet to work

Boards from https://www.aliexpress.com/item/1005005872059584.html (Official Espressif store)

### What is the actual behavior?

First I followed exactly the steps outlined here:
https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html
I get the errors in the attached log
Then I deleted the esp-idf directory,, and started again with 
git clone -b v5.2.1 --recursive https://github.com/espressif/esp-idf.git

Now I get this error

-- Configuring done
-- Generating done
-- Build files have been written to: /opt/esp-thread-br/examples/basic_thread_border_router/build
ninja: error: '/opt/esp-idf/components/openthread/lib/esp32h2/libopenthread_br.a', needed by 'esp_ot_br.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the /opt/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stderr_output_21847 and /opt/esp-thread-br/examples/basic_thread_border_router/build/log/idf_py_stdout_output_21847

### Steps to reproduce.

1. Step : use the exact steps in https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html
2. Step 2 : change to 5.2.1

...

### Debug Logs.

```plain
root@E1210:/opt/esp-idf/examples/openthread/ot_rcp# idf.py -p /dev/serial/by-id/                                                                                                                                                             usb-Espressif_USB_JTAG_serial_debug_unit_3C:84:27:17:F7:E4-if00   monitor
Executing action: monitor
Running idf_monitor in directory /opt/esp-idf/examples/openthread/ot_rcp
Executing "/root/.espressif/python_env/idf5.2_py3.11_env/bin/python /opt/esp-idf                                                                                                                                                             /tools/idf_monitor.py -p /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_u                                                                                                                                                             nit_3C:84:27:17:F7:E4-if00 -b 115200 --toolchain-prefix riscv32-esp-elf- --targe                                                                                                                                                             t esp32h2 --revision 0 --decode-panic backtrace /opt/esp-idf/examples/openthread                                                                                                                                                             /ot_rcp/build/esp_ot_rcp.elf -m '/root/.espressif/python_env/idf5.2_py3.11_env/b                                                                                                                                                             in/python' '/opt/esp-idf/tools/idf.py' '-p' '/dev/serial/by-id/usb-Espressif_USB                                                                                                                                                             _JTAG_serial_debug_unit_3C:84:27:17:F7:E4-if00'"...
[... 386 lines cut ...]

I (556) obtr_web: http://0.0.0.0:80/index.html

I (556) obtr_web: <===========================================================>

I (556) main_task: Returned from app_main()
I(596) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I(616) OPENTHREAD:[I] Settings------: Read NetworkInfo {rloc:0xd000, extaddr:0a3                                                                                                                                                             788376c57504f, role:router, mode:0x0f, version:4, keyseq:0x2, ...
I(616) OPENTHREAD:[I] Settings------: ... pid:0x2560e5c9, mlecntr:0x13bd6, maccn                                                                                                                                                             tr:0xc26, mliid:9b45186f95bb3f2f}
I (636) esp_ot_br: Internal RCP Version: openthread-esp32/e7771c75bd-456c44828;                                                                                                                                                              esp32h2;  2024-05-09 11:23:18 UTC
I (636) esp_ot_br: Running  RCP Version: openthread-esp32/e7771c75bd-456c44828;                                                                                                                                                              esp32h2;  2024-05-09 11:23:18 UTC
I (636) OPENTHREAD: OpenThread attached to netif
> E (736) w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
E (736) w5500.mac: emac_w5500_init(795): vefiry chip ID failed
I (736) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                                                                                                                                             n: 0| Intr:0
E (736) esp_eth: esp_eth_driver_install(228): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x10a (ESP_ERR_INVALID_VERSION) at 0x42020ec3
file: "/IDF/examples/common_components/protocol_examples_common/eth_connect.c" l                                                                                                                                                             ine 153
func: eth_start
expression: esp_eth_driver_install(&config, &s_eth_handle)

abort() was called at PC 0x4037df13 on core 0

Backtrace: 0x40375b8e:0x3fcb8d50 0x4037df1d:0x3fcb8d70 0x40384d82:0x3fcb8d90 0x4                                                                                                                                                             037df13:0x3fcb8e00 0x42020ec3:0x3fcb8e30 0x42020ae3:0x3fcb8f50 0x42008faa:0x3fcb                                                                                                                                                             8f70
```

francisp2, 2024-05-19:
Tried to compile again, same error with 2.5.1

[log_otbr.txt](https://github.com/espressif/esp-idf/files/15370018/log_otbr.txt)

There seems to be a similar issue here, but closed

zwx1995esp, 2024-05-21:
Hi, @francisp2  For the ethernet, did you use the `Sub-Ethernet daughter board` provided by ESP or other boards? If you used the ESP Sub-Ethernet board, could you please double check the configurations shown by the step: [2.1.3.2. Ethernet based Thread Border Router]( https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#ethernet-based-thread-border-router) ?

francisp2, 2024-05-21:
It are the Ethernet boards provided with the development kit
[image]

I will double check the settings this evening

francisp2, 2024-05-21:
Ok, seems better now, but now another problem. I have 2 esp32-h2/esp32s3 development kits, and 2 Ethernet daughterboards, and both daughterboards seem to have the same mac : 02:00:00:12:34:56
is that somewhere coded in the firmware ?

zwx1995esp, 2024-05-22:
Hi, @francisp2 After a quick code reviewing, you can manually set the ETH mac [here](https://github.com/espressif/esp-idf/blob/v5.1.3/examples/common_components/protocol_examples_common/eth_connect.c#L158). I'm not very familiar with ETH driver code, I will discuss it with the colleague who is responsed to this feature, and then share the result here.

francisp2, 2024-05-23:
Patch is working 

> I (645) esp_eth.netif.netif_glue: cc:8d:a2:29:a2:83

closed 2024-05-23 by francisp2