espressif/esp-idf #13809: Failed to strip unused libstdc++code after f8319f08cca75236c29eecfce01886a671f4fc17

andylinpersonal, 2024-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Linker should strip unused ```libstdc++``` code, such as ```iostream```, which is so bulky that I noticed this regression immediately :(

### What is the actual behavior?

After f8319f08cca75236c29eecfce01886a671f4fc17 on release/v5.2, the unused `libstdc++` functions remain inside the final binary.

This regression can only be reproduced on ESP32-S3. (and possible for other Xtensa targets)

### Steps to reproduce.

Just update the IDF after f8319f08cca75236c29eecfce01886a671f4fc17, use some standard c++ functions and see the size of final binary by ```idf.py size-components```.

MWE:
```C++
#include <iterator>
// Other C++ standard library functions implemented in the libstdc++.a can also be used.
#include <mutex>
#include <cmath>
#include <cstdio>

extern "C" void app_main(void)
{
	std::mutex mt;
	std::lock_guard lg(mt);
}

```

Following lines might cause this regression. I just reverted these lines and everything looks ok now.

[image]

### Build or installation Logs.

```plain
# idf.py size-components before f8319f08cca75236c29eecfce01886a671f4fc17

Total sizes:
Used static IRAM:   57486 bytes ( 304754 remain, 15.9% used)
      .text size:   56459 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   16972 bytes ( 328884 remain, 4.9% used)
      .data size:   11724 bytes
      .bss  size:    5248 bytes
Used Flash size :  157443 bytes
           .text:  103987 bytes
[... 68 lines cut ...]
             libxtensa.a       1060       0        2057         424         3541         119        48          0        3708
             libnewlib.a        164     200        1593           0         1957        1209        95          0        3061
             libesp_mm.a          4      44         459           0          507        2420       166          0        3049
         libmbedcrypto.a         56      12           0           0           68        2651        56          0        2763
                libsoc.a          0       0         245           0          245           0      2020          0        2265
             libdriver.a         36       0           0           0           36        1844       281          0        2161
      libesp_partition.a          0       8           0           0            8        1669       168          0        1837
         libesp_common.a          0       0           0           0            0          51      1742          0        1793
          libesp_timer.a         32      36         512           0          580        1060        57          0        1661
                libcxx.a          8      16           0           0           24         946         0          0         954
                liblog.a          8     272         238           0          518         636        21          0         903
 libbootloader_support.a          0       0         812           0          812          34        40          0         886
            libesp_rom.a          0       0         446           0          446          78         0          0         524
     libesp_app_format.a          0      10           0           0           10         184         4        256         444
             libxt_hal.a          0       0         405           0          405           0        32          0         437
         libapp_update.a          0       4           0           0            4         152        30          0         182
               libmain.a          0       0           0           0            0          45         0          0          45
              libefuse.a          0       0           0           0            0          17         0          0          17
                   (exe)          0       0           0           3            3           3         8          0          14
            libesp_phy.a          0       0           0           0            0           5         0          0           5
   libnvs_sec_provider.a          0       0           0           0            0           5         0          0           5
        libesp_ringbuf.a          0       0           0           0            0           0         0          0           0
                libgcc.a          0       0           0           0            0           0         0          0           0
```
```

sudeep-mohanty, 2024-05-20:
Hello @andylinpersonal, I can confirm that I observe the same behavior. We will take a look and update you once we have more information. Thanks!

AxelLin, 2024-06-06:
@Lapshin Is there any update for the fix? 
v5.2 is a release branch, I think this should be fixed ASAP.
(If it really takes time to fix it, maybe you can consider revert the bad commit first.)

Lapshin, 2024-06-13:
@AxelLin , thank you for your attention on this. The commit https://github.com/espressif/esp-idf/commit/5320ec20f9302f260265b27d55ade1bc92b92ea9 was ready a few weeks ago but waited to be synced with GitHub repo.

Closing this issue because fixed for all affected branches

closed 2024-06-13 by Lapshin

andylinpersonal, 2024-06-13:
Fixed in master so close this ticket