espressif/esp-idf #13810: ESPC32-H2: Example HA_ON_OFF_SWITCH renders W (509) ESP_ZB_ON_OFF_SWITCH: Network(0x3700) closed, devices joining not allowed.

phelicksg, 2024-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

Chip is ESP32-H2 (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-H2-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

According to the example (I just xcopied the HA_on_off_switch without any modifications), this is what I expected (part of the README.md):
This is the expected behaviour as given in the README.md from the standard Zigbee HA_on_off_switch example that comes with ESP-IDF 5.2.1 (stable):

I (388) main_task: Calling app_main()
I (398) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (398) phy_init: phy_version 220,2dbbbe7,Sep 25 2023,20:39:25
I (478) phy: libbtbb version: 90c587c, Sep 25 2023, 20:39:57
I (488) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (488) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
I (488) ESP_ZB_ON_OFF_SWITCH: Start network formation
I (498) main_task: Returned from app_main()
I (998) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (998) ESP_ZB_ON_OFF_SWITCH: Formed network successfully (Extended PAN ID: 60:55:f9:00:00:f6:07:b4, PAN ID: 0x2a74, Channel:13)
I (1468) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (1468) ESP_ZB_ON_OFF_SWITCH: Network steering started
I (14228) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (14728) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (14788) ESP_ZB_ON_OFF_SWITCH: New device commissioned or rejoined (short: 0xe399)
I (14858) ESP_ZB_ON_OFF_SWITCH: Found light
I (14858) ESP_ZB_ON_OFF_SWITCH: Try to bind On/Off
I (14858) ESP_ZB_ON_OFF_SWITCH: Bound successfully!
I (14858) ESP_ZB_ON_OFF_SWITCH: The light originating from address(0xe399) on endpoint(10)
I (15338) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (15408) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (35838) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (38548) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
I (39598) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
I (41148) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command

### What is the actual behavior?

The example builds, and I can flash it. When I do idf.py monitor, the program does not behave as indicated in the README.md from the example. Instead it renders errors:

W (509) ESP_ZB_ON_OFF_SWITCH: Network(0x9e84) closed, devices joining not allowed.
I (519) ESP_ZB_ON_OFF_SWITCH: Device started up in non factory-reset mode
I (519) ESP_ZB_ON_OFF_SWITCH: Device rebooted

I also tried rebooting the devboard several times (and even power-cycled it) - same behaviour. See full logs below in debug logs.

### Steps to reproduce.

Downloaded esp-idf 5.2.1 (stable) using the Windows installer. Standard install. Also installed the support for ESP32-H2 and ESP32-C6.
mkdir HA_on_off_switch
xcopy /e /i %IDF_PATH%\examples\bee\light_sample\HA_on_off_switch HA_on_off_switch
cd HA_on_off_switch
idf.py set-target esp32h2
idf.py menuconfig (did not change anything)
idf.py build
idf.py erase-flash
idf.py flash
idf.py monitor

### Debug Logs.

```plain
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x17ac
load:0x4083cfd0,len:0xf04
load:0x4083efd0,len:0x2ca8
entry 0x4083cfda
I (23) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (24) boot: compile time May 19 2024 19:19:11
I (25) boot: chip revision: v0.1
I (27) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (36) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 000e1000
I (79) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=42090020 size=11cd0h ( 72912) map
I (129) esp_image: segment 1: paddr=00021cf8 vaddr=40800000 size=06320h ( 25376) load
I (139) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=8d67ch (579196) map
I (317) esp_image: segment 3: paddr=000b56a4 vaddr=40806320 size=080c0h ( 32960) load
I (332) boot: Loaded app from partition at offset 0x10000
I (333) boot: Disabling RNG early entropy source...
I (344) cpu_start: Unicore app
W (353) clk: esp_perip_clk_init() has not been implemented yet
I (360) cpu_start: Pro cpu start user code
I (360) cpu_start: cpu freq: 96000000 Hz
I (361) cpu_start: Application information:
I (363) cpu_start: Project name:     light_switch
I (368) cpu_start: App version:      1
I (373) cpu_start: Compile time:     May 19 2024 19:18:50
I (379) cpu_start: ELF file SHA256:  9ff5c20a6...
I (384) cpu_start: ESP-IDF:          v5.2.1
I (389) cpu_start: Min chip rev:     v0.0
I (394) cpu_start: Max chip rev:     v0.99
I (399) cpu_start: Chip rev:         v0.1
I (403) heap_init: Initializing. RAM available for dynamic allocation:
I (411) heap_init: At 408180B0 len 000352D0 (212 KiB): RAM
I (417) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (424) spi_flash: detected chip: generic
I (427) spi_flash: flash io: dio
W (431) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (445) sleep: Configure to isolate all GPIO pins in sleep state
I (451) sleep: Enable automatic switching of GPIO sleep configuration
I (459) main_task: Started on CPU0
I (459) main_task: Calling app_main()
I (469) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (479) phy: phy_version: 230,2, 9aae6ea, Jan 15 2024, 11:17:12
I (479) phy: libbtbb version: 944f18e, Jan 15 2024, 11:17:25
I (489) main_task: Returned from app_main()
I (499) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (499) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
W (509) ESP_ZB_ON_OFF_SWITCH: Network(0x9e84) closed, devices joining not allowed.
I (519) ESP_ZB_ON_OFF_SWITCH: Device started up in non factory-reset mode
I (519) ESP_ZB_ON_OFF_SWITCH: Device rebooted

In other words, it never gets to the network formation.
I erased the devboard's flash before flashing the example.
```

### More Information.

I have tried the same on a different devkit: ESP32-C6-DevKitM-1. The same problem occurs as well using that set-up. Of course I have done idf.py set-target esp32c6, idf.py menuconfig, idf.py build, idf.py erase-flash, idf.py flash, idf.py monitor and I have also power-cycled the devkit.
I am using USB UART bridge (not sure this makes any difference).

cy-lhe, 2024-05-20:
Have the same issue, example does not open the network for registration as "advertised" in example Readme.
Tried on ESP32-C6.

phelicksg, 2024-05-20:
I just ran off.py erase-flash again, switched off power to devboard. Followed by IDF.pt flash and  powered off.devboard again. Powered on followed by idf.py monitor. Same behaviour, unfortunately, still does not work. Built on 5.2.1

MeryaneF, 2024-06-20:
I have the same problem, I tried modifying the code, but without success. The network remains closed.

xieqinan, 2024-07-17:
@phelicksg @cy-lhe @MeryaneF ,

In this example, the Zigbee network will be closed when the device is not factory reset. This means that if the device is on a network and reboots, the network is closed by default. You can call `esp_zb_bdb_open_network()` to open the network again.
```
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Deferred driver initialization %s", deferred_driver_init() ? "failed" : "successful");
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network formation");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_FORMATION);
            } else {
                esp_zb_bdb_open_network(180); // Open network 180s
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
            ESP_LOGE(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        }
        break;
```

By the way, the above logic will be added to these examples in the future.

closed 2024-11-13 by chshu