espressif/esp-idf #13811: ESP32-C3 CAN Bus (TWAI) Filter

procoderer, 2024-05-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Can someone help me calculate the acceptance code and acceptance mask for receiving as close to possible only CAN messages with Standard IDs 0x355 and 0x356?

I already tried 
  acceptance code: 0x354 << 21
  acceptance mask: 0x003 << 21
which accepted 0x356 but not 0x355

Thanks

wanckl, 2024-05-21:
@procoderer You remain only one step to success!

[image]

Refer to programming guide, using single filter mode for std-id, it can filter out not only ID phase but also some data bytes,, 

So if you only care about the ID, you need bypass data check in mask like: `(0x3 << 21) | 0x1FFFFF`

nopnop2002, 2024-05-21:
@procoderer 
https://www.esp32.com/viewtopic.php?t=16575

procoderer, 2024-05-21:
> @procoderer You remain only one step to success! [...]

Thank you. This seems to work perfectly. 

Would you say that anytime I want to bypass data check I should also just add on `| 0x1FFFFF`?

wanckl, 2024-05-22:
@procoderer 
`| 0x1FFFFF` in mask means bypass check for all other bits except ID domain, If you using ext-id or daul-filter mode, you change the bits position and number relatively

wanckl, 2024-05-27:
@nopnop2002  can you help closing this issue if no more question, thanks

nopnop2002, 2024-05-27:
@wanckl 

I am not the owner of this issue, so I cannot close it.

closed 2024-05-27 by procoderer