espressif/esp-idf #13812: idf.py menuconfig failed with IDF 5.2.1

lincyang, 2024-05-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I expected it to run idf.py menuconfig and show menu window.

### What is the actual behavior?

There is error and can not show menu window. You can see the log.

### Steps to reproduce.

1. copy the adf example play_http_mp3_example to my workspace
2. click ESP-IDF 5.2 PowerShell on the desktop
3. cd play_http_mp3_example path and run idf.py menuconfig

### Build or installation Logs.

```plain
PS D:\workplace\esp32\pipeline_http_mp3> idf.py menuconfig
Executing action: menuconfig
Running ninja in directory D:\workplace\esp32\pipeline_http_mp3\build
Executing "ninja menuconfig"...
[0/1] Re-running CMake...-- ccache will be used for faster recompilation
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Processing 4 dependencies:
[1/4] espressif/esp-dsp (1.4.12)
[2/4] espressif/jsmn (1.1.0)
[3/4] espressif/nghttp (1.58.0)
[4/4] idf (5.2.1)
-- Project sdkconfig file D:/workplace/esp32/pipeline_http_mp3/sdkconfig
Loading defaults file D:/workplace/esp32/pipeline_http_mp3/sdkconfig.defaults...
Loading defaults file D:/workplace/esp32/pipeline_http_mp3/sdkconfig.defaults.esp32...
D:/workplace/esp32/pipeline_http_mp3/sdkconfig.defaults.esp32:17 CONFIG_ESP32_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
-- Compiler supported targets: xtensa-esp-elf
-- App "play_http_mp3" version: 1
-- Adding linker script D:/workplace/esp32/pipeline_http_mp3/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-v5.2.1/components/soc/esp32/ld/esp32.peripherals.ld
-- Chip is esp32, D:/workplace/workplace/github/esp32-main/esp-adf/components/esp-adf-libs
-- Current board name is CONFIG_ESP_LYRATD_MSC_V2_2_BOARD
-- Components: adf_utils app_trace app_update audio_board audio_hal audio_mixer audio_pipeline audio_recorder audio_sal audio_stream battery_service bluetooth_service bootloader bootloader_support bt clouds cmock console coredump_upload_service cxx display_service driver dueros_service efuse esp-adf-libs esp-sr esp-tls esp_actions esp_adc esp_app_format esp_bootloader_format esp_codec_dev esp_coex esp_common esp_dispatcher esp_eth esp_event esp_event_cast esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_peripherals esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-dsp espressif__jsmn espressif__nghttp esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 input_key_service json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread ota_service partition_table perfmon playlist protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tone_partition ulp unity usb vfs wear_levelling wifi_provisioning wifi_service wpa_supplicant xtensa
-- Component paths: D:/workplace/workplace/github/esp32-main/esp-adf/components/adf_utils D:/Espressif/frameworks/esp-idf-v5.2.1/components/app_trace D:/Espressif/frameworks/esp-idf-v5.2.1/components/app_update D:/workplace/workplace/github/esp32-main/esp-adf/components/audio_board D:/workplace/workplace/github/esp32-main/esp-adf/components/audio_hal D:/workplace/workplace/github/esp32-main/esp-adf/components/audio_mixer D:/workplace/workplace/github/esp32-main/esp-adf/components/audio_pipeline D:/workplace/workplace/github/esp32-main/esp-adf/components/audio_recorder D:/workplace/workplace/github/esp32-main/esp-adf/components/audio_sal D:/workplace/workplace/github/esp32-main/esp-adf/components/audio_stream D:/workplace/workplace/github/esp32-main/esp-adf/components/battery_service D:/workplace/workplace/github/esp32-main/esp-adf/components/bluetooth_service D:/Espressif/frameworks/esp-idf-v5.2.1/components/bootloader D:/Espressif/frameworks/esp-idf-v5.2.1/components/bootloader_support D:/Espressif/frameworks/esp-idf-v5.2.1/components/bt D:/workplace/workplace/github/esp32-main/esp-adf/components/clouds D:/Espressif/frameworks/esp-idf-v5.2.1/components/cmock D:/Espressif/frameworks/esp-idf-v5.2.1/components/console D:/workplace/workplace/github/esp32-main/esp-adf/components/coredump_upload_service D:/Espressif/frameworks/esp-idf-v5.2.1/components/cxx D:/workplace/workplace/github/esp32-main/esp-adf/components/display_service D:/Espressif/frameworks/esp-idf-v5.2.1/components/driver D:/workplace/workplace/github/esp32-main/esp-adf/components/dueros_service D:/Espressif/frameworks/esp-idf-v5.2.1/components/efuse D:/workplace/workplace/github/esp32-main/esp-adf/components/esp-adf-libs D:/workplace/workplace/github/esp32-main/esp-adf/components/esp-sr D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp-tls D:/workplace/workplace/github/esp32-main/esp-adf/components/esp_actions D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_adc D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_app_format D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_bootloader_format D:/workplace/workplace/github/esp32-main/esp-adf/components/esp_codec_dev D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_coex D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_common D:/workplace/workplace/github/esp32-main/esp-adf/components/esp_dispatcher D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_eth D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_event D:/workplace/workplace/github/esp32-main/esp-adf/components/esp_event_cast D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_gdbstub D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_hid D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_http_client D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_https_ota D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_https_server D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_hw_support D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_lcd D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_local_ctrl D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_mm D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_netif D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_netif_stack D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_partition D:/workplace/workplace/github/esp32-main/esp-adf/components/esp_peripherals D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_phy D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_pm D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_psram D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_ringbuf D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_rom D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_system D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_timer D:/Espressif/frameworks/esp-idf-v5.2.1/components/esp_wifi D:/Espressif/frameworks/esp-idf-v5.2.1/components/espcoredump D:/workplace/esp32/pipeline_http_mp3/managed_components/espressif__esp-dsp D:/workplace/esp32/pipeline_http_mp3/managed_components/espressif__jsmn D:/workplace/esp32/pipeline_http_mp3/managed_components/espressif__nghttp D:/Espressif/framew [... cut ...] config --defaults D:/workplace/esp32/pipeline_http_mp3/sdkconfig.defaults --defaults D:/workplace/esp32/pipeline_http_mp3/sdkconfig.defaults.esp32 --env-file D:/workplace/esp32/pipeline_http_mp3/build/config.env --env IDF_TARGET=esp32 --env IDF_TOOLCHAIN=gcc --env IDF_ENV_FPGA= --env IDF_INIT_VERSION=5.2.1 --output config D:/workplace/esp32/pipeline_http_mp3/sdkconfig"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information.

my system is Windows 11 家庭中文版

Honza0297, 2024-05-21:
Hello @lincyang and thank you for reaching us! 

This issue should be already fixed in the latest version of `release/v5.2` branch. 

I would suggest you to install ESP-IDF directly form the github and use the latest version of `release/v5.2` branch. Please, let me know if it fixes the problem. Thank you!

dobairoland, 2024-05-31:
I assume that is fixed and closing it. Please let me know if I'm mistaken.

v5.2.2 passed the tests and will be released soon.

closed 2024-05-31 by dobairoland