espressif/esp-idf #13814: [v5.1.4][Nimble] ble_hs_adv_fields memory is mutated after calling start_nimble_port_run

chipweinberger, 2024-05-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-s3

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

I keep `static struct ble_hs_adv_fields advFields;` in global memory. I expect these fields to not change unless I change them.

However, after calling `start_nimble_port_run`, the first 4 bytes of `advFields.mfg_data` gets mutated

**Expected vs Actual:**
```
advFields.mfg_data original data              : ce73ce87e5665669e3c73a7c5bf5f79f
advFields.mfg_data after start_nimble_port_run: a748be6ee5665669e3c73a7c5bf5f79f
```

**I use a watchpoint to determine the offending code (see logs below):**
```
static struct ble_hs_adv_fields advFields;

static void start_gap_advertise(void)
{

   ....

    // jamcorderUuid
    pd_128_bit_t jamcorderUuid = pd_jamcorder_uuid_bytes();
    advFields.mfg_data = jamcorderUuid.bytes;
    advFields.mfg_data_len = 16;

    // Set a breakpoint on the specified memory address
    esp_err_t err = esp_cpu_set_watchpoint(0, &(advFields.mfg_data[0]), 1, ESP_CPU_WATCHPOINT_STORE);
    if (err != ESP_OK){
        ESP_LOGE(TAG, "esp_cpu_set_watchpoint %s", esp_err_to_name(err));
    }
     ...
     
     rc = ble_gap_adv_rsp_set_fields(&bAdvFields);
     
     rc = ble_gap_adv_start(ownAddrType, NULL, BLE_HS_FOREVER, &params, handle_gap_event, NULL);
}
```

### What is the actual behavior?

`advFields.mfg_data` get mutated after calling `start_nimble_port_run`

### Steps to reproduce.

1. set nimble `mfg_data`
2. call `start_nimble_port_run`

### Debug Logs.

```plain
// These logs come from the watch point

I (22314) NimBLE: GAP procedure initiated: advertise; 
I (22314) NimBLE: disc_mode=2
I (22314) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (22324) NimBLE: 

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Watchpoint 0 triggered 
Core  0 register dump:
PC      : 0x4038812e  PS      : 0x00060636  A0      : 0x80384b50  A1      : 0x3fce7a20  
0x4038812e: xTaskGetSchedulerState at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5152

A2      : 0xe54b5901  A3      : 0x3fca88ac  A4      : 0x00060620  A5      : 0xe54b5901  
A6      : 0x3fca88ac  A7      : 0xb33fffff  A8      : 0xe54b5901  A9      : 0x3fce7a00  
A10     : 0x00060823  A11     : 0x00000000  A12     : 0x00060820  A13     : 0x3fca88ac  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x4038812b:0x3fce7a20 0x40384b4d:0x3fce7a50 0x40384c7a:0x3fce7a90 0x420a0681:0x3fce7ac0 0x42092136:0x3fce7af0 0x42097e15:0x3fce7b20 0x4209201a:0x3fce7b60 0x4209fe8e:0x3fce7b90 0x40379a14:0x3fce7bc0 0x4201a860:0x3fce7bf0 0x4038502e:0x3fce7c20
0x4038812b: xTaskGetSchedulerState at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5152
0x40384b4d: xQueueSemaphoreTake at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1662
0x40384c7a: xQueueTakeMutexRecursive at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:783
0x420a0681: npl_freertos_mutex_pend at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:508
0x42092136: ble_hs_lock at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:210
0x42097e15: ble_l2cap_sig_extract_expired at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap_sig.c:1906
 (inlined by) ble_l2cap_sig_timer at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap_sig.c:1985
0x4209201a: ble_hs_timer_exp at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:418
 (inlined by) ble_hs_timer_exp at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:408
0x4209fe8e: npl_freertos_event_run at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:451
0x40379a14: ble_npl_event_run at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:279
0x4201a860: start_nimble_port_run(void*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jcommon/bluetooth/pd_bluetooth.cpp:300
0x4038502e: vPortTaskWrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

chipweinberger, 2024-05-20:
I'm not sure if this is an actual issue. Everything works fine. But it is unexpected behavior and broke some of my assumptions.

ESPAbhinav, 2024-05-29:
Hi @chipweinberger ,
I tried the steps mentioned at your end but could not reproduce the issue.
Steps I followed. Using bleprph application to reproduce scenario
1. Set the data manufacture data before nimble_port_run()
2. Start nimble_port_run()
3. Check the manufacturer data just before starting the advertisement.

Below is Screenshot
[image]

When the host successfully starts, the application is notified via the ble_hs_cfg.sync_cb callback which is bleprph_on_sync in my case and hence advertisement will be started. 

Referring to your code manufacture data is assigned again after the nimble_port_run() which will change the mfg data. 
```
static void start_gap_advertise(void)
{

    // jamcorderUuid
    pd_128_bit_t jamcorderUuid = pd_jamcorder_uuid_bytes();
    advFields.mfg_data = jamcorderUuid.bytes;
    advFields.mfg_data_len = 16;

     rc = ble_gap_adv_rsp_set_fields(&bAdvFields);
     
     rc = ble_gap_adv_start(ownAddrType, NULL, BLE_HS_FOREVER, &params, handle_gap_event, NULL);
}
```

To know more about your issue, please send your application code/ Changes by which you face the issue.

chipweinberger, 2024-05-29:
Hi, thank you for your support. Very curious.

This is all of my bluetooth code:

[pd_bluetooth.txt](https://github.com/espressif/esp-idf/files/15487800/pd_bluetooth.txt)

```
// Enables advertising with the following parameters:
//     - General discoverable mode.
//     - Undirected connectable mode.
static void start_gap_advertise(void)
{
    // already advertising?
    if (ble_gap_adv_active() == true) {
        return;
    }

    ESP_LOGI(TAG, "***** start advertising *****");

    // Set advertisement data
    memset(&advFields, 0, sizeof(ble_hs_adv_fields));

    // Advertise flags
    advFields.flags = BLE_HS_ADV_F_DISC_GEN | // general discoverability == advertise forever
                      BLE_HS_ADV_F_BREDR_UNSUP; // BR/EDR unsupported

    // jamcorder name
    advFields.name = (uint8_t *)bleDeviceName;
    advFields.name_len = strbuflen(bleDeviceName);
    advFields.name_is_complete = 1;

    // jamcorderUuid
    pd_128_bit_t jamcorderUuid = pd_jamcorder_uuid_bytes();
    advFields.mfg_data = jamcorderUuid.bytes;
    advFields.mfg_data_len = 16;

    // set advertising data
    int rc;
    rc = ble_gap_adv_set_fields(&advFields);
    if (rc != 0) {
        // raise issue
        pd_raise_e2(PD_ISSUE_MAJOR_INTERNAL_BLUETOOTH_ISSUE, 
            "bleAdvFail", TAG,
            "error setting advertisement data; 0x%04x %s, %s",
                rc, ble_rc_category_str(rc), ble_rc_str(rc));
        return;
    }

    // Set Scan Resp Fields
    memset(&bAdvFields, 0, sizeof(ble_hs_adv_fields));

    // Advertise flags
    bAdvFields.flags = BLE_HS_ADV_F_DISC_GEN | // general discoverability == advertise forever
                   BLE_HS_ADV_F_BREDR_UNSUP; // BR/EDR unsupported

    // midi svc uuid
    bAdvFields.uuids128 = &pd_ble_uuid_svc_midi;
    bAdvFields.num_uuids128 = 1;
    bAdvFields.uuids128_is_complete = true;

    // set Scan Resp Fields
    rc = ble_gap_adv_rsp_set_fields(&bAdvFields);
    if (rc != 0) {
        // raise issue
        pd_raise_e2(PD_ISSUE_MAJOR_INTERNAL_BLUETOOTH_ISSUE, 
            "bleAdvFail2", TAG,
            "error setting advertisement response fields; 0x%04x %s, %s",
                rc, ble_rc_category_str(rc), ble_rc_str(rc));
        return;
    }

    // Advertising params
    // Unirected advertising (UND): accept connection from any peer device.
    memset(&params, 0, sizeof(params));
    params.conn_mode = BLE_GAP_CONN_MODE_UND;
    params.disc_mode = BLE_GAP_DISC_MODE_GEN;
    params.itvl_min = advFields.adv_itvl;       // units: 0.625 milliseconds
    params.itvl_max = advFields.adv_itvl * 1.2; // units: 0.625 milliseconds

    // Begin advertising
    rc = ble_gap_adv_start(ownAddrType, NULL, BLE_HS_FOREVER, &params, handle_gap_event, NULL);
    if (rc != 0 && rc != BLE_HS_EALREADY) {
        // raise issue
        pd_raise_e2(PD_ISSUE_MAJOR_INTERNAL_BLUETOOTH_ISSUE, "bleAdvStart", TAG,
            "error enabling advertisement; 0x%04x %s, %s",
                rc, ble_rc_category_str(rc), ble_rc_str(rc));
        return;
    }
}

```

chipweinberger, 2024-05-29:
I think the problem is 

```
    // jamcorderUuid
    pd_128_bit_t jamcorderUuid = pd_jamcorder_uuid_bytes(); 
    advFields.mfg_data = jamcorderUuid.bytes;
    advFields.mfg_data_len = 16;
```

Yes you are right. Here, `advFields.mfg_data` is pointing to the stack, so it makes sense that after `start_nimble_port_run` the stack memory will change dramatically.

My apologies!

closed 2024-05-29 by chipweinberger