espressif/esp-idf #13819: assert failed: vTaskPriorityDisinheritAfterTimeout tasks.c:5160 (pxTCB->uxMutexesHeld)

MR-VENOM, 2024-05-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-307-g99daa62628

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

Trying to create a program with GATTs when Sent CM create a task and When sent CN it deletes the same task. The watchdog is diablesd and the loop for the task has no delay.

### What is the actual behavior?

the task is create successfully when sent CM but   After sending CN it sometimes HALT the entire system or panic with bellow log.

assert failed: vTaskPriorityDisinheritAfterTimeout tasks.c:5160 (pxTCB->uxMutexesHeld)

```
Backtrace: 0x40375a56:0x3fcb5790 0x40380cf9:0x3fcb57b0 0x40387f69:0x3fcb57d0 0x4038383a:0x3fcb58f0 0x40381720:0x3fcb5910 0x40387d28:0x3fcb5950 0x420a1d72:0x3fcb5970 0x40387cb5:0x3fcb59a0 0x42009da6:0x3fcb59f0 0x42009966:0x3fcb5ca0 0x42024b06:0x3fcb5cd0 0x420255e9:0x3fcb5cf0 0x4204b271:0x3fcb5d40 0x4204d77b:0x3fcb5d60 0x4038185d:0x3fcb5d90
0x40375a56: panic_abort at /home/nikhi/esp/v5.2.1/esp-idf/components/esp_system/panic.c:472
0x40380cf9: esp_system_abort at /home/nikhi/esp/v5.2.1/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x40387f69: __assert_func at /home/nikhi/esp/v5.2.1/esp-idf/components/newlib/assert.c:81
0x4038383a: vTaskPriorityDisinheritAfterTimeout at /home/nikhi/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5160 (discriminator 1)
0x40381720: xQueueSemaphoreTake at /home/nikhi/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1842
0x40387d28: esp_log_impl_lock_timeout at /home/nikhi/esp/v5.2.1/esp-idf/components/log/log_freertos.c:48
0x420a1d72: esp_log_writev at /home/nikhi/esp/v5.2.1/esp-idf/components/log/log.c:207
0x40387cb5: esp_log_write at /home/nikhi/esp/v5.2.1/esp-idf/components/log/log.c:225
0x42009da6: gatts_profile_a_event_handler(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/nikhi/disk/playcomputer/main/pc_ble.cpp:412 (discriminator 1)
0x42009966: gatts_event_handler(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/nikhi/disk/playcomputer/main/pc_ble.cpp:638
0x42024b06: btc_gatts_cb_to_app at /home/nikhi/esp/v5.2.1/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:40
0x420255e9: btc_gatts_cb_handler at /home/nikhi/esp/v5.2.1/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:808
0x4204b271: btc_thread_handler at /home/nikhi/esp/v5.2.1/esp-idf/components/bt/common/btc/core/btc_task.c:277
0x4204d77b: osi_thread_run at /home/nikhi/esp/v5.2.1/esp-idf/components/bt/common/osi/thread.c:165
0x4038185d: vPortTaskWrapper at /home/nikhi/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

[test.txt](https://github.com/espressif/esp-idf/files/15377384/test.txt)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/15377390/sdkconfig.txt)

### Steps to reproduce.

1. Create n example project with gatt server
2. Add the condition mentioned on the main.c
3. Send the bytes through NRF connect app

### Debug Logs.

```plain
assert failed: vTaskPriorityDisinheritAfterTimeout tasks.c:5160 (pxTCB->uxMutexesHeld)

Backtrace: 0x40375a56:0x3fcb5790 0x40380cf9:0x3fcb57b0 0x40387f69:0x3fcb57d0 0x4038383a:0x3fcb58f0 0x40381720:0x3fcb5910 0x40387d28:0x3fcb5950 0x420a1d72:0x3fcb5970 0x40387cb5:0x3fcb59a0 0x42009da6:0x3fcb59f0 0x42009966:0x3fcb5ca0 0x42024b06:0x3fcb5cd0 0x420255e9:0x3fcb5cf0 0x4204b271:0x3fcb5d40 0x4204d77b:0x3fcb5d60 0x4038185d:0x3fcb5d90
0x40375a56: panic_abort at /home/nikhi/esp/v5.2.1/esp-idf/components/esp_system/panic.c:472
0x40380cf9: esp_system_abort at /home/nikhi/esp/v5.2.1/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x40387f69: __assert_func at /home/nikhi/esp/v5.2.1/esp-idf/components/newlib/assert.c:81
0x4038383a: vTaskPriorityDisinheritAfterTimeout at /home/nikhi/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:5160 (discriminator 1)
0x40381720: xQueueSemaphoreTake at /home/nikhi/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1842
0x40387d28: esp_log_impl_lock_timeout at /home/nikhi/esp/v5.2.1/esp-idf/components/log/log_freertos.c:48
0x420a1d72: esp_log_writev at /home/nikhi/esp/v5.2.1/esp-idf/components/log/log.c:207
0x40387cb5: esp_log_write at /home/nikhi/esp/v5.2.1/esp-idf/components/log/log.c:225
0x42009da6: gatts_profile_a_event_handler(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/nikhi/disk/playcomputer/main/pc_ble.cpp:412 (discriminator 1)
0x42009966: gatts_event_handler(esp_gatts_cb_event_t, unsigned char, esp_ble_gatts_cb_param_t*) at /home/nikhi/disk/playcomputer/main/pc_ble.cpp:638
0x42024b06: btc_gatts_cb_to_app at /home/nikhi/esp/v5.2.1/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:40
0x420255e9: btc_gatts_cb_handler at /home/nikhi/esp/v5.2.1/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:808
0x4204b271: btc_thread_handler at /home/nikhi/esp/v5.2.1/esp-idf/components/bt/common/btc/core/btc_task.c:277
0x4204d77b: osi_thread_run at /home/nikhi/esp/v5.2.1/esp-idf/components/bt/common/osi/thread.c:165
0x4038185d: vPortTaskWrapper at /home/nikhi/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### More Information.

If i enable the WDT and give a delay of  vTaskDelay(1) it works fine but for my requirement the loop should run without any delay.

Dazza0, 2024-05-23:
@MR-VENOM Hard to tell what is happening here. Could you enable GDB stub on panic and by enabling the [`CONFIG_ESP_SYSTEM_PANIC_GDBSTUB`](https://docs.espressif.com/projects/esp-idf/en/v5.2.1/esp32/api-reference/kconfig.html#config-esp-system-panic) option and dumping some variable values:

- Navigate to the `vTaskPriorityDisinheritAfterTimeout()` frame and call `p pxTCB`. This print the information of the task holding the mutex
- Navigate to the `xQueueSemaphoreTake()` frame and call `p pxQueue`

Dazza0, 2024-05-29:
@MR-VENOM I had a brief look at your code, I think what's happening is the following:

```c
void test_task1(void *args)
{
    while (1)
    {
        printf("Task1 running \n");
    }
    
}

static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param) {
    ...
    vTaskDelete(task_main1);
}
```
When the `task_main1` task calls `printf()`, it will internally hold a mutex while while a log is being output. Concurrently in a separate thread, `gatts_profile_a_event_handler()` deletes the `task_main1` task while `task_main1` is still holding the mutex. As a result, on the next call to `printf()`, we crash due to the mutex having a corrupted state.

Before you delete a task, please ensure that all locks/mutexes/resources used by a task are released.

closed 2024-06-12 by Dazza0