espressif/esp-idf #13825: TG1WDT_SYS_RST

gl-chenxiaosheng, 2024-05-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

tag-v5.1.2-482a8fb

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Device work fine. Not report TG1WDT_SYS_RST. 

### What is the actual behavior?

When I frequently read and write NVS, I will read the log information stored in the flash during this period. This will result in the following error message, I am unable to locate the specific location where the error occurred, and it will give me the impression that the memory has been damaged.

```shell
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x420047d3
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1764
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2e4c
entry 0x403c9908
I (31) boot: ESP-IDF v5.1.2-1-g2714a8ad0e-dirty 2nd stage bootloader
I (31) boot: compile time May 21 2024 16:58:20
I (32) boot: Multicore bootloader
I (36) boot: chip revision: v0.2
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 16MB
W (54) boot.esp32s3: PRO CPU has been reset by WDT.
W (60) boot.esp32s3: APP CPU has been reset by WDT.
I (65) boot: Enabling RNG early entropy source...
```

### Steps to reproduce.

1. frequently read and write NVS
2. read the log information stored in the flash
3. restart and retry it
...

Dazza0, 2024-05-22:
@gl-chenxiaosheng 

Could you decode the address pointed to by `Saved PC:0x420047d3`

Do you any sample code we can run to recreate this issue?

gl-chenxiaosheng, 2024-05-22:
decode 0x420047d3:  panic_handler at /home/helpsen/esp/esp-idf/components/esp_system/port/panic_handler.c:145 (discriminator 3)

It's hard to determine what caused this, I wrote sample and read and wrote NVS and read the log in flash without replicating it. At first I thought it was all this. 

Our program has an HTTP interface for mobile app calls, and there is a chance that frequent import configurations (written to NVS) , rebooting the device, and looking at the log in flash will cause this problem.

gl-chenxiaosheng, 2024-05-23:
sometime got coredump as follow:

0x42004ad1: frame_to_panic_info at panic_handler.c:?
0x4037604c: xt_highint4 at /home/helpsen/esp/esp-idf/components/esp_system/port/soc/esp32s3/highint_hdl.S:108
0x00040022: ?? ??:0

Dazza0, 2024-05-29:
@gl-chenxiaosheng

Given that the `0x420047d3` points to  ` panic_handler()`, I think whats happening is that there is some initial error that triggers the panic handlers, but the panic handler itself get stuck or runs into another error. 

Could you try enabling `CONFIG_ESP_PANIC_HANDLER_IRAM` and see if some register dump is given on the initial panic?

gl-chenxiaosheng, 2024-06-05:
There different error in same firmware.  Can you give some suggestions?

```c
0x4037626e: panic_abort at /home/helpsen/esp/esp-idf/components/esp_system/panic.c:452
0x4038275d: esp_system_abort at /home/helpsen/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x4038da7d: __assert_func at /home/helpsen/esp/esp-idf/components/newlib/assert.c:81
0x403899a8: block_trim_free at /home/helpsen/esp/esp-idf/components/heap/tlsf/tlsf.c:496
 (inlined by) block_prepare_used at /home/helpsen/esp/esp-idf/components/heap/tlsf/tlsf.c:579
 (inlined by) tlsf_malloc at /home/helpsen/esp/esp-idf/components/heap/tlsf/tlsf.c:1005
0x403892c2: multi_heap_malloc_impl at /home/helpsen/esp/esp-idf/components/heap/multi_heap.c:207
0x40376b69: heap_caps_malloc_base at /home/helpsen/esp/esp-idf/components/heap/heap_caps.c:176
0x40376c17: heap_caps_malloc_default at /home/helpsen/esp/esp-idf/components/heap/heap_caps.c:233
0x4038da8d: malloc at /home/helpsen/esp/esp-idf/components/newlib/heap.c:24
0x4209ba9d: mem_malloc at /home/helpsen/esp/esp-idf/components/lwip/lwip/src/core/mem.c:209
0x4038b423: pbuf_alloc at /home/helpsen/esp/esp-idf/components/lwip/lwip/src/core/pbuf.c:284
0x420a0eca: tcp_pbuf_prealloc at /home/helpsen/esp/esp-idf/components/lwip/lwip/src/core/tcp_out.c:263 (discriminator 2)
0x420a14ce: tcp_write at /home/helpsen/esp/esp-idf/components/lwip/lwip/src/core/tcp_out.c:610
0x421366a6: lwip_netconn_do_writemore at /home/helpsen/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c:1794
0x4038a790: lwip_netconn_do_write at /home/helpsen/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c:1921 (discriminator 2)
0x4213b27d: tcpip_thread_handle_msg at /home/helpsen/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:162
0x4038c2ab: tcpip_thread at /home/helpsen/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
0x40385c7d: vPortTaskWrapper at /home/helpsen/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

```c
0x4037626e: panic_abort at /home/helpsen/esp/esp-idf/components/esp_system/panic.c:452
0x4038275d: esp_system_abort at /home/helpsen/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x4038da7d: __assert_func at /home/helpsen/esp/esp-idf/components/newlib/assert.c:81
0x403899a8: block_trim_free at /home/helpsen/esp/esp-idf/components/heap/tlsf/tlsf.c:496
 (inlined by) block_prepare_used at /home/helpsen/esp/esp-idf/components/heap/tlsf/tlsf.c:579
 (inlined by) tlsf_malloc at /home/helpsen/esp/esp-idf/components/heap/tlsf/tlsf.c:1005
0x403892c2: multi_heap_malloc_impl at /home/helpsen/esp/esp-idf/components/heap/multi_heap.c:207
0x40376b69: heap_caps_malloc_base at /home/helpsen/esp/esp-idf/components/heap/heap_caps.c:176
0x40376c17: heap_caps_malloc_default at /home/helpsen/esp/esp-idf/components/heap/heap_caps.c:233
0x4038daf4: _calloc_r at /home/helpsen/esp/esp-idf/components/newlib/heap.c:65
0x4038db18: calloc at /home/helpsen/esp/esp-idf/components/newlib/heap.c:29
0x420097cf: esp_vfs_select at /home/helpsen/esp/esp-idf/components/vfs/vfs.c:989
0x420b514a: base_poll_write at /home/helpsen/esp/esp-idf/components/tcp_transport/transport_ssl.c:190 (discriminator 3)
0x4212d48d: esp_transport_poll_write at /home/helpsen/esp/esp-idf/components/tcp_transport/transport.c:164
0x420b5659: tcp_write at /home/helpsen/esp/esp-idf/components/tcp_transport/transport_ssl.c:231
0x4212d44d: esp_transport_write at /home/helpsen/esp/esp-idf/components/tcp_transport/transport.c:148
0x4202d8ac: esp_mqtt_write at /home/helpsen/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:670
0x4202f873: esp_mqtt_client_publish at /home/helpsen/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:2086
0x4200ed25: gl_mqtt_client_publish at /home/helpsen/temp/gl-iot-esp-sdk/applications/s20_ble_gateway/main/gl_mqtt/src/gl_mqtt.c:56
0x4202b3ed: gl_evt_repo_data_timer_callback at /home/helpsen/temp/gl-iot-esp-sdk/applications/s20_ble_gateway/main/gl_ble/src/gl_ble_filter.c:500
0x4200889e: timer_process_alarm at /home/helpsen/esp/esp-idf/components/esp_timer/src/esp_timer.c:451
0x420088f1: timer_task at /home/helpsen/esp/esp-idf/components/esp_timer/src/esp_timer.c:477 (discriminator 1)
0x40385c7d: vPortTaskWrapper at /home/helpsen/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

```

ESP-Marius, 2024-07-08:
tlsf related errors are usually do to heap corruption see: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/heap_debug.html#heap-corruption-detection for debugging advice

ESP-Marius, 2024-08-08:
Closing due to lack of activity. Feel free to re-open if you still suspect there is an issue in the esp-idf code.

closed 2024-08-08 by ESP-Marius