espressif/esp-idf #13826: "NimBLE: error setting advertisement data; rc=8" issue while using spp_server code from esp-idf/examples

BestBot7, 2024-05-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have taken the spp_server template code of nimble stack from esp-idf/examples. 

I have build and flashed on esp32c3 to verify BLE advertisement and connection feature. 
The number of BLE concurrent connections is set to 5 in the component config under menuconfig BLE feature. 
The extended advertising under Enable BLE 5 feature is also enabled with number of instances to 4. 

I have tried with esp-idf version 5.2.1, 5.1.1, and 5.1.2 and got the error saying **"NimBLE: error setting advertisement data; rc=8"**. What exactly is this error and how to resolve this?

rahult-github, 2024-05-21:
Hi @BestBot7  , Please disable BT_NIMBLE_EXT_ADV and use.

BestBot7, 2024-05-22:
Hi @rahult-github, I have tried both enabling and disable and ended up with the same error.

rahult-github, 2024-05-22:
Hi @BestBot7 ,

I quickly tried on esp32c3 on 5.2.1 and i see it working ok. 

> I (234) cpu_start: App version:      v5.2.1-dirty
I (240) cpu_start: Compile time:     May 22 2024 10:37:56
I (246) cpu_start: ELF file SHA256:  3d90e430b...
I (251) cpu_start: ESP-IDF:          v5.2.1-dirty
I (256) cpu_start: Min chip rev:     v0.3
I (261) cpu_start: Max chip rev:     v1.99 
I (266) cpu_start: Chip rev:         v0.3
I (271) heap_init: Initializing. RAM available for dynamic allocation:
I (278) heap_init: At 3FC97020 len 00028FE0 (163 KiB): RAM
I (284) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (291) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (298) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (305) spi_flash: detected chip: generic
I (309) spi_flash: flash io: dio
W (313) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (326) sleep: Configure to isolate all GPIO pins in sleep state
I (333) sleep: Enable automatic switching of GPIO sleep configuration
I (340) coexist: coex firmware version: 77cd7f8
I (345) coexist: coexist rom version 9387209
I (350) main_task: Started on CPU0
I (350) main_task: Calling app_main()
I (360) BLE_INIT: BT controller compile version [30b57c4]
I (360) BLE_INIT: Bluetooth MAC: 7c:df:a1:a2:9b:f2
I (370) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I (410) NimBLE: BLE server UART_task started
I (410) NimBLE: BLE Host Task Started
I (410) NimBLE: GAP procedure initiated: stop advertising.
I (410) NimBLE: Device Address: 
I (410) NimBLE: 7c:df:a1:a2:9b:f2
I (420) NimBLE: 
I (420) NimBLE: GAP procedure initiated: advertise; 
I (430) NimBLE: disc_mode=2
I (430) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (440) NimBLE: 
I (440) main_task: Returned from app_main()

Can you please help share your sdkconfig ?

rahult-github, 2024-05-22:
Hi @BestBot7  

> I have taken the spp_server template code of nimble stack from esp-idf/examples

I have assumed you are running the ble_spp/spp_server example from IDF. If not, then can you please try the IDF example to confirm it works ?

rahult-github, 2024-05-22:
Hi @BestBot7 , 

In sdkconfig shared: 

> CONFIG_BT_NIMBLE_EXT_ADV=y 

Extended Advertising is enabled. 

> Please disable BT_NIMBLE_EXT_ADV and use.  

Please do this.

rahult-github, 2024-05-22:
The way to disable EXT_ADV is :

Component config → Bluetooth → NimBLE Options → Enable BLE 5 feature   and ensure Enable extended advertising is not selected

BestBot7, 2024-05-23:
I have disabled the extended advertisement and here are the error log image and the corresponding sdkconfig file attached.
[image]
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/15412662/sdkconfig.zip)

rahult-github, 2024-05-23:
Hi @BestBot7 , your original issue was advertising data not being set because ext adv was enabled. As seen in above logs, that issue is resolved.

But now a new issue is observed on your side where attempt to enable advertising fails since advertising is already in progress.  I will try to reproduce at my end and update

BestBot7, 2024-05-23:
Hi @rahult-github, this is the template code from esp32c3 examples. I have changed nothing in it, except that the number of concurrent connections set to 5. My aim is to verify the multiconnection in esp32c3 bluetooth and hence I configured the number of concurrent connections to 5. After connecting and disconnecting from a single device, the error is logged as I mentioned. So, please help me with what all should I be careful while enabling multiple BLE connections.

BestBot7, 2024-06-12:
Hi @rahult-github, Even after using the example code ble_multi_conn, the issue still persists. I have disabled the ext adv and tried and found no difference. On boot we don't get this error. Only on disconnecting after connection we find this log. Can you suggest some way on how to use this feature properly?

BestBot7, 2024-06-24:
Hi @rahult-github can you suggest some other way to resolve it?

rahult-github, 2024-06-24:
Hello @BestBot7  , can you share your complete example code changes with me . I can try to check your code and see what may be causing this behaviour.

BestBot7, 2024-06-24:
[main.zip](https://github.com/user-attachments/files/15949799/main.zip)

Here is the code file that I am trying to flash and monitor the output. 

Also attaching the image which has the error log output.

[image]

rahult-github, 2024-06-24:
Hi @BestBot7 , quickly ran with the code, and i see it working as expected: 

[image]

as seen in image above , it advertised, a device connected and then it started advertising again .  

Attached is my sdkconfig file. I still feel that EXT_ADV is enabled at your end .You can replace your example's sdkconfig with this attached sdkconfig and check . 

[sdkconfig.txt](https://github.com/user-attachments/files/15949956/sdkconfig.txt)

Please ensure to rename sdkconfig.txt to sdkconfig in your setup

BestBot7, 2024-06-24:
static void ble_spp_server_advertise(void)
{
    struct ble_gap_adv_params adv_params;
    struct ble_hs_adv_fields fields;
    const char *name;
    bool  device_connected = false;
    int rc;

    /** Advance Advertising data */
    icm_gap_adv_UpdateAdv();

    /** Scan  response advertisement data */
    /**
     *  Set the advertisement data included in our advertisements:
     *     o Advertising tx power.
     *     o Device name.
     */

    memset(&fields, 0, sizeof fields);

    name = ble_svc_gap_device_name();
    fields.name = (uint8_t *)name;
    fields.name_len = strlen(name);
    /* Indicate that the TX power level field should be included; have the
     * stack fill this value automatically.  This is done by assigning the
     * special value BLE_HS_ADV_TX_PWR_LVL_AUTO.
     */
    fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;
    fields.tx_pwr_lvl_is_present = 1;
    fields.name_is_complete = 1;

    rc = ble_gap_adv_rsp_set_fields(&fields);
    if (rc != 0) {
        ESP_LOGE(g_log_ble_tag, "Line:%d, %s, Failed with error: %s",__LINE__, __func__, esp_err_to_name(rc));
        return;
    }

    /* Begin advertising. */
    ESP_LOGI(g_log_ble_tag, "Line:%d, %s, Begin Advertising",__LINE__, __func__);
    ESP_LOGI(g_log_ble_tag, "Line:%d, %s, Device Name [%s]", __LINE__, __func__,name);
    memset(&adv_params, 0, sizeof adv_params);
    adv_params.conn_mode   = BLE_GAP_CONN_MODE_UND;
    adv_params.disc_mode   = BLE_GAP_DISC_MODE_GEN;
    adv_params.channel_map = BLE_GAP_ADV_DFLT_CHANNEL_MAP;
    adv_params.filter_policy = BLE_HCI_ADV_FILT_NONE;
    rc = ble_gap_adv_start(OwnAddrType, NULL, BLE_HS_FOREVER,
                           &adv_params, ble_spp_server_gap_event, NULL);

    if (rc != 0) {
        ESP_LOGW(g_log_ble_tag, "Line:%d, %s, Error Enabling advertisement procedure. rc:%d. "
                                " NOTE: This error also appears in esp-idf template code"
                              , __LINE__, __func__, rc);
    }

    for (int idx = 1; idx <= CONFIG_BT_NIMBLE_MAX_CONNECTIONS; idx++) {

        if (g_conn_handle_subs[idx] == true)
        {
            device_connected = true;
            break;
        }
    }
}

This is the function in my code where the error is popping. Please let me know if you need other parts of the code.

rahult-github, 2024-06-24:
Hi @BestBot7  , 

I took the source code main.zip you shared [here](https://github.com/espressif/esp-idf/issues/13826#issuecomment-2185583643)  and ran the same at my end for which output was as [above](https://github.com/espressif/esp-idf/issues/13826#issuecomment-2186091343)  

Your main.zip also has example code of ble_multi_conn_prph .  The comment [here](https://github.com/espressif/esp-idf/issues/13826#issuecomment-2186091343) has a code which was not present in main.zip file you shared.  

So is there a different application ?. I believe different codes are causing confusion. So, 

1. Give me one single piece of your application code that i can try at my end.
2. Please also share your sdkconfig. 

I can re-run your application at my end and see if it has any discrepancy.

rahult-github, 2025-03-04:
Closing issue. Feel free to reopen in case of any update

closed 2025-03-04 by rahult-github