espressif/esp-idf #13828: ESP-IDF V4.4.6 - Content in NVS erasing when ESP32 resets

FrancoBisciglia, 2024-05-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.6

### Espressif SoC revision.

ESP32-WROVER-E

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Greetings. I'm currently using a custom board that has a ESP32-WROVER-E module. In this board, I have a button that acts as a "factory reset", which means that if I hold it for 3 seconds, some variables are reset to their default value, and then saved to NVS flash. Furthermore, if I continue to hold the button for 10 seconds in total, the program resets via a "esp_restart()" call.

When the program starts, it loads these previously saved variables from NVS flash. 

### What is the actual behavior?

Now, regarding the issue, what happens is the following:

1) I hold the button for 3 seconds, so that the variables are reset to their default value.

2) I continue to hold it for another 7 seconds, so the program resets via an "esp_restart()".

3) When the program starts again, instead of their default values, all the variables are now erased to 0.

This only happens if I follow that specific order of events. For example, if I press the button for 3 seconds, so that the variables are reset to their default values, but then I relase the button and turn off and on the board power supply, the variables are loaded correctly.

Additionally, I have tried the ESP-IDF NVS example, that just modifies and loads a numeric variable in NVS flash, and then restarts with an "esp_restart()" call, and in that case, the variable doesn't reset to 0.

### Steps to reproduce.

1) I hold the button for 3 seconds, so that the variables are reset to their default value.

2) I continue to hold it for another 7 seconds, so the program resets via an "esp_restart()".

3) When the program starts again, instead of their default values, all the variables are now erased to 0.

In a generic form, this will be something like:

1) Save a variable in NVS flash.

2) Call "esp_restart()".

3) When the program starts again, load that variable from NVS flash.

Here is an example of a function that I use to save one of those variables in flash:

```
void flash_save_login()
{
	ESP_LOGI(TAG, "%s", __FUNCTION__);
	
	lock_flash();
	
	nvs_handle_t nvsHandle;
	nvs_open(CONFIG_NAMESPACE, NVS_READWRITE, &nvsHandle);

	nvs_set_str(nvsHandle, KEY_LOGIN, str(HR_LOGIN));
	
	nvs_commit(nvsHandle);
	nvs_close(nvsHandle);
	
	unlock_flash();
}
```

This is an example of a function that I use to load that variable from flash:

```
void flash_load_configuration()
{
	ESP_LOGI(TAG, "%s", __FUNCTION__);
	
	lock_flash();
	
	nvs_stats_t nvs_stats;
	nvs_get_stats(NULL, &nvs_stats);
	ESP_LOGW(TAG, "UsedEntries = (%d), FreeEntries = (%d), AllEntries = (%d)",
			nvs_stats.used_entries, nvs_stats.free_entries, nvs_stats.total_entries);
	
	nvs_handle_t nvsHandle;
	size_t size = 0;
	nvs_open(CONFIG_NAMESPACE, NVS_READWRITE, &nvsHandle);

	// Security code
	size = SIZE_LOGIN * 2;
	esp_err_t err_nvs = nvs_get_str(nvsHandle, KEY_LOGIN, str(HR_LOGIN), &size);
	
	nvs_close(nvsHandle);	
	
	unlock_flash();
}
```

The "lock_flash()" and "unlock_flash()" functions just use an FreeRTOS Semaphore.

This is the FreeRTOS task that controls the program reset via the "factory reset" button:

```
void io_task(void* pvParameters)
{
	int64_t config_factory_reset_timer = set_timeout(3000);
	int64_t firmware_factory_reset_timeout = set_timeout(10000);

	while (1)
	{
		uint32_t result = ulTaskNotifyTake(pdTRUE, 0);

		if (!get_io(FIRM_FACT))
		{
			if (check_timeout(config_factory_reset_timer))
			{
				factory_default();
				flash_save_configuration();

				vTaskSuspend(_led_task_handle);
				for (int i = 0; i < 10; i++)
				{
					set_io(GREEN_LED, !get_io(GREEN_LED));
					vTaskDelay(pdMS_TO_TICKS(100));
				}
				vTaskResume(_led_task_handle);
				
				config_factory_reset_timer = set_timeout(10000);
			}

			if (check_timeout(firmware_factory_reset_timeout))
			{
				ESP_LOGW("rtu_safe_restart", "Firmware factory reset");
				rtu_safe_restart();
			}
		}
		else
		{
			config_factory_reset_timer = set_timeout(3000);
			firmware_factory_reset_timeout = set_timeout(10000);
		}

		vTaskDelay(pdMS_TO_TICKS(50));
	}
}
```

### Debug Logs.

```plain
This is what I get when the program restarts:

I (00:00:24.893) i2c: i2c_device_detect
I (00:00:24.898) i2c: 0x20 found
I (00:00:24.916) main: I/O Expander: OK
I (00:00:24.922) main: NETIF: OK
I (00:00:24.925) main: EVENT LOOP: OK
I (00:00:24.929) flash: init_flash
I (00:00:25.155) flash: flash_load_configuration
W (00:00:25.156) flash: UsedEntries = (0), FreeEntries = (10080), AllEntries = (10080)
I (00:00:25.170) flash: ERR_NVS: ESP_ERR_NVS_NOT_FOUND
I (00:00:25.171) flash: HR_LOGIN:
```

Notice that "UsedEntries" is 0, I got an "ESP_ERR_NVS_NOT_FOUND" when calling "nvs_get_str()", and that the variable "HR_LOGIN" is empty.
```

rrtandler, 2024-05-29:
Hi @FrancoBisciglia ,
Could you please post the definition of `rtu_safe_restart` function as well as the code, where you call `nvs_flash_init` ?

FrancoBisciglia, 2024-05-29:
Hi @rrtandler 

Of course, here's the code of the "rtu_safe_restart" function:

```
void rtu_safe_restart()
{
	ESP_LOGW(__FUNCTION__, "Rebooting..");
	lock_log();
	lock_flash();
	esp_restart();
}
```

Where the functions "lock_log()" and "lock_flash()" are the following:

```
void lock_log()
{
	xSemaphoreTakeRecursive(_log_mutex, portMAX_DELAY);
}

void lock_flash()
{
	xSemaphoreTakeRecursive(_flash_mutex, portMAX_DELAY);
}
```

In addition, here's the code where I call the function "nvs_flash_init()":

```
esp_err_t init_flash()
{
	ESP_LOGI(TAG, "%s", __FUNCTION__);
	
	esp_err_t err = nvs_flash_init();
	
	if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND)
	{
        // NVS partition was truncated and needs to be erased
		ESP_LOGW(TAG, "nvs partition erased");
        nvs_flash_erase();
        nvs_flash_init();
    }
	
	_flash_mutex = xSemaphoreCreateRecursiveMutex();

	if (get_io(FIRM_FACT) == 0)
	{
		ESP_LOGW(TAG, "factory default pressed");
		factory_default();
		flash_save_configuration();
	}
	else
	{
		flash_load_configuration();
	}

	return ESP_OK;
}
```

Please, let me know if you need anything else. Thank you for your help.

rrtandler, 2024-05-29:
@FrancoBisciglia  thanks for your answer. From the code you have provided, I cannot see direct cause of the problem. Perhaps you can also post definition of `set_timeout()` `check_timeout()` and the `str()` used in statement `esp_err_t err_nvs = nvs_get_str(nvsHandle, KEY_LOGIN, str(HR_LOGIN), &size);`

FrancoBisciglia, 2024-05-29:
Hi @rrtandler .

Regarding those function you mention, yes, of course I can post the code. But before that, I will mention that the functions "set_timeout()" and "check_timeout()" are not related at all with the NVS flash. They are just used as a pseudo-timer, based on the ESP-IDF function "esp_timer_get_time()".

Anyway, here are the respective definitions of those functions:

```
#define set_timeout(time)		MS_TIMER + time
#define MS_TIMER				(esp_timer_get_time() / 1000)

#define check_timeout(time)		(((int64_t)(MS_TIMER - (time))) >= 0)

#define str(s)		((char*)&s)
```

The meaning of the "str()" function is related to the definition of the "HR_LOGIN" type of variables. This type of variables are defined as follows:

```
extern EXT_RAM_ATTR int16_t HoldingRegistersProtected[SIZE_MODBUS_HOLDING_REGISTERS_PROTECTED];
#define HR_LOGIN HoldingRegistersProtected[4453]
```

They are just specific cells of the global array "HoldingRegistersProtected", which is of type "EXT_RAM_ATTR".

Let me know if you need anything else. Thank you.

rrtandler, 2024-05-30:
Thanks @FrancoBisciglia ,
Is the variable holding the name of NVS key i.e. `KEY_LOGIN` stored also in the `EXT_RAM_ATTR` prefixed memory location ?
For better insight into what happens in NVS, you can selectively raise the log level of NVS component. Put the `esp_log_level_set("nvs", ESP_LOG_VERBOSE);`before the first `nvs_init(...)` call and please publish the logs after good and bad case. 
For reference, how to use the selective logging, see [Logging library](https://docs.espressif.com/projects/esp-idf/en/v4.4.7/esp32/api-reference/system/log.html)

Cicrex, 2024-05-31:
Hi @rrtandler 

Regardin "KEY_LOGIN", that's correct, as you can se in this definitions I posted before:

```
extern EXT_RAM_ATTR int16_t HoldingRegistersProtected[SIZE_MODBUS_HOLDING_REGISTERS_PROTECTED];
#define HR_LOGIN HoldingRegistersProtected[4453]
```

"HR_LOGIN" is defined as a specific index of the array "HoldingRegistersProtected", which is of type "EXT_RAM_ATTR".

Regardin the NVS, great, as you said, I will try to change the log level of the NVS component with "esp_log_level_set()", and then I will comment again with the corresponding results.

Thank you.

rrtandler, 2024-06-03:
Hi @Cicrex,

I see some misunderstanding in your answer. I was asking for `KEY_LOGIN` declaration while you have re-confirmed the declaration of `HR_LOGIN`. Could you please post the declaration of `KEY_LOGIN` ?
Thanks,

Radek Tandler

FrancoBisciglia, 2024-06-03:
Hi @rrtandler 

Firstly, I apologize, I made that last comment with my other account.

Secondly, you are right, I missread the word and I posted the definition of "HR_LOGIN". The definition of "KEY_LOGIN" is the following:

```
#define KEY_LOGIN "LOGIN"
```

As I said, as soon as I have the time, I will try what you suggested. Thank you.

FrancoBisciglia, 2024-06-05:
Hi @rrtandler 

I have tried your suggestion of raising the log level of the NVS flash with "esp_log_level_set("nvs", ESP_LOG_VERBOSE);". The result is similar to what I posted in the first comment, but here they are:

**Bad case**

```
I (19:00:52.170) main: I/O Expander: OK
I (19:00:52.176) main: NETIF: OK
I (19:00:52.179) main: EVENT LOOP: OK
I (19:00:52.183) flash: init_flash
I (19:00:52.416) flash: flash_load_configuration
W (19:00:52.417) flash: UsedEntries = (0), FreeEntries = (10080), AllEntries = (10080)
I (19:00:52.424) flash: ERR_NVS: ESP_ERR_NVS_NOT_FOUND
I (19:00:52.426) flash: HR_LOGIN:
E (19:00:52.430) flash: DEBOUNCE(0): 0
I (19:00:52.434) main: FLASH MEMORY: OK
I (19:00:56.431) io: init_io
```

**Good case**

```
I (00:00:00.168) flash: init_flash
I (00:00:00.452) flash: flash_load_configuration
W (00:00:00.453) flash: UsedEntries = (308), FreeEntries = (9772), AllEntries = (10080)
I (00:00:00.509) flash: ERR_NVS: ESP_OK
I (00:00:00.509) flash: HR_LOGIN: Datatress
E (00:00:00.510) flash: DEBOUNCE(0): 50
I (00:00:00.512) main: FLASH MEMORY: OK
I (00:00:04.516) io: init_io
```

As you can see, in the bad case, the "UsedEntires" parameter is 0 and the password variable "HR_LOGIN" is empty. On the other hand, in the good case, that same parameter has a value of 308, and "HR_LOGIN" is filled with the password string.

Please let me know if I can do anything else. Thank you.