espressif/esp-idf #13830: DHCP fails to get IP with Personal/Enterprise WiFi

tomasznowik, 2024-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-wroom-32

### Power Supply used.

USB

### What is the expected behavior?

DHCP works always and with any Personal and Enterprise WiFi

### What is the actual behavior?

DHCP sends discovery requests and we see it with wireshark but they are ignored by the router.
We are able to recreate it by switching between Personal and Enterprise network on Mist but it always happens on Aruba router with Enterprise PEAP network.
We didn't notice this issue on Rockus yet.
Cellphones on the same time has no issues connecting these networks.

### Steps to reproduce.

Just connect to Enterprise PEAP wifi created by Aruba router
or 
Switch between two networks Personal and Enterprise PEAP created by one Mist router 

### Debug Logs.

```plain
I 24392 | WiFi             Starting connection
I 24422 | wifi             new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I 24422 | wifi             state: init -> auth (b0)
I 24462 | wifi             state: auth -> assoc (0)
I 24462 | wifi             state: assoc -> run (10)
I 24552 | wifi             connected with it_personal_x, aid = 3, channel 11, BW20, bssid = 5c:5b:35:01:f7:18
I 24552 | wifi             security: WPA2-PSK, phy: bgn, rssi: -43
I 24562 | wifi             pm start, type: 1
W None |dhcp_start(netif=0x3ffcefb4) st2
W None |dhcp_start(): restarting DHCP configuration
W None |dhcp_start(): starting DHCP configuration
W None |dhcp_discover()
W None |dhcp_discover(): dhcp state is DISCOVER
W None |dhcp_select: could not allocate DHCP request
W None |transaction id xid(dc7b4519)
W None |dhcp_discover: making request
W None |dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
W None |dhcp_discover: deleting()ing
W None |dhcp_discover: SELECTING
W None |dhcp_discover(): set request timeout 500 msecs
I 24602 | WiFi             Connected
I 24612 | wifi             AP's beacon interval = 102400 us, DTIM period = 2
W None |dhcp_coarse_tmr()
W None |dhcp_fine_tmr(): request timeout
W None |dhcp_timeout()
W None |dhcp_timeout(): restarting discovery
W None |dhcp_discover()
W None |dhcp_discover(): dhcp state is DISCOVER
W None |dhcp_select: could not allocate DHCP request
W None |transaction id xid(dc7b4519)
W None |dhcp_discover: making request
W None |dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
W None |dhcp_discover: deleting()ing
W None |dhcp_discover: SELECTING
W None |dhcp_discover(): set request timeout 1000 msecs
W None |dhcp_coarse_tmr()
W None |dhcp_fine_tmr(): request timeout
W None |dhcp_timeout()
W None |dhcp_timeout(): restarting discovery
W None |dhcp_discover()
W None |dhcp_discover(): dhcp state is DISCOVER
W None |dhcp_select: could not allocate DHCP request
W None |transaction id xid(dc7b4519)
W None |dhcp_discover: making request
W None |dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
W None |dhcp_discover: deleting()ing
W None |dhcp_discover: SELECTING
W None |dhcp_discover(): set request timeout 2000 msecs
W None |dhcp_coarse_tmr()
W None |dhcp_coarse_tmr()
W None |dhcp_fine_tmr(): request timeout
W None |dhcp_timeout()
W None |dhcp_timeout(): restarting discovery
W None |dhcp_discover()
W None |dhcp_discover(): dhcp state is DISCOVER
W None |dhcp_select: could not allocate DHCP request
W None |transaction id xid(dc7b4519)
W None |dhcp_discover: making request
W None |dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
W None |dhcp_discover: deleting()ing
W None |dhcp_discover: SELECTING
W None |dhcp_discover(): set request timeout 4000 msecs
W None |dhcp_coarse_tmr()
W None |dhcp_coarse_tmr()
```

### More Information.

Changing settings to 
CONFIG_LWIP_DHCP_DISABLE_CLIENT_ID=y
CONFIG_LWIP_DHCP_DOES_ARP_CHECK=n
didn't help.

kapilkedawat, 2024-05-23:
Hi @tomasznowik, Does router ack these dhcp packets(ignored ones)?

tomasznowik, 2024-05-23:
This is old capture from WiFi directly showing that ESP32 never gets an answer for DHCP Discovery:
[image]

And that's a capture from PC with DHCP server. Looks like after switching ESP32 to personal1 WiFi the DHCP Offer is sent but ESP32 doesn't receive it:
[image]

And here's the last one - also from PC with DHCP server. Looks like after switching to other WiFi no packets from ESP32 are visible although previously ESP32 did receive IP on this WiFi SSID. And I see in ESP32 log that it sends DHCP Discovery a few times before switching to other WiFi:
[image]

tomasznowik, 2024-05-23:
Note: we have other device based on RTL chip and we didn't notice any DHCP issues with it in the same scenarios and tests.

kapilkedawat, 2024-05-24:
Hi @tomasznowik, in all these captures, esp32 is always sending dhcp request however AP is not forwarding the packets from DHCP server to esp32.

Could you please share sniffer capture of RTL device and esp32 device with the same AP so that I can analyze a bit?

tomasznowik, 2024-05-28:
@kapilkedawat This is a capture of two successful requests for ESP32 and RTL: [dhcp_issue_esp32_vs_rtl.zip](https://github.com/espressif/esp-idf/files/15469753/dhcp_issue_esp32_vs_rtl.zip)

From what I see ESP32 sends hostname and client identifier OPTIONS in DHCP Discovery package and client identifier OPTION in DHCP Request package and RTL doesn't do that. 

I will check if disabling both would help. In the past we tested CONFIG_LWIP_DHCP_DISABLE_CLIENT_ID alone but it didn't help.

AxelLin, 2024-06-05:
> Could you please share sniffer capture of RTL device and esp32 device with the same AP so that I can analyze a bit?

@kapilkedawat Did you check the log in https://github.com/espressif/esp-idf/issues/13830#issuecomment-2135133436 ? Any finding?

kapilkedawat, 2024-06-05:
Hi @tomasznowik, sorry for the delay, I missed the reply. These are both from the success case or later AP dropped these packets? Could you also attach over the air capture of the same instances?

tomasznowik, 2024-06-05:
@kapilkedawat First capture I've attached is over the air (wifi) but it's no so easy to setup the test to grab it. But I have some logs with DHCP in debug mode  where I've added DHCP client stop & start in case IP is not received in 10 sec. Maybe it will help: 
[dhcp_reset_helped.txt](https://github.com/user-attachments/files/15589396/dhcp_reset_helped.txt)

tomasznowik, 2024-10-15:
Yes, currently it's mitigated by reconnect to the wifi network and we do not debug it anymore

closed 2025-03-07 by Alvin1Zhang