espressif/esp-idf #13831: esp_ble_gattc_read_char Read 21 bytes {00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00}

shellus, 2024-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

After I connect to the peer ble and I use esp_ble_gattc_read_char without sending any data, I get the event “ESP_GATTC_READ_CHAR_EVT”

I can confirm that the opposite end is not sending any data, and the opposite end is another development board with verified ble interaction with the Android phone.

```c++
        case ESP_GATTC_READ_CHAR_EVT:
            if(param->read.status == ESP_GATT_OK){
                m_readlen = param->read.value_len;
                ESP_LOGI(LOG_TAG, "GATTDevice.cpp Read %d bytes", m_readlen);
                ESP_LOG_BUFFER_HEX(LOG_TAG, param->read.value, m_readlen);
                memcpy(m_readbuff, param->read.value, m_readlen);
                m_readbuff[m_readlen] = 0;
            }
            else{
                m_readlen = 0;
            }
            m_lock->Unlock();
            break;
```

```
outputs:
I (43311) GATTDevice: GATTDevice.cpp Read 21 bytes
I (43311) GATTDevice: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (43321) GATTDevice: 00 00 00 00 00
```

I'd like to know what the 21 bytes means.

shellus, 2024-05-22:
Another noteworthy piece of information is that
I will see these logs when I call esp_ble_gattc_write_char or  esp_ble_gattc_read_char after:
```
outputs:
W (7100) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (7100) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
W (7110) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
W (7120) GATTDevice: GATT event 0x5
W (7120) GATTDevice: ESP_GATTC_CLOSE_EVT status:0 conn_id:0 remote_bda:cd:fb:50:e6:33:fe reason:0x0
W (7130) GATTDevice: GATT event 0x29
W (7130) GATTDevice: ESP_GATTC_DISCONNECT_EVT reason:0x16 conn_id:0 remote_bda:cd:fb:50:e6:33:fe
```

shellus, 2024-05-22:
In the idf sdk I see that the esp_ble_gattc_open I'm using is defined in the esp-idf-v5.2.1\components\bt\host\bluedroid\api\esp_gattc_api.c
The #if (BLE_42_FEATURE_SUPPORT == TRUE) scope.

Does this mean I am using BLE 4.2 GATTC?
If the peer is not BLE4.2, could that be the cause of the problem?

shellus, 2024-05-23:
Resolved.
The problem with 21-bit 0 is that the slave sets these 0 in characteristic.
The reason for the disconnection may be that I did not receive the slave's notify, which caused the slave to think that it had timed out, so it actively disconnected.

So my problem is because I thought I should read char on this feature, but the slave is designed to reply to the data via notify, not read.
Knowing this, I have solved all my problems.

closed 2024-05-23 by shellus

zhp0406, 2024-05-24:
Thank you for your report. Feel free to ask if you encounter any other issues later on.