espressif/esp-idf #13832: esp_https_server: performing session handshake / mbedtls_mpi causes watchdog to be trigerred even with CONFIG_MBEDTLS_HARDWARE_MPI=n

RVC69, 2024-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2.1

### Espressif SoC revision.

chip revision: v0.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C6

### Power Supply used.

USB

### What is the expected behavior?

During esp_https_server session handshake watchdog is triggered even with CONFIG_MBEDTLS_HARDWARE_MPI=n
Expected not to be trigerred or MPI really deactivated when using sdkconfig option.

### What is the actual behavior?

Even with CONFIG_MBEDTLS_HARDWARE_MPI=n, mpi acceleration is called.

0x420d5608: mbedtls_mpi_core_mla at D:/Programs/esp-idf/v5.2.1/esp-idf/components/mbedtls/mbedtls/library/bignum_core.c:480
0x420d57de: mbedtls_mpi_core_montmul at D:/Programs/esp-idf/v5.2.1/esp-idf/components/mbedtls/mbedtls/library/bignum_core.c:545

### Steps to reproduce.

        httpd_ssl_config_t conf = HTTPD_SSL_CONFIG_DEFAULT();
        conf.servercert = cert_start;
        conf.servercert_len = cert_end - cert_start;
        conf.prvtkey_pem = privkey_start;
        conf.prvtkey_len = privkey_end - privkey_start;
        conf.httpd.lru_purge_enable = true;
        conf.httpd.stack_size = 8192;
        conf.httpd.max_open_sockets   =  16 - MQTT_BROKER_MAX_CLIENTS;
        conf.httpd.task_priority = WEBSERVER_TASK_PRIORITY;
        conf.httpd.recv_wait_timeout = 20;
        conf.httpd.send_wait_timeout = 20;
        conf.port_insecure = WEBSERVER_INSECURE_PORT;
        conf.port_secure = WEBSERVER_SECURE_PORT;       

ret = httpd_ssl_start(&server, &conf);

### Debug Logs.

```plain
I (205100) esp_https_server: performing session handshake
E (210090) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:E (210090) task_wdt:  - IDLE (CPU 0)
E (210090) task_wdt: Tasks currently running:
E (210090) task_wdt: CPU 0: httpd
E (210090) task_wdt: Print CPU 0 (current core) backtrace
Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x420d3e7a  RA      : 0x420d4020  SP      : 0x408515b0  GP      : 0x408194a0  
0x420d3e7a: mbedtls_mpi_core_mla at D:/Programs/esp-idf/v5.2.1/esp-idf/components/mbedtls/mbedtls/library/bignum_core.c:480
0x420d4020: mbedtls_mpi_core_montmul at D:/Programs/esp-idf/v5.2.1/esp-idf/components/mbedtls/mbedtls/library/bignum_core.c:543

TP      : 0x40811680  T0      : 0x40030dca  T1      : 0x00000004  T2      : 0xc9261067
0x40811680: ppCalTxAMPDULength at ??:?
0x40030dca: memset in ROM

S0/FP   : 0x03dddb19  S1      : 0x4085d8d0  A0      : 0xa78b4eeb  A1      : 0x0470d456
A2      : 0x4085e550  A3      : 0x00000000  A4      : 0xbfb0316a  A5      : 0x4085d930
A6      : 0x00000001  A7      : 0x00000001  S2      : 0x00000042  S3      : 0x0000003e
S4      : 0x00000040  S5      : 0x4085e4f0  S6      : 0x408575fc  S7      : 0x88c99fbf
S8      : 0x00000041  S9      : 0x4085d0c8  S10     : 0x4085d0c8  S11     : 0x00000002
T3      : 0xffffffff  T4      : 0x89b18cab  T5      : 0x18ae8979  T6      : 0x8c57b7ab
MSTATUS : 0x00000000  MTVEC   : 0x00000022  MCAUSE  : 0x00000001  MTVAL   : 0x0000000e
MHARTID : 0x00000002
I (211900) esp_https_server: performing session handshake
E (216760) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:E (216760) task_wdt:  - IDLE (CPU 0)
E (216760) task_wdt: Tasks currently running:
E (216760) task_wdt: CPU 0: httpd
E (216760) task_wdt: Print CPU 0 (current core) backtrace
Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x420d3e52  RA      : 0x420d4020  SP      : 0x408515b0  GP      : 0x408194a0  
0x420d3e52: mbedtls_mpi_core_mla at D:/Programs/esp-idf/v5.2.1/esp-idf/components/mbedtls/mbedtls/library/bignum_core.c:480
0x420d4020: mbedtls_mpi_core_montmul at D:/Programs/esp-idf/v5.2.1/esp-idf/components/mbedtls/mbedtls/library/bignum_core.c:543

TP      : 0x40811680  T0      : 0x40030dca  T1      : 0x00000000  T2      : 0x1c6b6a9d
0x40811680: ppCalTxAMPDULength at ??:?
0x40030dca: memset in ROM

S0/FP   : 0xc959029d  S1      : 0x408559ac  A0      : 0x1c61b279  A1      : 0x4265e42a
A2      : 0x40865724  A3      : 0x00000001  A4      : 0x80322959  A5      : 0x40855a8c
A6      : 0x00000001  A7      : 0x00000001  S2      : 0x00000042  S3      : 0x00000039
S4      : 0x00000040  S5      : 0x40865644  S6      : 0x4085ccb4  S7      : 0x88c99fbf
S8      : 0x00000041  S9      : 0x4085e9f8  S10     : 0x4085e9f8  S11     : 0x00000001
T3      : 0x00000000  T4      : 0x2301af8e  T5      : 0xb2d15ee0  T6      : 0xc15cc864
MSTATUS : 0x00000000  MTVEC   : 0x00000026  MCAUSE  : 0x00000001  MTVAL   : 0x00000004
MHARTID : 0x00000002
```

### More Information.

# CONFIG_MBEDTLS_ECP_RESTARTABLE is not set
CONFIG_MBEDTLS_CMAC_C=y
CONFIG_MBEDTLS_HARDWARE_AES=y
CONFIG_MBEDTLS_AES_USE_INTERRUPT=y
CONFIG_MBEDTLS_AES_INTERRUPT_LEVEL=0
**# CONFIG_MBEDTLS_HARDWARE_MPI is not set**
CONFIG_MBEDTLS_HARDWARE_SHA=y
CONFIG_MBEDTLS_HARDWARE_ECC=y
CONFIG_MBEDTLS_ECC_OTHER_CURVES_SOFT_FALLBACK=y
CONFIG_MBEDTLS_ROM_MD5=y

hmalpani, 2024-05-29:
Hello @RVC69 
I think the issue here is that you have set `recv_wait_timeout/send_wait_timeout` to 20. The default task watchdog timeout is set to 5 seconds. I think it is taking more than 5 seconds for connection to complete and therefore the watchdog is triggered. Can you try increasing the task watchdog timeout in menuconfig and see if the issue persists.

AxelLin, 2024-05-29:
> Hello @RVC69 I think the issue here is that you have set `recv_wait_timeout/send_wait_timeout` to 20. The default task watchdog timeout is set to 5 seconds. I think it is taking more than 5 seconds [...]

@hmalpani 
Can you reproduce this issue?
I believe above comment is wrong!
It's fine to set `recv_wait_timeout/send_wait_timeout` > watchdog timeout.

RVC69, 2024-05-29:
Thanks, it's not easy to isolate part of this code.

I have set CONFIG_TASK_WDT_TIMEOUT_S=20, removed CONFIG_MBEDTLS_HARDWARE_MPI=n, conf.httpd.recv_wait_timeout = 20; conf.httpd.send_wait_timeout = 20;

It is working now. Seems it was a watchog timeout issue
regards

AxelLin, 2024-05-30:
The default CONFIG_ESP_TASK_WDT_TIMEOUT_S=5 should work.
Otherwise, it means some task can occupy the CPU for 5 seconds, AFAICT, it's not normal.

mahavirj, 2024-07-08:
@AxelLin @RVC69 

Generally speaking the default value for task watchdog timeout is sufficient for most of the configurations, however for some cases this value may need fine tuning. (As an example) Signing operation for different cipher schemes may take long time based on the factors like CPU speed, state of hardware accelerator etc. Some performance numbers are highlighted for reference [here](https://github.com/espressif/mbedtls/wiki/Performance-Numbers).

For this particular issue, I would recommend:

1. Please highlight the scheme used to generate the private key for HTTPS server configuration
2. Please try with latest 5.2.2 release (just in case something was fixed/updated for C6 target)
3. Please supply mbedtls debug enabled logs

closed 2024-07-15 by Alvin1Zhang