espressif/esp-idf #13833: Installation does not work neither the link on Toolchain setup

yareach, 2024-05-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.7

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

The link: https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch5-win32.zip does not work.
I tried this link becuase I didn't succeed to install it from the VSC extension setup nor using the installation using the ESP-IDF Tools Installer. VSC installed it but idf.py didn't work - complaining something about packages. The installer fisnished with the ps or cmd windows. Both comlained that the env. variable for IDF was null. It was not true, the variable was set. Spent two days but no progress. I'm trying to do it manually now.

### What is the actual behavior?

Working installation from download up to flashing at max. 1 h of my time. Version 3.3.3 worked without any problems.
Less amount of data. It is just simple embedded board and I have to download giga bytes of tools.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

dobairoland, 2024-05-23:
You can download the file from https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch5/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch5-win32.zip

But I really recommend you to use the Windows Tools installer instead as it does several other setups as well (Windows defender, Antivirus, drivers), all of which you'll have to do it manually. Please check your permanently set environment variables and remove all ESP-IDF related ones. Then please re-run the installer and share the details about the issue you encounter.

yareach, 2024-05-25:
The installer does not work well on my computer. Have you tested it with Norton 360? Have you tested it with a non-English number format and date settings? For instance where decimal is not "." but ","? Also non-admin user?
I finally used the installer from the clonned repo. The ps1 didn't work, I had to use the bat. It does not set the path variable outside the terminal. I had to visualize the path after export was called and add it to the env. path manually. Now the compilation works - but not for the unit tests where a mock was used. I will sumit another issue for this.
Back to the installer and others. I think you should use the same path in the manual, installer, VSC plugin and command line tools. Now it is Espressif under profiles, or .espressif, or Espressif under profile/desktop, or Espressif in the root of C. This is very tiring.
And ff generates a warning during the compilation. This should not happen in a release version:
C:/Users/yarea_000/esp/esp-idf/components/fatfs/src/ff.c
C:/Users/yarea_000/esp/esp-idf/components/fatfs/src/ff.c:541:19: error: 'Ct855' defined but not used [-Werror=unused-const-variable=]
 static const BYTE Ct855[] = TBL_CT855

dobairoland, 2024-05-25:
> I had to use the bat. It does not set the path variable outside the terminal. I had to visualize the path after export was called and add it to the env. path manually.

Since you didn't use the installer you didn't get a shortcut on the desktop. Now in each terminal you need to run `export.bat` in order to set up the environment. I don't recommend to set any environment variable permanently. You will forgot about them and then you will have troubles when trying to switch to other versions.

> And ff generates a warning during the compilation. 

Please open a separate issue for that one.

If you happen to try the Windows installer again and encounter issues then please report it here: https://github.com/espressif/idf-installer/issues. Please try to be more descriptive and in case of the Windows installer share the installation log.

yareach, 2024-05-26:
There were those two shortcuts after the installer. None of them worked. They complained a variable (not sure if IDF_PATH) was null. I checket that and the variable was properly set. That's why I gave it on those scripts and removed the installation.
I use only one version, so in my case it is ok to have those vars in the system. If you are developping something, you don't want to change the version often. I would've stayed on 3.3.3 but it does not offer the new tools. So I'm now moving to the new one.
If there is a need of more versions, it should be controlled by an app build config where all the paths to the tools and sdk would be configured. Not by the path variable.

dobairoland, 2024-05-27:
Alright, I don't think we can pull out more information from the system since you already fixed your environment variables.

I think we can close this. Please give more information about those other issues you mentioned in separate issue reports.

closed 2024-05-27 by dobairoland