espressif/esp-idf #13836: Unable to build our code after include esp_ota_ops.h file

adityamilinddesai-eaton, 2024-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We are using https://github.com/espressif/esp-idf/tree/release/v5.1 and branch 5.1.2. We need to use latest set ota partition related API's which are present in app_update module inside esp_ota_ops.h file. In this version we need to include app_update module. If we include this module in cmakelist and include esp_ota_ops.h file in our code, then at the time of building , we are getting error of fatal error - esp_app_desc.h No such file or directory present inside the file esp_ota_ops.h. In actual this file is present inside esp_app_format module. But still it generate error. If we copy esp_app_desc.h file to another module eg esp_partition/include then same code will build without any issue. How to resolve this issue without copying this file to esp_partition module?

igrr, 2024-05-23:
Could you please also show the contents of CMakeLists.txt file of the component from which you are including esp_ota_ops.h?

adityamilinddesai-eaton, 2024-05-23:
Also we tried to add using REQUIRES app_update esp_app_format method also. But same error we got.

igrr, 2024-05-23:
I can see that you are adding include directories of other components by extending the INCLUDE_DIRS. This is not the right approach in CMake... Instead, you need to add the libraries you depend on to the REQUIRES or PRIV_REQUIRES lists.

Please try removing all the `INCLUDE_DIRS` which contain `../../../esp32_common/esp-idf`, then add the components you depend on into `REQUIRES`. Once you do this, please try to build again. If you get an error, please share the complete build log. Please upload the log as text rather than as a picture.

mahavirj, 2024-06-13:
Please refer to this build system guide [section](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#component-requirements) for more information on the component requirements. Hope this helps!

closed 2024-06-13 by mahavirj