espressif/esp-idf #13837: The issue about the frequency band wifi (2.4G or 5G) than ESP32 supports

NormanLuo, 2024-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32 supports to connect with 2.4G frequency band wifi,If a wifi is 2.4G and 5G mixing,Can the ESP32 stably connect with it?

NormanLuo, 2024-05-23:
Please Every Excellent Egineers Give me a solution

MaxwellAlan, 2024-05-24:
Hi @NormanLuo 

Yes, if Router's WiFi in mixed 2.4G and 5G mode, as long as the 2.4G band is functioning normally, the ESP32 can work with it.

Thanks.

NormanLuo, 2024-05-24:
> Hi @NormanLuo
> 
> Yes, if Router's WiFi in mixed 2.4G and 5G mode, as long as the 2.4G band is functioning normally, the ESP32 can work with it.
> 
> Thanks.

MaxwellAlan,Thanks for your answer

closed 2024-07-17 by Sherry616