espressif/esp-idf #13838: Unable to build code on esp-idf V5.1.2 due to error in file spi_flash_ll.h

adityamilinddesai-eaton, 2024-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We are using https://github.com/espressif/esp-idf/tree/release/v5.1 and branch 5.1.2. While building our code we are getting error in the file \esp-idf\components\hal\esp32\include\hal\spi_flash_ll.h line no 319 to 323 and 245 to 270. We are not directly calling any function from this file. Still we are getting below errors for this file. Previously we are using esp-idf 4.3. In that also for this file content was same. There we are not getting error. But after migration now we are getting error.
Also if we change the code in that file highlighted in the yellow color in right side, then build is successful.  
Is there any option need to select to resolve this issue?
Please refer the below comment screenshots for error.
In comparison image left side is original content and right side yellow highlight is working code due to modifications.

kolipakakondal, 2025-02-21:
Hello,

IDE version is not mentioned here. However, based on the editor style, it appears to be version Espressif-IDE 2.x, which does not support the new C/C++ language grammar used in ESP-IDF 5.x and later.

To resolve these indexer issues, the user needs to upgrade to Espressif-IDE 3.2.0, which is the latest version. This version C/C++ editor is based on the Language Server(clangd) and has the capability to understand the new C/C++ language grammar used in ESP-IDF 5.x.

However, since this is a major version upgrade, the user may encounter some breaking changes, such as:
- The old code formatter no longer working
- Changes in the ESP-IDF Manager installation process, etc

Please refer to this post https://developer.espressif.com/blog/espressif-ide-lsp-support/#why-moving-to-lsp-based-editor

kolipakakondal, 2025-02-21:
For some reason, If there is no immediate plan to upgrade the IDE to version 3.x, here is the workaround to disable the problem markers in the Editor.

> Configure Problem Markers in Preferences
> 	1.	Go to:
> 	•	Windows → Preferences (on Windows/Linux)
> 	•	Eclipse → Preferences (on macOS)
> 	2.	Navigate to C/C++ → Code Analysis.
> 	3.	Uncheck the markers you want to disable (e.g., specific warnings or errors).
> 	4.	Click Apply and Close.