espressif/esp-idf #13840: After network failure installer muzz CONTINUE from point of failure without RE-DOWNLOADING GIGABYTE(s) of sh#t

powerbroker, 2024-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2.1

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Just continue'n'sh#t, REUSING already downloaded ZIP package or just go on in case of GIT.
May ask if i want to continue with existent folder or restart everything from the beginning here or something else.

### What is the actual behavior?

Says "there is already not-empty folder 'esp-idf-v5.2.1' so please choose another one" and starts from blank point, RE-DOWNLOADING GIGABYTE(s) of sh#t over UNSTABLE NETWORK CONNECTION again.

### Steps to reproduce.

Start installation of ESP-IDF with downloaded online installer. Break network connection.
Restart installer, point it to install into previously selected location.
Enjoy the choice between restart from the beginning or huck yourself.

dobairoland, 2024-05-23:
@powerbroker Please try to be constructive and avoid venting your frustration.

I've opened https://github.com/espressif/idf-installer/issues/244 to track this request and closing this one as we will not track issues with vulgarism.

closed 2024-05-23 by dobairoland

dobairoland, 2024-05-23:
We recommend to use the offline installer for users with unreliable network connections. The offline installer can be downloaded with a downloading manager which is able to resume the download upon network failure.