espressif/esp-idf #13841: Installer muzz be pythonless or at least allow to choose python executable location

powerbroker, 2024-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.2.1

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Installer uses downloaded poothon with no choice, while poothon is known in causing compatibility troubles.
There muzz be no poothon usage at all during installation: tons of software work this way. Or, at least, possibility to enforce the pre-installed, pre-patched and known as COMPATIBLE poothon version.

### What is the actual behavior?

Installer tries to do some magic with it's internal poothon, fails and provides no way to recover.

### Steps to reproduce.

Install e.g. under Windows 7 - everything works near 'ok' but poothon completely fails.

dobairoland, 2024-05-23:
The installer deploys a virtual Python environment. Having all installations use one tested environment is much reliable than using an arbitrary installed version. Most of Pythons from Windows Store have some limitations.

The Python environment deployed by the installer is isolated from the rest of the system. One can have other Pythons installed and used for other purposes. Or one doesn't have to have any globally installed Python. 

> Install e.g. under Windows 7

Windows 7 is not supported. Please use Windows 10 or 11.

closed 2024-05-23 by dobairoland

powerbroker, 2024-05-25:
my dear friend,

i would totally agree with you if only i didn't PAY MY MONEY for the ESP device but got it for free instead.

i was right doing that, am i?
so could you please go on and proceed with the bug fix?

igrr, 2024-05-25:
I think what you are asking for (selecting Python interpreter in the installer) is supported, although not enabled by default for the reasons @dobairoland mentioned.

You can launch the installer with a command line option `/USEEMBEDDEDPYTHON=no` ([docs](https://github.com/espressif/idf-installer?tab=readme-ov-file#command-line-parameters)). This will enable Python interpreter selection screen in the installer.

Please note that the Python environment must still be recent enough — for example, IDF 5.2.1 requires at least Python 3.8 (which is the oldest Python version still officially supported at the moment).

dobairoland, 2024-05-25:
> for example, IDF 5.2.1 requires at least Python 3.8 (which is the oldest Python version still officially supported at the moment).

And that is the last one which still supports Windows 7. Python 3.9 doesn't run on Windows 7. I recommend to get it from https://www.python.org/downloads/windows/ instead of Windows Store.