espressif/esp-idf #13843: Task runtime percentage calculation

hitecSmartHome, 2024-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to calculate runtime percentages for different tasks and also gather additional information from them.
Everything goes well but the percentage calculations are always wrong and I don't know why.

### Here is my function to retrive the task information and put them into an ArduinoJsonDocument.

```cpp
void Sys::getTaskInfo(JsonObject& taskInfo) {
    JsonArray taskList = taskInfo["taskList"].to<JsonArray>();
    UBaseType_t uxArraySize;
    TaskStatus_t* taskStatusArray;
    uint32_t ulTotalRunTime, ulTaskRuntimePercentage;
    uxArraySize = uxTaskGetNumberOfTasks();
    taskStatusArray = (TaskStatus_t*)pvPortMalloc(uxArraySize * sizeof(TaskStatus_t));

    // Return if we couldn't retrive the task status array
    if (taskStatusArray == NULL) {
        return;
    }
    uxArraySize = uxTaskGetSystemState(taskStatusArray, uxArraySize, &ulTotalRunTime);

    if (ulTotalRunTime > 0) {
        // Convert ulTotalRunTime to milliseconds if it's in ticks
        // Assuming configTICK_RATE_HZ = 1000 for milliseconds
        ulTotalRunTime /= configTICK_RATE_HZ;

        taskInfo["totalRunTime"] = ulTotalRunTime;

        for (UBaseType_t i = 0; i < uxArraySize; i++) {
            JsonObject task = taskList.add<JsonObject>();
            TaskStatus_t* taskStatus = &taskStatusArray[i];

            // Convert task runtime to the same unit
            uint32_t taskRunTime = taskStatus->ulRunTimeCounter / configTICK_RATE_HZ;
            ulTaskRuntimePercentage = (taskRunTime * 100UL) / ulTotalRunTime;

            // Store the task information
            task["name"] = taskStatus->pcTaskName;
            task["state"] = taskStatus->eCurrentState;
            task["priority"] = taskStatus->uxCurrentPriority;
            task["basePriority"] = taskStatus->uxBasePriority;
            task["stackHighWaterMark"] = taskStatus->usStackHighWaterMark;
            task["coreId"] = taskStatus->xCoreID;
            task["number"] = taskStatus->xTaskNumber;
            task["runTimeCounter"] = taskRunTime;
            task["runTimePercent"] = ulTaskRuntimePercentage;
            task["stackSize"] = taskHandler.getStackSize(taskStatus->xHandle);
        }
    }

    vPortFree(taskStatusArray);
}
```

The documentation does not say how should it be calculated. I want the run time counts in milliseconds.
The [FreeRTOS Run TIme Stats](https://www.freertos.org/rtos-run-time-stats.html) only describes the `vTaskGetRunTimeStats()` API which does not go into detail. I need much finer controll than a printf.

Also the documentation of the `uxTaskGetSystemState` API links to the same `vTaskGetRunTimeStats()` docs.

If I understand correctly, `uxTaskGetSystemState` will give me the total run time in ticks.

### That is why my calculation is the following:

```cpp
// This should give me the total run time in milliseconds?
uint32_t ulTotalRunTime = ulTotalRunTime / configTICK_RATE_HZ;
// This should give me the task total run time in milliseconds?
uint32_t taskRunTime = taskStatus->ulRunTimeCounter / configTICK_RATE_HZ;
// This should give me the task total run time in percentages?
uint32_t ulTaskRuntimePercentage = (taskRunTime * 100UL) / ulTotalRunTime;
```

The numbers are correct to a point. Here is an example of my task list

```json
"taskList": [
      {
        "name": "httpd",
        "state": 0,
        "priority": 5,
        "basePriority": 5,
        "stackHighWaterMark": 17404,
        "coreId": 0,
        "number": 15,
        "runTimeCounter": 1025,
        "runTimePercent": 0,
        "stackSize": 0
      },
      {
        "name": "ComponentsTask",
        "state": 1,
        "priority": 1,
        "basePriority": 1,
        "stackHighWaterMark": 17700,
        "coreId": 2147483647,
        "number": 20,
        "runTimeCounter": 114892,
        "runTimePercent": 12,
        "stackSize": 20000
      },
      {
        "name": "HardwareConfigT",
        "state": 1,
        "priority": 1,
        "basePriority": 1,
        "stackHighWaterMark": 4100,
        "coreId": 2147483647,
        "number": 17,
        "runTimeCounter": 163839,
        "runTimePercent": 18,
        "stackSize": 7000
      },
      {
        "name": "systemTask",
        "state": 1,
        "priority": 1,
        "basePriority": 1,
        "stackHighWaterMark": 13076,
        "coreId": 1,
        "number": 10,
        "runTimeCounter": 90775,
        "runTimePercent": 10,
        "stackSize": 16384
      }
]
```

This seems good but if i let it run it can produce the following results:

```json
{
        "name": "IDLE",
        "state": 1,
        "priority": 0,
        "basePriority": 0,
        "stackHighWaterMark": 928,
        "coreId": 1,
        "number": 6,
        "runTimeCounter": 763230,
        "runTimePercent": 85,
        "stackSize": 0
      },
      {
        "name": "IDLE",
        "state": 1,
        "priority": 0,
        "basePriority": 0,
        "stackHighWaterMark": 892,
        "coreId": 0,
        "number": 5,
        "runTimeCounter": 3761665,
        "runTimePercent": 419,
        "stackSize": 0
      }
```

As you can see, one of the IDLE tasks has a 419% run time. Sometimes all the other task's runtime does this.
Is this a 32 bit int overflow? Is this a tick overflow? Is this a millis overflow? Where am I going wrong? Does the `uxTaskGetSystemState` correctly working?

Dazza0, 2024-05-23:
FreeRTOS run time stats are measured in units of the run time stats clock (i.e., one unit is whatever the one period of the run time stats clock is).

On the ESP32, the choice of the run time stats clock is set using the [`CONFIG_FREERTOS_RUN_TIME_STATS_CLK`](https://docs.espressif.com/projects/esp-idf/en/v5.2.1/esp32/api-reference/kconfig.html#config-freertos-run-time-stats-clk) option, with each clock choice having its own clock frequency/period.

- `CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER` uses the ESP Timer as the run time stats clock. This clock runs at 1MHz
- `CONFIG_FREERTOS_RUN_TIME_STATS_USING_CPU_CLK` uses the internal CPU clock as the run time stats clock. This clock has the same frequency as the CPU (which itself is configurable to 160/240 MHz)

hitecSmartHome, 2024-05-24:
Oh I see. It is indeed set to `CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER`.
So in theory the `uxTaskGetSystemState` returns the total run time in microseconds already and the `taskStatus->ulRunTimeCounter` is also in microseconds.

AxelLin, 2024-05-24:
@Dazza0 

I have below settings: (Test on v5.2.1-528-g70fc0e3037, esp32s3)
CONFIG_FREERTOS_RUN_TIME_COUNTER_TYPE_U64=y
CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER=y

But after running for a few hours, I got:

IDLE1           2720836312              414%
IDLE0           2335057183              400%

I'm wondering if portALT_GET_RUN_TIME_COUNTER_VALUE() needs fix:
```
diff --git a/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos/portmacro.h b/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos/portmacro.h
index c33b775a85..b00bd5bd6c 100644
--- a/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos/portmacro.h
+++ b/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos/portmacro.h
@@ -589,7 +589,7 @@ void vPortTCBPreDeleteHook( void *pxTCB );
 #define portGET_RUN_TIME_COUNTER_VALUE() 0
 #ifdef CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER
 /* Coarse resolution time (us) */
-#define portALT_GET_RUN_TIME_COUNTER_VALUE(x)    do {x = (uint32_t)esp_timer_get_time();} while(0)
+#define portALT_GET_RUN_TIME_COUNTER_VALUE(x)    do {x = (configRUN_TIME_COUNTER_TYPE)esp_timer_get_time();} while(0)
 #endif
 
 // --------------------- TCB Cleanup -----------------------
diff --git a/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h b/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h
index 91085eb1a9..4300a30638 100644
--- a/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h
+++ b/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h
@@ -520,7 +520,7 @@ extern void _frxt_setup_switch( void );     //Defined in portasm.S
  */
 #define portGET_RUN_TIME_COUNTER_VALUE() xthal_get_ccount()
 #ifdef CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER
-#define portALT_GET_RUN_TIME_COUNTER_VALUE(x) do {x = (uint32_t)esp_timer_get_time();} while(0)
+#define portALT_GET_RUN_TIME_COUNTER_VALUE(x) do {x = (configRUN_TIME_COUNTER_TYPE)esp_timer_get_time();} while(0)
 #endif
 
 // --------------------- TCB Cleanup -----------------------
```

hitecSmartHome, 2024-05-24:
So I rewrote it to not scale to anything and just use the total run times as it is, with `CONFIG_FREERTOS_RUN_TIME_STATS_USING_ESP_TIMER` enabled.

Will wait and see how it goes.

```cpp
void getTaskInfo() {
    // Get the task list
    JsonArray taskList = taskInfo["taskList"].to<JsonArray>();
    UBaseType_t uxArraySize;
    TaskStatus_t* taskStatusArray;
    uint32_t ulTotalRunTime, ulTaskRuntimePercentage;
    uxArraySize = uxTaskGetNumberOfTasks();
    taskStatusArray = (TaskStatus_t*)pvPortMalloc(uxArraySize * sizeof(TaskStatus_t));

    // Return if we couldn't retrive the task status array
    if (taskStatusArray == NULL) { return; }

    uxArraySize = uxTaskGetSystemState(taskStatusArray, uxArraySize, &ulTotalRunTime);

    if (ulTotalRunTime > 0) {
        for (UBaseType_t i = 0; i < uxArraySize; i++) {
            TaskStatus_t* taskStatus = &taskStatusArray[i];

            uint32_t taskRunTime = taskStatus->ulRunTimeCounter;
            uint64_t tempPercentage = (static_cast<uint64_t>(taskRunTime) * 100UL) / ulTotalRunTime;
            ulTaskRuntimePercentage = static_cast<uint32_t>(tempPercentage);
        
            printf("\nTask name: %s\n",taskStatus->pcTaskName);
            printf("taskRunTime : %u\n",taskRunTime);
            printf("ulTaskRuntimePercentage : %u\n",ulTaskRuntimePercentage );
        }
    }

    vPortFree(taskStatusArray);
}
```

hitecSmartHome, 2024-05-24:
It looks good so far. The two IDLE tasks are taking up most of the cpu time for some reason, that one i have to figure out but the help from @Dazza0 did the trick. I just compute it as milliseconds.

Thank you very much!

hitecSmartHome, 2024-05-24:
Maybe the uint32_t is not a good choice for these numbers

atanisoft, 2024-05-24:
FWIW here is how I display these in my code and I don't see odd results:
```
        std::vector<TaskStatus_t> taskList;
        uint64_t now = esp_timer_get_time();
        uint32_t ulTotalRunTime{0};
        if ((now - lastTaskList_) > taskListInterval_ || !lastTaskList_)
        {
            taskList.resize(taskCount);
            UBaseType_t retrievedTaskCount =
                uxTaskGetSystemState(&taskList[0], taskCount, &ulTotalRunTime);
            // adjust this time so we can use it for percentages.
            ulTotalRunTime /= 100;
            // sort by runtime
            std::sort(taskList.begin(), taskList.end(),
            [](const TaskStatus_t &left, const TaskStatus_t &right)
            {
                return left.ulRunTimeCounter > right.ulRunTimeCounter;
            });
            for (int task = 0; task < retrievedTaskCount; task++)
            {
                printf("%-16s id:%3d (%p), priority:%2d/%2d, free stack:%5" PRIu32
                    ", core:%4s, cpu%%:%-3" PRIu32 ", state:%s",
                    esp_log_system_timestamp(),
                    taskList[task].pcTaskName,
                    taskList[task].xTaskNumber,
                    taskList[task].xHandle,
                    taskList[task].uxCurrentPriority,
                    taskList[task].uxBasePriority,
                    taskList[task].usStackHighWaterMark,
                    taskList[task].xCoreID == tskNO_AFFINITY ? "ANY" :
                    taskList[task].xCoreID == PRO_CPU_NUM ? "PRO" :
                    taskList[task].xCoreID == APP_CPU_NUM ? "APP" : "UNK",
                    (taskList[task].ulRunTimeCounter / portNUM_PROCESSORS) / ulTotalRunTime,
                    taskList[task].eCurrentState == eRunning ? "Running" :
                    taskList[task].eCurrentState == eReady ? "Ready" :
                    taskList[task].eCurrentState == eBlocked ? "Blocked" :
                    taskList[task].eCurrentState == eSuspended ? "Suspended" :
                    taskList[task].eCurrentState == eDeleted ? "Deleted" : "Unknown"
                );
            }
            lastTaskList_ = now;
        }
```

I use the above code to dump the running task list every ~15sec (`taskListInterval_`). You should be able to use similar code to dump the parameter data into a json payload for serializing.

Dazza0, 2024-05-28:
@hitecSmartHome 

> I think the runTimeCounter is overflowing and break the calculation

FreeRTOS run time stats do not handle overflows, thus all run time stats calculations overflow eventually. If you would like to measure task run times over a fixed period (which shouldn't overflow), the [real time stats example](https://github.com/espressif/esp-idf/tree/v5.2.1/examples/system/freertos/real_time_stats) demonstrates how to do this.

Dazza0, 2024-05-28:
@AxelLin 

> I'm wondering if portALT_GET_RUN_TIME_COUNTER_VALUE() needs fix:

That has already been fixed on master in https://github.com/espressif/esp-idf/commit/40684a0e1cd467260d26b1ccb1be84a025e2bd9f. Fix is currently awaiting backport to v5.2 branch

closed 2024-06-12 by Dazza0

hitecSmartHome, 2024-06-12:
Thank you very much. Unfortunatelly I'm using IDF 4.4.7 because of the Arduino dependency in PIO. I can't test it with 5.2