espressif/esp-idf #13845: about Timers and relaunching

diplfranzhoepfinger, 2024-05-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

using this: 

https://github.com/diplfranzhoepfinger/esp_timer_example

it is the Example: 

https://github.com/espressif/esp-idf-cxx/blob/main/examples/esp_timer_cxx/main/esp_timer_example.cpp

and modify it like this:

igrr, 2024-05-24:
> and it works, the Timer can be started 2 times.

I think it makes sense, because the timer is first started for 200 ms, so after a 550 ms delay, it is not running. Hence, we can start it again without getting an error.

If you want to start an already-running timer, you first need to call esp_timer_stop to cancel it.

diplfranzhoepfinger, 2024-05-24:
this is my Error. 

i wanted to test a Timer started 2 times which is still running.

Dazza0, 2024-05-28:
@diplfranzhoepfinger Closing this issue as it's not a bug. ESP Timer start APIs are designed to return `ESP_ERR_INVALID_STATE` if the timer is already running.

closed 2024-05-28 by Dazza0