espressif/esp-idf #13855: 程式只有单独SPI读取并没有问题，但在另一个核心执行不同任务时，却会影响到SPI的读取

zhaohaiyan75, 2024-05-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.0.4

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1U-N4

### Power Supply used.

USB

### What is the expected behavior?

我使用FreeRTOS(xTaskCreatePinnedToCore)进行多核分工，core 1做SPI高速的读取不间断，core 0做WiFi的连结与传输。希望两个线程之间相互不影响

### What is the actual behavior?

如果程式只有单独SPI读取并没有问题，但在另一个核心执行不同任务时，却会影响到SPI的读取，造成读的资料错误，这是什么原因呢?
明明是在不同核心下工作，怎么会互相影响到?尤其是在wifi连接瞬间，SPI读到错误值越多。

SPI是master，CLK速率40M
ESP32_CPU_Freq为240M，160M时错误更多

### Steps to reproduce.

参见上面的说明

nopnop2002, 2024-05-26:
What happens if you use xTaskCreate instead of xTaskCreatePinnedToCore?

zhaohaiyan75, 2024-05-26:
> What happens if you adjust the priority of SPI reading tasks and other tasks?
> 
> What happens if you use xTaskCreate instead of xTaskCreatePinnedToCore?

Core 1 only one task, its function is read data by spi, its priority is the highest. I have used xTaskCreatePinnedToCore to create task to assign cpu core.

nopnop2002, 2024-05-27:
When you enable WiFi, tiT, esp_timer, wifi, and sys_evt tasks will be added.
There is a possibility that these are running on Core1.

```
I (6454) MAIN: pcTaskName=tiT eCurrentState=2 xTaskNumber=8, uxCurrentPriority=18
I (6474) MAIN: pcTaskName=esp_timer eCurrentState=3 xTaskNumber=3, uxCurrentPriority=22
I (6484) MAIN: pcTaskName=wifi eCurrentState=2 xTaskNumber=10, uxCurrentPriority=23
I (6494) MAIN: pcTaskName=sys_evt eCurrentState=2 xTaskNumber=9, uxCurrentPriority=20
```

zhaohaiyan75, 2024-06-03:
> > Core 1 only one task
> 
> Enabling WiFi will add a tiT task.
> 
> And note that the core number for this task is -1 (Not pinned).
> 
> Just by enabling WiFi, tasks running on core #1 will be affected.
> 
> ```
> main            X       1       1004    4       0
> IDLE1           R       0       916     6       1
> IDLE0           R       0       920     5       0
> tiT             B       18      2104    8       -1
> ipc0            S       24      480     1       0
> Tmr Svc         B       1       1504    7       0
> esp_timer       S       22      3312    3       0
> wifi            B       23      656     10      0
> ipc1            S       24      492     2       1
> sys_evt         B       20      840     9       0
> ```

> > Core 1 only one task
> 
> Enabling WiFi will add a tiT task.
> 
> And note that the core number for this task is -1 (Not pinned).
> 
> Just by enabling WiFi, tasks running on core #1 will be affected.
> 
> ```
> main            X       1       1004    4       0
> IDLE1           R       0       916     6       1
> IDLE0           R       0       920     5       0
> tiT             B       18      2104    8       -1
> ipc0            S       24      480     1       0
> Tmr Svc         B       1       1504    7       0
> esp_timer       S       22      3312    3       0
> wifi            B       23      656     10      0
> ipc1            S       24      492     2       1
> sys_evt         B       20      840     9       0
> ```

thank you for your informatiom.
When I set read spi task priority high tiT, spi read is normal, This problem is solved

Alvin1Zhang, 2025-01-06:
Thanks for reporting, glad to hear that the issue was resolved, feel free to reopen.

closed 2025-01-06 by Alvin1Zhang