espressif/esp-idf #13858: ff compilation warning on 4.4.7 - 'Ct855' defined but not used

yareach, 2024-05-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When compiling an app which uses 4.4.7 I got a warning of something used. It requires turning off this check on gcc.
C:/Users/yarea_000/esp/esp-idf/components/fatfs/src/ff.c
C:/Users/yarea_000/esp/esp-idf/components/fatfs/src/ff.c:541:19: error: 'Ct855' defined but not used [-Werror=unused-const-variable=]
static const BYTE Ct855[] = TBL_CT855

igrr, 2024-05-26:
Could you please attach your sdkconfig file? Thank you.

yareach, 2024-05-27:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/15459032/sdkconfig.zip)

omnitex, 2026-11-03:
Hi @yareach, thanks for the report, and sorry for the late reply.

The warning (caused by `CONFIG_FATFS_CODEPAGE_DYNAMIC`) was fixed in IDF v5.0+ by the FatFs update to R0.14b: bb8587a1399. It was not backported to v4.4.

If you can, please upgrade to v5.0 or newer, since v4.4 is EOL. If you need to stay on v4.4, you can either select a fixed code page in menuconfig instead of the dynamic one, or patch it locally by adding `855` / `Ct855` to `validcp[]` / `tables[]` in `f_setcp()` (`components/fatfs/src/ff.c`).

Closing for now, feel free to reopen if you still see it on v5.x.

closed 2026-11-03 by omnitex