espressif/esp-idf #13859: Check that mutex semaphore is given by owner task - does not work

yareach, 2024-05-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.7

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board with ESP32 16M

### Power Supply used.

USB

### What is the expected behavior?

I looked into the code and I think the configCHECK_MUTEX_GIVEN_BY_OWNER define should be checked in more place. For instance I changed line 1724 of queue.c to: #if ( configUSE_MUTEXES == 1 && configCHECK_MUTEX_GIVEN_BY_OWNER == 1)
and now it works. It only complains that some stuff is defined but not used. I assume this check should be somewhere around to make the menuconfig option count.

### What is the actual behavior?

I've moved to 4.4.7 from 3.3.3. I ran my unit tests where checked the mutex is working by controling it from the unit test. The owner is a another task which is running. When I lock the mutex from the unit test code and check a function which uses the mutex it goes panic. This didn't happen with 3.3.3. I started with 4.4.7.
I found out the option in the menuconfig: Check that mutex semaphore is given by owner task
and I turned it off. But no effect, still the same.

### Steps to reproduce.

Create a task and a static mutex with timeout in a file. The task should be using the mutex. Add another function, that uses the mutex.
In the unit test include that file, init the mutex and run the task.
Lock the mutex from the unit test: TEST_ASSERT_EQUAL(pdTRUE, xSemaphoreTake(hSemUnitTimeCalback, UNIT_TIME_CALLBACK_SET_TIMEOUT / portTICK_RATE_MS));
Call the function that uses the mutex: TEST_ASSERT_FALSE(UnitTimeRegisterTimeCallback(TimeCallback));
Now it should go panic.

The function is simple:
bool UnitTimeRegisterTimeCallback(unit_time_callback_t pfcTimeCallback)
{
	/* Time callback set mutex */
	if (atomic_load(&bUnitTimeInitialized) &&
		(xSemaphoreTake(hSemUnitTimeCalback, UNIT_TIME_CALLBACK_SET_TIMEOUT_TICKS) == pdTRUE))
	{
		/* Sets the time callback */
		pfcUnitTimeCallback = pfcTimeCallback;
	
		/* Releases the time callback mutex */
		xSemaphoreGive(hSemUnitTimeCalback);

		return true;
	}

	return false;
}

The task's while loop does this:
/* Time callback set mutex */
if (xSemaphoreTake(hSemUnitTimeCalback, UNIT_TIME_CALLBACK_TIMEOUT_TICKS) == pdTRUE)
{
	/* Periodical callback indicating the current time */
	if (pfcUnitTimeCallback != NULL)
	{
		pfcUnitTimeCallback(sTimeInfo, sTimeNow);
	}

	/* Releases the time callback mutex */
	xSemaphoreGive(hSemUnitTimeCalback);
}
vTaskDelayUntil(&lastWakeTime, UNIT_TIME_CONTROL_TASK_PERIOD_TICKS);

The timeouts are 500 to 1000 ms.

### Debug Logs.

```plain
Running UnitTimeRegisterTimeCallback, UnitTimeUnregisterTimeCallback...

assert failed: vTaskPriorityDisinheritAfterTimeout tasks.c:5034 (pxTCB != pxCurrentTCB[xPortGetCoreID()])

Backtrace: 0x40081ace:0x3ffb5030 0x40085a95:0x3ffb5050 0x4008b551:0x3ffb5070 0x400885ad:0x3ffb5190 0x40086a86:0x3ffb51b0 0x400d3b63:0x3ffb51f0 0x400d3dfb:0x3ffb5210 0x400da9a5:0x3ffb5260 0x400daaf1:0x3ffb5290 0x400dab5f:0x3ffb52b0 0x400dac2b:0x3ffb52d0 0x400eac64:0x3ffb52f0 0x40089029:0x3ffb5310
0x40081ace: panic_abort at C:/Users/yarea_000/esp/esp-idf/components/esp_system/panic.c:408

0x40085a95: esp_system_abort at C:/Users/yarea_000/esp/esp-idf/components/esp_system/esp_system.c:137

0x4008b551: __assert_func at C:/Users/yarea_000/esp/esp-idf/components/newlib/assert.c:85

0x400885ad: vTaskPriorityDisinheritAfterTimeout at C:/Users/yarea_000/esp/esp-idf/components/freertos/tasks.c:5034 (discriminator 1)

0x40086a86: xQueueSemaphoreTake at C:/Users/yarea_000/esp/esp-idf/components/freertos/queue.c:1741

0x400d3b63: UnitTimeUnregisterTimeCallback at c:\users\yarea_000\source\repos\prj_unit\unit\prj_ble_server\components\unit_time/unit_time.c:389 (discriminator 1)

0x400d3dfb: test_func_524 at C:/Users/yarea_000/source/repos/prj_Unit/Unit/prj_ble_server/components/unit_time/test/test_unit_time.c:584 (discriminator 3)

0x400da9a5: unity_default_test_run at C:/Users/yarea_000/esp/esp-idf/components/unity/unity_runner.c:78

0x400daaf1: unity_run_single_test at C:/Users/yarea_000/esp/esp-idf/components/unity/unity_runner.c:125

0x400dab5f: unity_run_all_tests at C:/Users/yarea_000/esp/esp-idf/components/unity/unity_runner.c:184 (discriminator 3)

0x400dac2b: app_main at C:/Users/yarea_000/source/repos/prj_Unit/Unit/prj_ble_server/test/main/prj_test.c:73

0x400eac64: main_task at C:/Users/yarea_000/esp/esp-idf/components/freertos/port/port_common.c:141

0x40089029: vPortTaskWrapper at C:/Users/yarea_000/esp/esp-idf/components/freertos/port/xtensa/port.c:142
```

igrr, 2024-05-26:
We can definitely check if `configCHECK_MUTEX_GIVEN_BY_OWNER` should be considered in more places. However please do note that disabling this check doesn't make it legal to give a mutex which was taken by another task. In older FreeRTOS versions this case wasn't checked, and sometimes it could result in adding a task to the FreeRTOS internal task list twice, which eventually caused the scheduler to enter an infinite loop. The exact scenario when could be hard to reproduce, and the resulting issues were complex to debug. Therefore an assert was added to catch such issues earlier.

In any case, FreeRTOS mutexes should only be released by the task which owns the mutex. This has been confirmed a few times by the author of FreeRTOS (e.g. [here](https://forums.freertos.org/t/mutex-acquired-by-one-task-xsemaphoretake-can-be-released-xsemaphoregive-by-another/9799/2)). You'll probably need to refactor your code to take this limitation into account.

yareach, 2024-05-27:
Well, I don't use this option in the applicaiton code. I only need it in the unit test code where I lock the mutex manually from the unit test and then call the function to verify that the function where the mutex is used. The function will return false because it was not able to take the mutex and timed out.
In such case I would need to create another dummy task to just lock the mutex, sleep and unlock. Isn't it useless effort if it is just a unit test?

Dazza0, 2024-05-28:
@yareach 

> In such case I would need to create another dummy task to just lock the mutex, sleep and unlock. Isn't it useless effort if it is just 
a unit test?

In your test case:
```c
 TEST_ASSERT_EQUAL(pdTRUE, xSemaphoreTake(hSemUnitTimeCalback, UNIT_TIME_CALLBACK_SET_TIMEOUT / portTICK_RATE_MS));
TEST_ASSERT_FALSE(UnitTimeRegisterTimeCallback(TimeCallback));
```
the same mutex gets taken twice from the same task, once in `test_func_524()`, and the second attempt in `UnitTimeUnregisterTimeCallback()`.

In order for a mutex to be taken by the same task multiple times, you must use a [recursive mutex](https://www.freertos.org/RTOS-Recursive-Mutexes.html).

yareach, 2024-05-28:
@Dazza0 I got that. As I mentioned, I don't use it in the app. This is only a case for the unit test where I want to lock the mutex a verify the function handles it well.

Dazza0, 2024-06-26:
@yareach Closing issue as FreeRTOS non-recursive mutexes are not designed to be recursively taken.

closed 2024-06-26 by Dazza0