espressif/esp-idf #13860: esp32s3 i2s做slave播放不正常

crl6, 2024-05-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

写了个i2s播放wav音乐代码。
1. esp32做master 进行播放
2. rk3588做slave模式，使用tinycap进行录音。
3. 测试正常，录到的wav文件在电脑播放正常。
esp32s3端代码什么都不改，只改成slave模式也就是 .role = I2S_ROLE_MASTER改成 .role = I2S_ROLE_SLAVE。
rk3588端代码什么都不改，只改成slave模式。进行如下操作：
1. rk3588使用tinycap进行录音（提供时钟）
2. esp32s3上电开始播放wav。
3. 把录到的wav拉到电脑播放。声音听着变快了一倍。
请问esp32s3 i2s做slave有什么限制，或者特别配置吗？
附件是录到的wav文件 
[wav文件.zip](https://github.com/espressif/esp-idf/files/15446846/wav.zip)

L-KAYA, 2024-05-27:
I2S的采样速率是由主机决定的，所以应该是rk3588的时钟有关的寄存器配置没配对，只不过在ESP做主机的时候按ESP的时钟采样了，所以是正常的，rk3588做主机的时候就按rk3588的时钟采样了。

crl6, 2024-05-29:
> I2S的采样速率是由主机决定的，所以应该是rk3588的时钟有关的寄存器配置没配对，只不过在ESP做主机的时候按ESP的时钟采样了，所以是正常的，rk3588做主机的时候就按rk3588的时钟采样了。

esp32s3 做slave必须使用mclk引脚吗？

L-KAYA, 2024-05-31:
1. esp做i2s从机不是必须使用mclk引脚的，如果i2s主机有mclk信号的话可以接也可以不接，接了之后采样率能高点，不接常见的采样率（如48K以下）也基本都可以支持；
2. esp做i2s从机的时候，如果要接受来自主机的mclk时钟作为输入，需配置时钟源为外部输入`clk_src = I2S_CLK_SRC_EXTERNAL`，以及输入的mclk时钟频率`ext_clk_freq_hz`。

crl6, 2024-06-02:
rk3588，驱动问题

closed 2024-06-02 by crl6