espressif/esp-idf #13862: Possible memory leak in gptimer component

GreyCon, 2024-05-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1.2

### Espressif SoC revision.

ESP32-D0WD (revision v1.0)

### Operating System used.

macOS

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Firebeetle ESP32 IOT

### Power Supply used.

USB

### What is the expected behavior?

I expect gptimer_del_timer(gptimer) should free memory associated with the timer callback handler.

### What is the actual behavior?

After some time RTOS crashes, having run out of heap memory. 

### Steps to reproduce.

1.  I have a class which wraps a gptimer. It has a method which a client of the class calls;
```
void TimerManager::sleepFor(uint8_t hours, uint8_t minutes, uint8_t seconds) {
	time_t timeNow;
	time( &timeNow );

	if (alarmTaskHandle == nullptr) {
		// Create the RTOS Task to handle alarm triggers
		ESP_LOGI(TAG, "Creating GPTimer Callback handler task.");
		xTaskCreate(timerTask, "TimerAlarmTask", StackSize, this, tskIDLE_PRIORITY+1, &alarmTaskHandle);
	}
	int noSecondsToSleep = (hours*60*60 + minutes*60 + seconds);
	ESP_LOGI(TAG,"sleepFor, Will sleep for %d seconds",noSecondsToSleep );

	callerTaskHandle = xTaskGetCurrentTaskHandle();
	startTimerAndAlarm(noSecondsToSleep);
}
```
This method in turn calls code to create and start a gptimer;
```
static void startTimerAndAlarm(int seconds) {
	ESP_LOGI(TAG, "Start timer for %d seconds", seconds);

    gptimer_config_t timer_config = {
        .clk_src = GPTIMER_CLK_SRC_DEFAULT,
        .direction = GPTIMER_COUNT_UP,
        .resolution_hz = 1000000, // 1MHz, 1 tick=1uS
    };
    ESP_ERROR_CHECK(gptimer_new_timer(&timer_config, &gptimer));

	gptimer_alarm_config_t alarmCfg = {
			.alarm_count = (uint64_t)seconds * 1000000,
			.reload_count = 0, // counter will reload with 0 on alarm event
	};
	ESP_ERROR_CHECK(gptimer_set_alarm_action(gptimer, &alarmCfg));
	gptimer_event_callbacks_t cbs = {
			.on_alarm = timerGroupISRCallback,
	};

	ESP_LOGI(TAG,"scheduleTasks->calling register_event_callbacks");
	ESP_ERROR_CHECK(gptimer_register_event_callbacks(gptimer, &cbs, alarmTaskHandle));
	ESP_LOGI(TAG,"scheduleTasks->calling enable");
	ESP_ERROR_CHECK(gptimer_enable(gptimer));
	ESP_LOGI(TAG,"scheduleTasks->calling start");
	ESP_ERROR_CHECK(gptimer_start(gptimer));
}

```

The callback handler simply posts to a waiting task, which in turn kills the timer;
```
static bool IRAM_ATTR timerGroupISRCallback(gptimer_handle_t timer, const gptimer_alarm_event_data_t *edata, void *userData) {
    BaseType_t high_task_awoken = pdFALSE;

    TaskHandle_t taskToNotify = (TaskHandle_t)userData;

	ESP_ERROR_CHECK(gptimer_stop(gptimer));

	// Alarm has popped, let our RTOS Alarm task handle it.
	vTaskNotifyGiveFromISR(taskToNotify, & high_task_awoken);
    return (high_task_awoken == pdTRUE);
}

static void timerTask(void *userData) {
	uint32_t notifiedValue;
	uint64_t timeLeft;

    while (1) {
    	showMyStack("timerTask");
    	ESP_LOGI(TAG,"timerTask waiting...");
    	// RTOS Task to sleep until timer pops.    	xTaskGenericNotifyWait(uxIndexToWaitOn, ulBitsToClearOnEntry, ulBitsToClearOnExit, pulNotificationValue, xTicksToWait), bits_exit_clear, value, wait_time)
		xTaskNotifyWaitIndexed( 0, 0x00, ULONG_MAX, &notifiedValue, portMAX_DELAY );
    		// Timer popped , notify calling task.
    		ESP_LOGI(TAG,"timerTask triggered...");
			ESP_ERROR_CHECK(gptimer_get_raw_count(gptimer, &timeLeft));
			ESP_LOGI(TAG,"RawValue: %llu",timeLeft);
			ESP_ERROR_CHECK(gptimer_disable(gptimer));
			ESP_ERROR_CHECK(gptimer_del_timer(gptimer));

    		xTaskNotifyGive(callerTaskHandle);
    }
}
```

Example usage in an RTOS task:
```
static void waitForTask(void *userData) {
	unsigned long int notifiedValue;
	time_t timeNow;

	while (1) {
		ESP_LOGI(TAG,"waitForTask, sleeping for a while...");

		showMyStack();
		timerManager->sleepFor(0,0,1);
		xTaskNotifyWait(0, 0,  &notifiedValue, portMAX_DELAY );
		time ( &timeNow );
		ESP_LOGI(TAG,"waitForTask, Current Time is %s",ctime(&timeNow) );
	}
}
```

This will run for some time, and eventually die from lack of heap. It looks to me like the gptimer_register_event_callbacks method calls code which allocates some callback handler, but this is not freed when gptimer_del_timer(gptimer) is called.

### Debug Logs.

```plain
Analysis of stack for failure:
#0  remove_free_block (sl=6, fl=0, block=0x3ffd8828, control=0x3ffc0274) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/heap/tlsf/tlsf.c:332
#1  block_locate_free (size=<optimized out>, control=<optimized out>) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/heap/tlsf/tlsf.c:567
#2  tlsf_malloc (tlsf=0x3ffc0274, size=<optimized out>) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/heap/tlsf/tlsf.c:1004
#3  0x4008f4dd in multi_heap_malloc_impl (heap=0x3ffc0260, size=8) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/heap/multi_heap.c:207
#4  0x40082280 in heap_caps_malloc_base (size=8, caps=2052) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/heap/heap_caps.c:176
#5  0x400822d8 in heap_caps_malloc (size=8, caps=2052) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/heap/heap_caps.c:197
#6  0x400d5406 in esp_intr_alloc_intrstatus (source=14, flags=2062, intrstatusreg=1073082528, intrstatusmask=1, handler=0x4008850c <gptimer_default_isr>, arg=0x3ffd87c0, ret_handle=ret_handle@entry=0x3ffd87f4) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/esp_hw_support/intr_alloc.c:529
#7  0x401b0002 in gptimer_register_event_callbacks (timer=0x3ffd87c0, cbs=0x3ffd7c70, user_data=0x3ffd9020) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/driver/gptimer/gptimer.c:246
#8  0x400df835 in startTimerAndAlarm (seconds=<optimized out>) at /Users/concunningham/Documents/git-repos/PharmaSlave/PSMonitor/main/Utilities/TimerManager.cpp:122
#9  0x400df934 in TimerManager::sleepFor (this=<optimized out>, hours=0 '\\000', minutes=0 '\\000', seconds=<optimized out>) at /Users/concunningham/Documents/git-repos/PharmaSlave/PSMonitor/main/Utilities/TimerManager.cpp:96
#10 0x400dc4f1 in waitForTask (userData=<optimized out>) at /Users/concunningham/Documents/git-repos/PharmaSlave/PSMonitor/main/OperationalMode/OperationalController.cpp:49
#11 0x4008b3bc in vPortTaskWrapper (pxCode=0x400dc4c8 <waitForTask(void*)>, pvParameters=0x3ffb0320 <delay1>) at /Users/concunningham/Documents/ESPIDF/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

No.

GreyCon, 2024-05-26:
Closing this  - I looked at the gptimer code and I may have been hasty. It may well be that my application is running out of heap, and gptimer_register_event_callbacks can't obtain any free heap memory. Whether this is as a result of fragmentation by the repeated create/use/destroy gptimer remains to be seen. I'll trim it down to a small test program which does nothing else but that, and see how it goes.

GreyCon, 2024-05-26:
After more analysis, I don't believe that my application , which has been stable, is using all the heap. The error being trapped, as shown in the coredump is at line 332 of tlsf.c, where a free block in the heap has been successfully identified, and is supposedly being removed for use from the heap. See below.

So, maybe this is a more general memory issue, which only arises when I create and destroy gptimers a lot....

```
static inline __attribute__((always_inline)) void remove_free_block(control_t* control, block_header_t* block, int fl, int sl)
{
	block_header_t* prev = block->prev_free;
	block_header_t* next = block->next_free;
	tlsf_assert(prev && "prev_free field can not be null");
	tlsf_assert(next && "next_free field can not be null");
	next->prev_free = prev;   //  ******This is the offending line - I can't see how it would cause a crash.
	prev->next_free = next;
```

I will attach the full coredump if needed.

GreyCon, 2024-05-26:
Oh, I keep forgetting I'm not working in a protected memory environment. I guess my larger application could be corrupting the heap somehow - I found some docs on heap debugging (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html)   I can add some tracing to the main application and see if I can see anything bad happening.

If you think I'm thinking about this wrongly, please feel free to just close the issue, and tell me so.

suda-morris, 2024-05-27:
`gptimer_register_event_callbacks` will only do memory allocation for the first time it install interrupt handler. The interrupt resources will be recycled in `gptimer_destroy`. IDF has many test cases for testing the memory leak, and this `gptimer_register_event_callbacks` and `gptimer_del_timer` are for sure being covered.

After reading your code, I would prefer to call `gptimer_new_timer` and `gptimer_register_event_callbacks` in the constructor of class `TimerManager`.

GreyCon, 2024-05-27:
Hey, thanks for your reply. I know you must be busy. I've been through the gptimer code, and it all looks fine. The failure clearly occurs in a malloc, so my heap is somehow getting corrupted. I'm paring back my main app a little at a time to see what is causing it. The problem is it might only happen every few hours - but I am sure you are familiar with this :-)

Cheers,
Con

closed 2024-05-27 by GreyCon