espressif/esp-idf #13863: RISC-V ULP GPIO becomes disconnected when rtc_gpio_deinit() is called on unrelated pin

funkBuild, 2024-05-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

After initialising a GPIO as RTC, the RISC-V ULP should retain access even when `rtc_gpio_deinit(io_num)` is called on an unrelated pin. This is especially important because  `gpio_config()` will automatically call `rtc_gpio_deinit` for any pins that can be used as RTC IO

### What is the actual behavior?

After initialising a RTC GPIO, calling `rtc_gpio_deinit()` on a different pin (via  `gpio_config()` ) will cause all RTC GPIO to be inaccessible from the ULP.

### Steps to reproduce.

```
// main.c

#include "driver/gpio.h"
#include "driver/rtc_io.h"
#include "esp_sleep.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "soc/rtc_cntl_reg.h"
#include "soc/rtc_periph.h"
#include "soc/sens_reg.h"
#include "ulp_main.h"
#include "ulp_riscv.h"
#include <stdio.h>

extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[] asm("_binary_ulp_main_bin_end");

static void init_ulp_program(void);

void app_main(void) {
  /* If this call to gpio_config is removed, the example works fine */

  gpio_config_t io_conf = {};
  io_conf.intr_type = GPIO_INTR_DISABLE;
  io_conf.mode = GPIO_MODE_OUTPUT;
  io_conf.pin_bit_mask = (1ULL << 1);
  io_conf.pull_down_en = 0;
  io_conf.pull_up_en = 0;

  gpio_config(&io_conf);

  /* Initialize selected GPIO as RTC IO, enable input, disable pullup and
   * pulldown */
  rtc_gpio_init(GPIO_NUM_18);
  rtc_gpio_set_direction(GPIO_NUM_18, RTC_GPIO_MODE_INPUT_ONLY);
  rtc_gpio_pulldown_dis(GPIO_NUM_18);
  rtc_gpio_pullup_dis(GPIO_NUM_18);
  rtc_gpio_hold_en(GPIO_NUM_18);

  esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
  /* not a wakeup from ULP, load the firmware */
  if (cause != ESP_SLEEP_WAKEUP_ULP &&
      esp_reset_reason() != ESP_RST_DEEPSLEEP) {
    printf("Not a ULP-RISC-V wakeup, initializing it! \n");
    init_ulp_program();
  }

  ulp_pulse2_gpio = 18;

  for (int i = 0; i < 10; i++) {
    printf("Pulse count %lu\n", ulp_pulse2_count);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }

  /* Go back to sleep, only the ULP Risc-V will run */
  printf("Entering in deep sleep\n\n");

  /* Small delay to ensure the messages are printed */
  vTaskDelay(100);

  // esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  esp_deep_sleep(1000000ULL); // 1s
}

static void init_ulp_program(void) {
  esp_err_t err = ulp_riscv_load_binary(
      ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start));
  ESP_ERROR_CHECK(err);

  /* The first argument is the period index, which is not used by the ULP-RISC-V
   * timer The second argument is the period in microseconds, which gives a
   * wakeup time period of: 20ms
   */
  ulp_set_wakeup_period(0, 20000);

  /* Start the program */
  err = ulp_riscv_run();
  ESP_ERROR_CHECK(err);
}
```

```
// ulp_main.c

#include "driver/gpio.h"
#include "driver/rtc_io.h"
#include "esp_sleep.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "soc/rtc_cntl_reg.h"
#include "soc/rtc_periph.h"
#include "soc/sens_reg.h"
#include "ulp_main.h"
#include "ulp_riscv.h"
#include <stdio.h>

extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[] asm("_binary_ulp_main_bin_end");

static void init_ulp_program(void);

void app_main(void) {
  /* Initialize selected GPIO as RTC IO, enable input, disable pullup and
   * pulldown */
  rtc_gpio_init(GPIO_NUM_18);
  rtc_gpio_set_direction(GPIO_NUM_18, RTC_GPIO_MODE_INPUT_ONLY);
  rtc_gpio_pulldown_dis(GPIO_NUM_18);
  rtc_gpio_pullup_dis(GPIO_NUM_18);
  rtc_gpio_hold_en(GPIO_NUM_18);

  rtc_gpio_deinit(0); // Comment this line out for the example to work correctly

  esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
  /* not a wakeup from ULP, load the firmware */
  if (cause != ESP_SLEEP_WAKEUP_ULP &&
      esp_reset_reason() != ESP_RST_DEEPSLEEP) {
    printf("Not a ULP-RISC-V wakeup, initializing it! \n");
    init_ulp_program();
  }

  ulp_pulse2_gpio = 18;

  for (int i = 0; i < 10; i++) {
    printf("Pulse count %lu\n", ulp_pulse2_count);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }

  /* Go back to sleep, only the ULP Risc-V will run */
  printf("Entering in deep sleep\n\n");

  /* Small delay to ensure the messages are printed */
  vTaskDelay(100);

  // esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  esp_deep_sleep(1000000ULL); // 1s
}

static void init_ulp_program(void) {
  esp_err_t err = ulp_riscv_load_binary(
      ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start));
  ESP_ERROR_CHECK(err);

  /* The first argument is the period index, which is not used by the ULP-RISC-V
   * timer The second argument is the period in microseconds, which gives a
   * wakeup time period of: 20ms
   */
  ulp_set_wakeup_period(0, 20000);

  /* Start the program */
  err = ulp_riscv_run();
  ESP_ERROR_CHECK(err);
}
```

pedrogsfaria, 2025-02-19:
I am facing a similar problem, any updates on how to solve it?

songruo, 2025-04-03:
This issue has been fixed in 51205a8fd8f7256e5121fe5d2c9bbb67a5c69003, and has been backported to previous versions.

closed 2025-04-03 by Alvin1Zhang