espressif/esp-idf #13865: BLE client authorization flag issue on ESP-IDF 5.1.4 after firmware update from 5.1.3

VNovytskyi, 2024-05-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# ble-auth-lost-issue-idf514
A test project to reproduce the issue (https://github.com/VNovytskyi/ble-auth-lost-issue-idf514) related to BLE client authentication when switching from version ESP-IDF 5.1.3 to version ESP-IDF 5.1.4. 
The problem is that the BLE bond (procedure or data format) on 5.1.3 is different from 5.1.4 and when upgrading to a new version, the authorization flag is reset, but the others do not change.

For example:
1) You have your own IoT project that uses the BLE feature. You are using the ESP-IDF 5.1.3 for developing new releases;
2) At every new release, the customers can connect to your products without entering a password because they are bonded;
3) You are using the next flags to protect data when creating characteristics: NIMBLE_PROPERTY::READ_AUTHEN, WRITE_AUTHEN, READ_ENC, WRITE_ENC;
4) Then you start using the ESP-IDF 5.1.4 and release a new version of your product;
5) Customers report that after a firmware update they have to re-enter the password to connect via BLE and this behaviour was not present in previous versions (or it can be another situation depending on BLE client: forever re-connect loop)
## Components
- ESP32 development board (rev 0)
- nRF Connect application
- installed ESP-IDF 5.1.3 and ESP-IDF 5.1.4
## Project base
Server example of ESP-NIMBLE-CPP (https://github.com/h2zero/esp-nimble-cpp)
## Steps to reproduce the issue
1) Clear bond with your ESP32 device
2) Build and flash the project with ESP-IDF 5.1.3
3) Bond with your ESP32 device (you can use nRF Connect application)
4) Reconnect
   
Now you must see:
```
--- onAuthenticationComplete ---
Authenticated: Yes
Encrypted:     Yes
Bonded:        Yes
```
5) Build and flash the project with ESP-IDF 5.1.4
6) Connect to ESP32

Now you must see:
```
--- onAuthenticationComplete ---
Authenticated: No
Encrypted:     Yes
Bonded:        Yes
```
## Expected result
The "Authenticated" flag after an firmware update is "Yes"
## Actual result
The "Authenticated" flag after an firmware update is "No"

rahult-github, 2024-05-29:
Hi @VNovytskyi , 

I tried the repo , but am facing compilation issue when i try to compile for 5.1.4 / 5.1.3    

Can you possibly: 

1. Share any IDF example based minimal project that can be used by us to reproduce the issue at our end ?
2. Or, can you enable nimble debug for 5.1.3 and 5.1.4 and share logs that can help us to debug more ? 
      (Top) → Component config → Bluetooth → NimBLE Options → NimBLE Host log verbosity : Set this to Debug
      (Top) → Component config → Log output → Default log verbosity :  Set this to Debug 
      
     
  If i look at the description, one reason may be that linkkey is getting lost . So please also confirm that on 5.1.3 also, BT_NIMBLE_NVS_PERSIST has been enabled, before migrating to 5.1.4 .  
  
  And, i am assuming that after upgrade, the flash is not getting erased explicitly in your use case.

rahult-github, 2024-09-10:
Hi @VNovytskyi  , do you have any further logging related update for this issue ?

rahult-github, 2024-09-24:
Closing this issue. Feel free to reopen in case of further updates.

closed 2024-09-24 by rahult-github