espressif/esp-idf #13867: MQTT: Outbox_Enqueue error

kovid-wakefit, 2024-05-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using ESP_IDF version 5.2.1(release). We have 75KB of data in a file inside the SD card, which we are chunking in 4KB and trying to publish on our server using MQTT_Client, but in runtime, we are getting **E (1038394) outbox: outbox_enqueue (53): Memory exhausted** error while uploading the chunk after uploading few files. After restarting, it starts working for a few minutes, uploads a few files, and then again shows the same error and stops uploading.

I have taken the reference from this example: https://github.com/espressif/esp-idf/tree/a322e6bdad4b6675d4597fb2722eea2851ba88cb/examples/protocols/mqtt/ssl_mutual_auth

Also, Please find the sdk_config file, for your reference.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/15465188/sdkconfig.txt)

[image]
[image]

euripedesrocha, 2024-05-28:
Hi @kovid-wakefit this is working as intended. I noticed from your log that you have a disconnection and after that you have the memory exhausted. 
I would suggest you to handle the disconnect event and wait until the reconnection to continue publishing.

kovid-wakefit, 2024-05-28:
this disconnection is happening after getting the "outbox:enqueue" error. The picture is swapped while posting.

euripedesrocha, 2024-05-28:
Are you using `esp_mqtt_client_publish` or `esp_mqtt_client_enqueue`?
And you are publishing with QoS>0 right?

Anyway the behavior still looks correct from my perspective. You are exhausting the memory so the outbox can't allocate more data and hence it's not possible to create a new message. 

I noticed that for QoS1 we could have a better outbox handling, cleaning it up faster, I will further investigate and apply it if it's the case. 

For now, I will suggest the workaround of tracking the outbox size and handle the case of full outbox as a possibility in your scenario, waiting to publish when there is enough space.

kovid-wakefit, 2024-05-28:
Yes, We are using ```esp_mqtt_client_publish```. QOS is 1. 
Here is steps which we are following : 
```
do{
     ret = esp_mqtt_client_publish(client, topic_name, buffer, size, 1, 0);
     delay(for 100 ms);
     retry++;
 } while((ret == -1) && (retry <=5));
```
How can we check the outbox size runtime and how we can clean it up? Can you share the APIs or ESP doc link.

euripedesrocha, 2024-05-28:
You can use [esp_mqtt_client_get_outbox_size](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/mqtt.html#_CPPv431esp_mqtt_client_get_outbox_size24esp_mqtt_client_handle_t). 
It is also possible to set the [outbox limit](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/mqtt.html#_CPPv4N24esp_mqtt_client_config_t15outbox_config_t5limitE).

kovid-wakefit, 2024-06-04:
Any further updates on this?

euripedesrocha, 2024-06-04:
@kovid-wakefit as I mention the behavior here seems to be correct. 
You need to control the amount of memory used in the system. Have you tried the proposed outbox control functions?  
Another alternative is that if you have SPIRAM in your system, you can select the destination of outbox to that memory by setting "Use external memory for outbox data" in menuconfig.

closed 2024-09-16 by euripedesrocha

Vaibhav0216, 2025-01-21:
Hi @kovid-wakefit ,
I am facing same issue how you solved the problem? 
Can you please guide me ?

euripedesrocha, 2025-01-21:
@Vaibhav0216 I suggest you open a new issue with the added details of your case. 

The behavior described here is the expected one. Take a look at the memory usage of the mqtt client and the ways to control it as pointed above.