espressif/esp-idf #13868: Should ESP-IDF support Nushell?

dobairoland, 2024-05-28:
### Describe the solution you'd like.

[Nushell](https://www.nushell.sh/) is a new shell supported on all platforms which ESP-IDF supports. We would like to ask you to let us know if you would like it to be officially supported as one of the shells for ESP-IDF.

Please react to this feature request, or leave us your opinion if you know it and use it. There is also a [PR](https://github.com/espressif/esp-idf/pull/13834) with an experimental Nushell export script for ESP-IDF.

portasynthinca3, 2024-05-31:
As the author of the aforementioned pull request, I would like to state that I would prefer it if ESP-IDF did, indeed, support Nushell :)

ChristianPavilonis, 2024-09-12:
I use nushell btw.

fhrbata, 2024-09-17:
Hello, the master branch now includes updated export scripts that utilize a new `tools/activate.py` script. Although there are still some bugs to fix, you can try using it with nushell. Note that for nushell, it doesn't function as an export script that modifies your current shell context; instead, it launches a new shell with the esp-idf environment configured.

Running `./tools/activate.py` from within nushell (nu) should spawn a new nushell where you can use esp-idf commands. You can exit this new shell with `exit` or `ctrl+d`. While nushell isn't supported through the export scripts, using `tools/activate.py` directly should enable you to work with nushell.

fhrbata, 2025-05-23:
Hello @Gremious ,

thank you for trying `/tools/activate.py`. You can use the `-d` debug option to see which virtual environment is being used. Check if you have any ESP-IDF environment variables, like `IDF_PYTHON_ENV_PATH`, set in your Nushell RC scripts (`~/.config/nushell/env.nu`). If they are set there, they will replace the values set by the activation script. This could be why the activation seems to work, but `idf.py` doesn't. To troubleshoot, after activation, run `echo $env.IDF_PYTHON_ENV_PATH` to make sure it matches the value shown by the `activate.py` script in debug mode. But maybe there is another issue.

It appears to be working for me, though.

```bash
$ ./tools/activate.py -s nu -d
debug:  command line: ['/home/fhrbata/work/esp-idf/tools/export_utils/activate_venv.py', '-s', 'nu', '-d']
Activating ESP-IDF 6.0
Setting IDF_PATH to '/home/fhrbata/work/esp-idf'.
debug:  IDF_PYTHON_ENV_PATH /home/fhrbata/.espressif/python_env/idf6.0_py3.12_env
* Checking python version ... 3.12.8
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... debug:  Shell explicitly stated: "nu"
nu
* Detecting outdated tools in system ... OK - no outdated tools found
debug:  Temporary script file path: /tmp/esp_idf_activate_fhrbata/activate_9smq8crh
Starting new 'nu' shell with ESP-IDF environment... (use "exit" command to quit)
     __  ,
 .--()°'.' Welcome to Nushell,
'|, . ,'   based on the nu language,
 !_-(_\    where all data is structured!

Please join our Discord community at https://discord.gg/NtAbbGn
Our GitHub repository is at https://github.com/nushell/nushell
Our Documentation is located at https://nushell.sh
Tweet us at @nu_shell
Learn how to remove this at: https://nushell.sh/book/configuration.html#remove-welcome-message

It's been this long since Nushell's first commit:
6yrs 12days 21hrs 37mins 5secs 328ms 949µs 507ns 

Startup Time: 9ms 88µs 825ns

~/work/esp-idf> echo $env.IDF_PYTHON_ENV_PATH                                                                  05/23/2025 04:36:17 PM
/home/fhrbata/.espressif/python_env/idf6.0_py3.12_env
~/work/esp-idf> idf.py --version                                                                               05/23/2025 04:36:24 PM
ESP-IDF v6.0-dev-309-g8afb0fb9e683
~/work/esp-idf>                                                                                                05/23/2025 04:36:30 PM

```

Thank you!