espressif/esp-idf #13870: BLE gattc service cache information is not deleted whent the configured BLE namespace is deleted

sergiutigu, 2024-05-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

when the configured namespace that stores the bonding information for a particular device is deleted , the cached service information to be also deleted.

### What is the actual behavior?

when the configured namespace that stores the bonding information for a particular device is deleted , the cached service information is not deleted. and there is no way of deleting that, except deleting the entire partition, which we can't do since we have data than we don't want to loose.

### Steps to reproduce.

1. In the menuconfig enable CONFIG_BT_GATTC_CACHE_NVS_FLASH.
2.  Start BLE as a central.
3. Set the namespace name that keeps the bonding information: 
  ErrorCode_s32 = esp_ble_config_file_path_update((char_t*)Filename_pac);
    if (ErrorCode_s32 != ESP_OK) {
    }
4. Bond a peripheral 
5. discover the services. This should save them into flash.
6. Disconnect.
7. delete the Filename_pac. This deletes the bonding information only.
8. Start the BLE again as central.
9. Bond again the peripheral.
10. Services are still stored inside nvs. 
...

zhp0406, 2024-05-29:
If you need to delete the service information within the NVS, you can use the API esp_ble_gattc_cache_clean.

[image]

exilat, 2024-06-14:
The issue here is that can only be applied if you BLE stack is running. 
But the need to deleted the nvs namespace is coming from the need to perform a factory setting of the device.
This must be implemented regardless of the state of the BLE stack.
What i am saying that there is a missing functioanlity in the BLE stack, and that is blocking us to implement factory settings,
Till now we could avoid it but not on the new platform.

sergiutigu, 2024-07-18:
is there any input that is expected from my side on this matter?

Weijian-Espressif, 2024-08-09:
1. esp_ble_config_file_path_update is not allowed to be called while the Bluetooth stack is running, and can only be used before bluedroid init, which is now supported by IDF v5.2 and above. esp_err_t esp_bt_config_file_path_update(const char *file_path);
2. f you need to delete the service information within the NVS, you can use the API esp_ble_gattc_cache_clean.

esp-zhp, 2024-08-09:
@sergiutigu 
"The issue is that this can only be applied if your BLE stack is running. You are correct.

Could you describe your use case? Why do you need to use this related API when the BLE stack is not running?"

sergiutigu, 2024-08-12:
we need to perform a factory settings. This can happen while the BT are running or not. 
Also we have 2 version of factory settings: BT Factory Settings/Wifi Factory Settings.
So we need to be able to clear "all BT specific only" nvs files, whiteout affecting the Wifi information stored in flash while BT service are off.
Factory settings performs a reset, and we do not have time to start the BT stack before the reset comes.
We only have time to delete files, namespace from nvs space.

esp-zhp, 2024-08-13:
.Based on the Bluedroid host architecture, the GATTC cache can only be cleared after Bluedroid has been initialized. We do not currently support this method. Please consider alternative approaches.clear the corresponding GATTC cache after initialization.

closed 2024-12-17 by Alvin1Zhang