espressif/esp-idf #13871: KConfig option SECURE_UART_ROM_DL_MODE unnecessarily depends on SECURE_BOOT_V2_ENABLED

jrahlf, 2024-05-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Target: esp32s3, but possibly others
Currently one cannot activate SECURE_ENABLE_SECURE_ROM_DL_MODE without also enabling secure boot or flash encryption.
This might make sense for devices with external flash, but there are chips like ESP32-S3FN8, which have internal flash. For these, one might just want to prevent firmware readout by enabling `SECURE_ENABLE_SECURE_ROM_DL_MODE=y`, without the disadvantages of secure boot and flash encryption.

Suggestion:
Remove `SECURE_BOOT_V2_ENABLED || SECURE_FLASH_ENC_ENABLED` dependency.
Instead give a textual warning that the option should normally be used in conjunction with secure boot or flash encryption.

AdityaHPatwardhan, 2024-06-06:
Hi @jrahlf We have the API for enabling secure ROM DL mode through the application.
https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/efuse/include/esp_efuse.h#L360
You can call this API at start to enable the Secure ROM DL mode irrespective of the status of security features.

mahavirj, 2024-07-08:
Closing, hope that API based approach as suggested above works for your case.

closed 2024-07-08 by mahavirj