espressif/esp-idf #13872: BLE there is no way of finding out the address type of the bonced peripheral if the IRK is not forwarded.

sergiutigu, 2024-05-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

We need to filter to allow connection only from bonded devices.
 For that purpotse we are using esp_ble_gap_update_whitelist() which requires address  type. 
Bonded peripheral address type to be available in the bonding information without the IRK being sent by the peripheral.

### What is the actual behavior?

Bonded peripheral address type is only available in the bonding information if the peripheral sents IRK.

### Steps to reproduce.

We need to filter the conection to allow connection only of the bonded devices.
For that we are using :
esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type)
the last parameter is the address type.
Now we are getting this information from the bonded list:
esp_err_t esp_ble_get_bond_device_list(int *dev_num, esp_ble_bond_dev_t *dev_list)
Definition of:
/**
* @brief  struct type of the bond device value
*/
typedef struct
{
    esp_bd_addr_t  bd_addr;               /*!< peer address */
    esp_ble_bond_key_info_t bond_key;     /*!< the bond key information */
} esp_ble_bond_dev_t;                     /*!< the ble bond device type */

/**
* @brief  struct type of the bond key information value
*/
typedef struct
{
    esp_ble_key_mask_t    key_mask;       /*!< the key mask to indicate witch key is present */
    esp_ble_penc_keys_t   penc_key;       /*!< received peer encryption key */
    esp_ble_pcsrk_keys_t  pcsrk_key;      /*!< received peer device SRK */
    esp_ble_pid_keys_t    pid_key;        /*!< peer device ID key */
} esp_ble_bond_key_info_t;                /*!< ble bond key information value type */

/**
* @brief  BLE pid keys
*/
typedef struct
{
    esp_bt_octet16_t          irk;           /*!< The irk value */
    esp_ble_addr_type_t       addr_type;     /*!< The address type */
    esp_bd_addr_t             static_addr;   /*!< The static address */
} esp_ble_pid_keys_t;

Basically you can get the addr_type only if the :
**    esp_ble_pid_keys_t    pid_key;        /*!< peer device ID key */** is filled. Which only happens if the IRK is being sent.
But as standard mentions, IRK is optional and only required for privacy.
The address type should be available regardless.
What is happening right now is that one of the peripheral that we connect to, doesn't provide the IRK. We can pair it, but then we can't re-connect to it since we need to filter at the reconnect only the bonded devices.
So either whitelist doesn't require the address type( don't know if that is possible) or the BLE stack needs to provied address type whithout IRK being sent.

sergiutigu, 2024-07-18:
is there any input expected from me on this issue?

esp-zhp, 2024-08-07:
We no longer maintain version 4.3. so, I have created a patch based on the latest master (commit 327683185c960ae145107fa54aea8442d779a9f0). This patch allows you to retrieve the address types of devices in the bond list.
[0001-feat-ble-bluedroid-Support-getting-BLE-address-type-.zip](https://github.com/user-attachments/files/16522716/0001-feat-ble-bluedroid-Support-getting-BLE-address-type-.zip)

esp-zhp, 2024-08-07:
If you encounter any issues during use this patch, please let me know.

closed 2024-09-11 by Indastri