espressif/esp-idf #13874: Acting as a USB Composite device with multiple CDC's

Flamabalistic, 2024-05-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is it possible for a device (ESP32S3) to show up as 2 seperate USB CDC ports? 

From what I've found, it can be done using vanilla tinyusb [link](https://github.com/hathach/tinyusb/blob/master/examples/device/cdc_dual_ports/src/main.c), and it is possible to have multiple USB configurations [link](https://github.com/espressif/esp-idf/blob/003f3bb5dc7c8af8b71926b7a0118cfc503cab11/examples/peripherals/usb/device/tusb_composite_msc_serialdevice/main/tusb_composite_main.c), however I can't seem to find anything on having multiple of the same type.

Any ideas?

roma-jam, 2024-05-29:
Hi @Flamabalistic 

You can refer to `tusb_serial_device` example in `esp-idf` ([link](https://github.com/espressif/esp-idf/tree/003f3bb5dc7c8af8b71926b7a0118cfc503cab11/examples/peripherals/usb/device/tusb_serial_device))

In the README you can find information how to enable double serial device (refer to section [How to use example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_serial_device#how-to-use-example))

After configuring, the USB CDC device will have two different ports while attached.

Flamabalistic, 2024-05-29:
Thank you!

Don't know how I missed the 2 CDC's in the example.

closed 2024-05-29 by Flamabalistic