espressif/esp-idf #13878: SPI Flash not getting detected

RahulGoyal-tech, 2024-05-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have created a custom ESP32S3 Board but when i am running esptool flash_id, or any other command, my ESP32S3 is not abe to identify flash ic. 
Gives following errors. How can i resolve this?
[image]

Alvin1Zhang, 2024-07-02:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-07-02 by Alvin1Zhang

RahulGoyal-tech, 2024-11-19:
@dobairoland @Alvin1Zhang Sorry for inactivity. Resolved this. Was a hardware issue