espressif/esp-idf #13881: Measuring DC power usage / voltage /current on Esp32-C6 module

ilker-aktuna, 2024-05-30:
### Is your feature request related to a problem?

Is it possible to measure  DC power usage on Esp32-C6 module (esp32-c6-devkit) without any additional hardware or connection ?

ilker-aktuna, 2024-06-15:
I'd appreciate any information please.

jpeg2600, 2024-06-17:
No, it's not possible

ilker-aktuna, 2024-06-17:
too bad. On Arduino it is possible with some boards on Arduino.
```

#include "esp_adc_cal.h"  

 esp_adc_cal_characteristics_t adc_chars;
 if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF) {
    vref = adc_chars.vref;
  }
  float voltage = analogRead(36) / 4096.0 * 6.566 * (vref / 1000.0);
```