espressif/esp-idf #13883: Add ability to have I2S driver in IRAM

nvx, 2024-05-30:
### Is your feature request related to a problem?

I have some code using the I2S driver and have noticed the first time it runs it's slow, but subsequent runs once the code is in the cache runs fast. My code is already in IRAM but it's leveraging the I2S driver which is not.

### Describe the solution you'd like.

The ability to enable the I2S to live in IRAM, or some easy way to flag specific functions from the IDF to be in IRAM would be a more general solution.

Specifically the following functions I'd love to see in IRAM:
* i2s_channel_enable
* i2s_channel_disable
* i2s_channel_read
* i2s_channel_write
* i2s_channel_preload_data

### Describe alternatives you've considered.

Modifying the ESP-IDF to make the required functions be in IRAM would work, but it would be difficult to maintain.

suda-morris, 2024-05-31:
> Modifying the ESP-IDF to make the required functions be in IRAM would work

Alternatively, you can control the function placement even in your linker.lf file. Example is here: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/spi_master/hd_eeprom/components/eeprom/linker.lf

Documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/linker-script-generation.html

nvx, 2024-05-31:
> Alternatively, you can control the function placement even in your linker.lf file. Example is here: [...]

That worked! From reading the docs it sounded like the linker script fragment had to be in the right component, but in hindsight it makes perfect sense that it doesn't matter which component it's in since linking happens all at once at the very end.

closed 2024-05-31 by nvx