espressif/esp-idf #13885: 为什么pClient->connect 后不会执行任何代码了？和卡死了一样

bbhxwl, 2024-05-31:
我下面代码，
不会输出Connected successfully!，也不会输出Failed to connect.，请问是硬件的bug么？
```
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEClient.h>
#include <BLEScan.h>
BLEClient* pClient;
BLEScan* pBLEScan;
const char* serviceUUID = "0000ffe5-0000-1000-8000-00805f9a34fb";
const char* charUUID = "0000ffe4-0000-1000-8000-00805f9a34fb";

// 自定义回调类，用于处理发现的 BLE 设备
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
  public:
    void onResult(BLEAdvertisedDevice advertisedDevice) {
        String name=advertisedDevice.getName().c_str();
        String mac=advertisedDevice.getAddress().toString().c_str();
        if(name.indexOf("WT901")!=-1||name.indexOf("InnoWT")!=-1){
          Serial.println(name);
          // 连接到设备
          pBLEScan->stop();
          Serial.println("1");
          pClient = BLEDevice::createClient();
          Serial.println("2");
          bool connected = pClient->connect(&advertisedDevice);
          if (connected) {
            Serial.println("Connected successfully!");
          } else {
            Serial.println("Failed to connect.");
          }
          

        }
    }
};

void setup() {
  Serial.begin(9600);
  BLEDevice::init("Inno");
  pBLEScan=BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);

}

void loop() {
  Serial.println("hello");
  pBLEScan->start(5);
  delay(5000);
}

```

weiyuhannnn, 2024-06-04:
@bbhxwl  Thanks for using the Espressif product. May I know which chip do you use?

Alvin1Zhang, 2025-03-14:
Thanks for reporting, will close due to short of details, feel free to reopen with more details.

closed 2025-03-14 by Alvin1Zhang